arendst/Tasmota #5886: PZEM-004T how to get values such as power factor and send to domoticz?

xury77, 2019-05-30:
Hallo. I using Tasmota on ESP-01  board with PZEM-004T.
Its works well, but I want to get values such as power factor,apparent power etc. and send to dummy sensor in domoticz. In Tasmota web configuration is not possible, but is there a solution to get it using console, rules, or another way?

ascillato2, 2019-05-30:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-05-30 by ascillato2

ammous, 2024-12-22:
> Hallo. I using Tasmota on ESP-01 board with PZEM-004T. Its works well, but I want to get values such as power factor,apparent power etc. and send to dummy sensor in domoticz. In Tasmota web configuration is not possible, but is there a solution to get it using console, rules, or another way?

hello ; can you help me, how did you do for programming esp (firmware) and the connection with PZEM