arendst/Tasmota #5887: INA219 with Wemos D1 Mini

vwtune, 2019-05-30:
I have a tasmota flashed (sensor version) wemos D1 mini, with version 6.5.0. I would like to monitor the voltage of a 12volt accu. I connected the ina219 on D4: SDA, D5: SCL, G: GND and 3V3: VCC.
The problem i have, in tasmota i can't see any values of the ina219. in the main window it shows: analog0    5
In get this in console: 
15:57:31 MQT: tele/WemosD1Mini2/SENSOR = {"Time":"2019-05-30T15:57:31","ANALOG":{"A0":5}}
15:57:45 CMD: I2Cscan 
15:57:45 MQT: stat/WemosD1Mini2/RESULT = {"I2CScan":"Error 4 at 0x01"}

Please help me out here, i realy would love to get this to work.

ascillato2, 2019-05-30:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

vwtune, 2019-05-30:
> Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you? [...]

what do you mean with template missing? Sorry, do i have to open a new issue? Is it the wrong section?

ascillato, 2019-05-30:
Please, click on the link provided (troubleshooting template), choose troubleshooting, and then complete all the information requested there. That will provide us with a lot of information for helping you.

Also remember that it is available the Tasmota Support Chat.

closed 2019-05-30 by ascillato2