arendst/Tasmota #5895: Enhancement of the color command (RGBW)

Vujagig, 2019-05-31:
An enhancement of the color command:
color 12 (using the real white)
color 13 (using the rgb & real white)

I noticed that Alexa does not know the command amber / bernstein and does not execute the command white at RGBW
A change from Scheme to Color is not possible via Alexa.
When a scheme is activated, the color command has no effect
The white command has FFF0F0 and color 12 FFFFFF. A bug?

ascillato2, 2019-05-31:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato, 2019-05-31:
Hi, sorry. There is no enough information. Please, use the troubleshooting template with status 0 output and the console output when this happens to you. Also, remember to have weblog set to 4 as the troubleshooting template ask for. Thanks

meingraham, 2019-05-31:
`color 12 (rgb white) command` - works in Africa ;-)

Jason2866, 2019-05-31:
Use latest dev version. This issues is solved with PR #5878

s-hadinger, 2019-05-31:
I have on my todo list a way to mix RGB and White seamlessly. Stay tuned.

ascillato2, 2019-06-01:
@s-hadinger Cool!!! Thanks.

PR #6032 (2019-07-07): Add blend RGB leds with White leds for better whites #5895 #5704
## Description:

When setting last argument of `rgbwwtable`to zero, like this `rgbwwtable 255,255,255,255,0`, color will be automatically blended between RGB and White. Any White component will be taken off RGB and used on White leds.

Changing the 4th argument can be used to calibrate White vs RGB: often white leds are more bright than RGB. For example, Sonoff B1 would require `rgbwwtable 255,255,255,35,0`.

Works only with lights with 4 (RGBW) or 5 (RGBCW) channels.

@arendst I bumped to version to 6.6.0.1.

**Related issue (if [... cut]
closed 2019-06-01 by ascillato2