arendst/Tasmota #5896: access countdown timer value in a rule

th-s-op, 2019-05-31:
Hi, I am using rules, and there is a countdown timer, i.e.

ruletimer1 3600

now when there is another event while the timer is still counting down, i.e. a button press, I want to access the countdown timer value at that time. Ideally, something like

Rule1 on Button1#State=1 do var1 %ruletimer1% done

(example: if I initialized the rule timer with 60 minutes and would press Button1 after 10 minutes, the %ruletimer1% value would be 50 minutes (or 3000).

Of course, there is nothing like a %ruletimer1% substitution, but would it be possible to implement it?

With such a feature, I could implement a state machine that is somewhat more complex - one timer being interrupted by something else, and continuing later (by saving a timer state/value, redefining it with a new value or stopping it, and later resetting it to the saved value).

meingraham, 2019-05-31:
Just spitballing, but if you invoke the command `RuleTimer` without a parameter, it returns the current value of every timer in a JSON payload. Not sure if you can use `#Tn` from the RESULT.

Alternatively, you may be able to do something with scripting instead of rules.

In any case, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. You may get some people with more ideas.

If efforts with rules and scripting fail, then you can submit a formal feature request **ensuring that you complete the template as requested**. The lack of that requested information will result in your request not receiving much attention. If you cannot take a few minutes to provide the requested information, would you expect people to take the time to respond?

Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

th-s-op, 2019-05-31:
Your supposed solution would work if there'd be an external logic, but I want pure Tasmota-inside logic. From my point of view, this is a feature request.

Feature: implement %ruletimer1% .. %ruletimer8% text replacements within rules (like %var1%, %time% and so on)

I usually fill out the template, but I see no need for it in this case.

meingraham, 2019-05-31:
Kewl. I suppose you also looked into this - https://github.com/arendst/Sonoff-Tasmota/blob/8ab60add17b1fe94340348ec4970d261c66092cd/scripter.md

But you've got it handled. Regards.

Jason2866, 2019-05-31:
@th-s-op  There will be no enhancement in rules for this feature you want.
There is a more powerful alternative as already @meingraham  mentioned.
Take a look at the new scripter engine. If this doesnt solve your wishes, sorry.
Tasmota is designed to be kept small and powerful and includes as many functions for general use cases.

th-s-op, 2019-05-31:
Thanks, both! I really missed that one. Great alternative I will immediately try!

closed 2019-05-31 by th-s-op