arendst/Tasmota #5897: Power monitoring improvements request

stepanov1975, 2019-05-31:
Hi,
I have a couple of feature requests.

1. Over/under voltage protection
AFAIU VoltageHigh, VoltageLow currently thresholds for MQTT reporting. It will be useful if it will be possible to turn off the load also based on those thresholds.

2. Power reporting
Power reporting thresholds currently set by PowerDelta in percents. The problem with percents always near zero. Usually, it is not very important to send a report when power changes from 1W to 1.1W (10%), but important when it changes from 1000W to 1100W that is also 10%. It will be better if PowerDelta will be in wats, but I suppose for back compatibility it is problematic to change this now. So it will be good if an additional parameter will be added that will prevent MQTT messages for small changes (set minimum change limit in W). I know it is PowerLow parameter, but I don't want to lose all low power reports completely.
Also, the possibility to set a minimum time between MQTT reports will be helpful. It seems now when power changes more than PowerDelta it will be reported each second. This may create a substantial load on the monitoring system for multiple devices.

Thank you.

Jason2866, 2019-05-31:
There are many features nice to have. Please keep the very very limited resources of the device in mind.
The features you want can be realized with rules or the new scripter engine.
Thx for your suggestions. 

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

meingraham, 2019-05-31:
@stepanov1975 

#1 Code a rule that triggers on `ENERGY#Voltage` or `ENERGY#Current`, or even `ENERGY#Power`. Since setting the thresholds cause telemetry to be posted, you can trigger on the value, e.g., `ON ENERGY#Voltage>150 DO Power OFF ENDON'

#2 I agree that percentages are problematic. An enhancement to `PowerDelta` could be 0-100 - treat as percentage. Greater than 100 - treat as an absolute offset by 100, e.g., `PowerDelta 500` would set a 400W threshold.

In the meantime... you can also use a rule to adjust `PowerDelta` based on the current Power value. Using the one-shot feature of rules, you can make the rule send a message only once when the threshold is crossed.

```
Backlog PowerDelta 0; Rule1 0; Rule1 5
 
Rule1 
ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK 
ON ENERGY#Power>=15 DO Backlog PowerDelta 25; Status 8 BREAK 
ON ENERGY#Power>5 DO Backlog PowerDelta 35; Status 8 BREAK 
ON ENERGY#Power<=5 DO Backlog PowerDelta 100 ENDON

Rule1 1
```

Which translates to:
```
Rule Pseudo Code
IF ENERGY#Power>=35  // ENERGY#Power GE 35
  DO Backlog PowerDelta 10; Status 8
ELSE IF ENERGY#Power>=15  // ENERGY#Power GE 15 and LT 35
  DO Backlog PowerDelta 25; Status 8
ELSE IF ENERGY#Power>5  // ENERGY#Power GT 5 and LT 15
  DO Backlog PowerDelta 35; Status 8
ELSE  // ENERGY#Power changed (i.e. LE 5)
  DO PowerDelta 100
```

Mike

closed 2019-06-01 by ascillato2