arendst/Tasmota #5901: Absolute PowerDelta

meingraham, 2019-06-01:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem?**
No - this is an enhancement request

**Describe the solution you'd like**
An enhancement to `PowerDelta` could be 0-100 - treat as percentage (to retain backward compatibility). PowerDelta greater than 100 - treat as an absolute power value change threshold offset by 100, e.g., `PowerDelta 500` would set a 400W threshold.

**Describe alternatives you've considered**
```
Backlog PowerDelta 0; Rule1 0; Rule1 5
 
Rule1 
ON ENERGY#Power>=35 DO Backlog PowerDelta 10; Status 8 BREAK 
ON ENERGY#Power>=15 DO Backlog PowerDelta 25; Status 8 BREAK 
ON ENERGY#Power>5 DO Backlog PowerDelta 35; Status 8 BREAK 
ON ENERGY#Power<=5 DO Backlog PowerDelta 100 ENDON

Rule1 1
```
**Additional context**
Using a rule to adjust `PowerDelta` based on the current Power value improves the situation, but it does not solve the problem when devices connected to the power monitor have large operating power ranges.

Cheers!

Mike

arendst, 2019-06-05:
Nice idea but currently the max allowed value for the variable is 255 (uint8_t).

Will consider.

meingraham, 2019-10-05:
@arendst Any re-consideration for this?

Along with this #6404

arendst, 2019-10-06:
Will consider this one.

Won't consider the sensor one as it will need too many resources; a delta register for all available sensor types.

meingraham, 2019-10-17:
I'm rooting for this one, obviously. But implementing this one would get us to just one issue remaining. Go @arendst, go! Go @arendst, go! :wink:

arendst, 2019-10-18:
How can I resist....

arendst, 2019-10-18:
Pfew.

Jason2866, 2019-10-18:
Michael is happy as a child😀

andrethomas, 2019-10-18:
Go @meingraham Go... close this one :)

meingraham, 2019-10-18:
Thank you. Wiki updated!

closed 2019-10-18 by meingraham