arendst/Tasmota #5903: Device not detected by Alexa on Echo Plus

smoebot, 2019-06-02:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  I have a Brilliant Smart Wifi Plug that was flashed with tuya convert.  Immediately after flashing I upgraded to the latest firmware, 6.5.0.13 using 2.5.2 core binaries , sourced from http://thehackbox.org/tasmota/020502/
I have applied and activated the template from here:
https://blakadder.github.io/templates/brilliant_plug.html

Alexa does not detect the device.  I am using an Amazon Echo Plus 1st Generation, with built in hub

I have tried both Belkin WeMo and Hue Bridge emulation, and I get the same result: :No new devices found"

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Brilliant Smart Wifi Plug
- [X] Tasmota binary firmware version number used : 6.5.0.13 (pre-compiled)
- [X] Development IDE - Compiler / Upload tools used : N/A
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
11:15:02 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:15:02 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:21","StartupUTC":"2019-06-02T10:11:41","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":64,"SaveAddress":"FA000"}}
11:15:02 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.13(f61f095-sonoff)","BuildDateTime":"2019-05-31T22:02:03","Boot":4,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
11:15:02 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["aegaeon","aegaeon"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
11:15:02 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","001783A0","23B617CC","01003BC0"]}}
11:15:02 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7944","IPAddress":"192.168.0.8","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:E5:9F:08","Webserver":2,"WifiConfig":4}}
11:15:02 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E59F08","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:15:02 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 02 10:15:02 2019","Local":"Sun Jun 02 11:15:02 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
11:15:02 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-02T11:15:02"}}
11:15:02 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-02T11:15:02","Uptime":"0T00:03:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"aegaeon","BSSId":"54:25:EA:38:84:A0","Channel":3,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:04"}}}11:15:02 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:15:02 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:21","StartupUTC":"2019-06-02T10:11:41","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":64,"SaveAddress":"FA000"}}
11:15:02 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0.13(f61f095-sonoff)","BuildDateTime":"2019-05-31T22:02:03","Boot":4,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
11:15:02 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["aegaeon","aegaeon"],"TelePeriod":300,"Resolution":"55C180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
11:15:02 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":428,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","001783A0","23B617CC","01003BC0"]}}
11:15:02 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-7944","IPAddress":"192.168.0.8","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:E5:9F:08","Webserver":2,"WifiConfig":4}}
11:15:02 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E59F08","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:15:02 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 02 10:15:02 2019","Local":"Sun Jun 02 11:15:02 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
11:15:02 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-02T11:15:02"}}
11:15:02 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-02T11:15:02","Uptime":"0T00:03:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"aegaeon","BSSId":"54:25:EA:38:84:A0","Channel":3,"RSSI":68,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```
**(Please, remember to close the issue when the problem has been addressed)**

smoebot, 2019-06-02:
It was suggested by another user on the sonoff-users group that I tag @s-hadinger

Jason2866, 2019-06-02:
Please try latest version from a few minutes

s-hadinger, 2019-06-02:
The code for Alexa discovery hasn't changed for some time. It has probably no connection with the latest lightId change.
Can you please set log level 4 and cut/paste logs when you trigger an Alexa discover?

smoebot, 2019-06-02:
these are the logs during the alexa discover period:
```23:01:54 APP: Boot Count 11
23:01:54 CFG: Saved to flash at F8, Count 66, Bytes 3584
23:02:00 RSL: tele/sonoff/UPTIME = {"Time":"2019-06-02T23:02:00","Uptime":"0T00:00:16"}
23:02:10 WIF: Checking connection...
23:02:10 WIF: Connected
23:02:30 WIF: Checking connection...
23:02:30 WIF: Connected
23:02:50 WIF: Checking connection...
23:02:50 WIF: Connected
23:03:10 WIF: Checking connection...
23:03:10 WIF: Connected
23:03:30 WIF: Checking connection...
23:03:30 WIF: Connected```

meingraham, 2019-06-02:
It never completes booting? This shows uptime of only 16 seconds.

What is your logging level (`WegLog`)?

smoebot, 2019-06-02:
Here's some more logs.
Device is booted, can access the web GUI fine, and have performed an alexa discovery.  I've also updated to 6.5.0.14 as per request from @Jason2866 

```00:00:00 CFG: Loaded from flash at FA, Count 72
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.5.0.14(d0502c1-sonoff)-2_5_2
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 aegaeon in mode 11N as sonoff-7944...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-7944 with IP address 192.168.0.8
00:00:04 UPP: Multicast (re)joined
00:00:04 APP: (UTC) Sun Jun 02 22:23:02 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
23:23:08 APP: Boot Count 14
23:23:08 CFG: Saved to flash at F9, Count 73, Bytes 3584
23:23:09 HTP: Information
23:23:22 WIF: Checking connection...
23:23:22 WIF: Connected
23:23:42 WIF: Checking connection...
23:23:42 WIF: Connected
23:24:02 WIF: Checking connection...
23:24:02 WIF: Connected
23:24:22 WIF: Checking connection...
23:24:22 WIF: Connected
23:24:42 WIF: Checking connection...
23:24:42 WIF: Connected
23:25:02 WIF: Checking connection...
23:25:02 WIF: Connected
23:25:22 WIF: Checking connection...
23:25:22 WIF: Connected
23:25:42 WIF: Checking connection...
23:25:42 WIF: Connected
23:26:02 WIF: Checking connection...
23:26:02 WIF: Connected
23:26:22 WIF: Checking connection...
23:26:22 WIF: Connected
23:26:42 WIF: Checking connection...
23:26:42 WIF: Connected
23:27:02 WIF: Checking connection...
23:27:02 WIF: Connected
23:27:22 WIF: Checking connection...
23:27:22 WIF: Connected
23:27:42 WIF: Checking connection...
23:27:42 WIF: Connected
23:27:57 RSL: tele/sonoff/STATE = {"Time":"2019-06-02T23:27:57","Uptime":"0T00:04:59","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"aegaeon","BSSId":"54:25:EA:38:84:A0","Channel":3,"RSSI":70,"LinkCount":1,"Downtime":"0T00:00:04"}}
23:28:02 WIF: Checking connection...
23:28:02 WIF: Connected
23:28:22 WIF: Checking connection...
23:28:22 WIF: Connected
23:28:26 HTP: Main Menu
23:28:30 HTP: Console
23:28:42 WIF: Checking connection...
23:28:42 WIF: Connected
23:29:02 WIF: Checking connection...
23:29:02 WIF: Connected
23:29:22 WIF: Checking connection...
23:29:22 WIF: Connected
23:29:42 WIF: Checking connection...
23:29:42 WIF: Connected
23:30:02 WIF: Checking connection...
23:30:02 WIF: Connected
23:30:22 WIF: Checking connection...
23:30:22 WIF: Connected
23:30:42 WIF: Checking connection...
23:30:42 WIF: Connected
23:31:02 WIF: Checking connection...
23:31:02 WIF: Connected
23:31:10 CMD: weblog
23:31:10 SRC: WebConsole from 192.168.0.17
23:31:10 RSL: Received Topic /weblog, Data Size 0, Data 
23:31:10 RSL: Group 0, Index 1, Command WEBLOG, Data 
23:31:10 RSL: stat/sonoff/RESULT = {"WebLog":4}```

s-hadinger, 2019-06-03:
This is what I suspected. Your Tasmota device didn't receive a single UDP packet.

Philips Hue and Belkin Wemo discovery relies on UDP/Multicast. Isn't multicast blocked at your router level?

smoebot, 2019-06-03:
I'll investigate and confirm.  

I do have actual Philips Hue lights that are discovered without issue

smoebot, 2019-06-03:
@s-hadinger @meingraham @Jason2866 
Device is now responding.  It was entirely related to multicast, it seems.  Even with root access on my ISP provided router, I was not able to configure anything to allow the multicast to be handed along.
Workaround - put another (better) router in Wireless router mode, connected to the ISP router.  This second router will now handle all of the home IoT stuff - Alexa, lights, whatever else.
Thanks for your help.
Logs for show:
```00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 hermes in mode 11N as sonoff-7944...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-7944 with IP address 192.168.1.102
00:00:04 UPP: Multicast (re)joined
00:00:05 CFG: Saved to flash at F9, Count 90, Bytes 3584
00:00:07 APP: (UTC) Mon Jun 03 10:27:42 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:27:45 APP: Boot Count 19
11:27:45 CFG: Saved to flash at F8, Count 91, Bytes 3584
11:27:59 WIF: Checking connection...
11:27:59 WIF: Connected
11:28:13 UDP: Packet (122)
11:28:13 UDP: Packet (125)
11:28:16 UDP: Packet (125)
11:28:19 WIF: Checking connection...
11:28:19 WIF: Connected
11:28:39 WIF: Checking connection...
11:28:39 WIF: Connected
11:28:59 WIF: Checking connection...
11:28:59 WIF: Connected
11:29:00 UDP: Packet (174)
11:29:01 UDP: Packet (174)
11:29:03 UDP: Packet (174)
11:29:04 UDP: Packet (174)
11:29:13 UDP: Packet (94)
11:29:13 UDP: Packet (101)
11:29:13 UDP: Packet (94)
11:29:13 UDP: Packet (101)
11:29:13 UDP: Packet (94)
11:29:13 UDP: Packet (122)
11:29:13 UDP: Packet (125)
11:29:15 UPP: Hue 3 response packets sent to 192.168.1.101:50000
11:29:15 HTP: Hue setup
11:29:15 HTP: Hue API ()
11:29:15 HTP: Hue POST args ({"devicetype": "Echo"})
11:29:15 HTP: Hue API (/e59f08/lights)
11:29:15 HTP: Hue Result ({"240775297":{"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"}})
11:29:15 HTP: Hue API (/e59f08/lights/240775297)
11:29:15 HTP: Hue Result ({"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"})
11:29:16 HTP: Hue API (/e59f08/lights/240775297)
11:29:16 HTP: Hue Result ({"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"})
11:29:16 HTP: Hue API (/e59f08/lights/240775297)
11:29:16 HTP: Hue Result ({"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"})
11:29:16 HTP: Hue API (/e59f08/lights/240775297)
11:29:16 HTP: Hue Result ({"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"})
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:18 UDP: Packet (199)
11:29:19 UDP: Packet (199)
11:29:19 WIF: Checking connection...
11:29:19 WIF: Connected
11:29:39 WIF: Checking connection...
11:29:39 WIF: Connected
11:29:45 HTP: Main Menu
11:29:58 SRC: WebGui from 192.168.1.100
11:29:58 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
11:29:58 RSL: stat/sonoff/POWER = OFF
11:29:58 CFG: Saved to flash at F7, Count 92, Bytes 3584
11:29:58 SRC: WebGui from 192.168.1.100
11:29:58 RSL: stat/sonoff/RESULT = {"POWER":"ON"}
11:29:58 RSL: stat/sonoff/POWER = ON
11:29:59 CFG: Saved to flash at F6, Count 93, Bytes 3584
11:29:59 WIF: Checking connection...
11:29:59 WIF: Connected
11:30:01 HTP: Console
11:30:13 UDP: Packet (122)
11:30:13 UDP: Packet (125)
11:30:16 UDP: Packet (125)
11:30:20 WIF: Checking connection...
11:30:20 WIF: Connected
11:30:40 WIF: Checking connection...
11:30:40 WIF: Connected
11:30:42 HTP: Hue API (/e59f08/lights/240775297)
11:30:42 HTP: Hue Result ({"state":{"on":true,"bri":254,"alert":"none","effect":"none","reachable":true},"type":"Extended color light","name":"Sonoff","modelid":"LCT007","uniqueid":"dc:4f:22:e5:9f:08:00:11-1","swversion":"5.50.1.19085"})
11:31:00 WIF: Checking connection...
11:31:00 WIF: Connected
11:31:00 UDP: Packet (174)
11:31:01 UDP: Packet (174)
11:31:03 UDP: Packet (174)
11:31:04 UDP: Packet (174)```

Jason2866, 2019-06-03:
Glad it works now. Since the question is answered, please close this issue.
Thx.

closed 2019-06-03 by ascillato2