arendst/Tasmota #5904: APDS9960 rule doesn't work

artafakhari, 2019-06-02:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Hello everyone
Unfortunately none of below rules work on the my tasmotized Wemos
rule on APDS9960#Ambient<100 do power on endon on APDS9960#Ambient>110 do power off endon
"on APDS9960#Long=1 do power toggle endon on APDS9960#Up=1 do dimmer + endon on APDS9960#Down=1 do dimmer - endon on APDS9960#Left=1 do ct + endon on APDS9960#Right=1 do ct - endon

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
16:37:50 RSL: stat/sonoff1/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff1","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:37:50 RSL: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:47:38","StartupUTC":"2019-06-02T13:50:12","Sleep":50,"CfgHolder":14617,"BootCount":12,"SaveCount":106,"SaveAddress":"3FA000"}}
16:37:50 RSL: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-06-02T17:05:16","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
16:37:50 RSL: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Arta",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
16:37:50 RSL: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":611,"Free":3464,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","0FFAE394","000783A0","63B617CE","00003BC0"]}}
16:37:50 RSL: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-1916","IPAddress":"192.168.1.231","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"3C:71:BF:29:47:7C","Webserver":2,"WifiConfig":4}}
16:37:50 RSL: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_29477C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
16:37:50 RSL: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 02 15:37:50 2019","Local":"Sun Jun 02 16:37:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:50","Sunset":"20:45"}}
16:37:50 RSL: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2019-06-02T16:37:50","DHT11":{"Temperature":26.2,"Humidity":19.0},"TempUnit":"C"}}
16:37:50 RSL: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2019-06-02T16:37:50","Uptime":"0T01:47:38","Vcc":3.053,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":1008,"POWER1":"ON","Dimmer":10,"Color":"1919","HSBColor":"0,0,10","Channel":[9,9],"CT":500,"Fade":"OFF","Speed":1,"LedTable":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"Arta","BSSId":"B0:4E:26:FA:64:7E","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
16:36:38 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-02T16:36:38","DHT11":{"Temperature":25.9,"Humidity":18.0},"APDS9960":{"Left":1},"TempUnit":"C"}
16:36:40 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-02T16:36:40","DHT11":{"Temperature":25.9,"Humidity":18.0},"APDS9960":{"Right":1},"TempUnit":"C"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
16:39:49 RSL: stat/sonoff1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":342,"Rules":"on APDS9960#Long do power toggle endon on APDS9960#Up do dimmer + endon on APDS9960#Down do dimmer - endon on APDS9960#Left do ct + endon on APDS9960#Right do ct - endon"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
This is the rule status in the console
16:39:49 RSL: stat/sonoff1/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":342,"Rules":"on APDS9960#Long do power toggle endon on APDS9960#Up do dimmer + endon on APDS9960#Down do dimmer - endon on APDS9960#Left do ct + endon on APDS9960#Right do ct - endon"}
This rule works perfectly:
16:44:10 RSL: stat/sonoff1/RESULT = {"POWER1":"OFF"}
16:44:10 RSL: stat/sonoff1/POWER1 = OFF
16:44:12 RUL: DHT11#TEMPERATURE>10.9 performs "power1 toggle"
16:44:12 RSL: stat/sonoff1/RESULT = {"POWER1":"ON"}
16:44:12 RSL: stat/sonoff1/POWER1 = ON

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2019-06-02:
Please try with prefix 'Tele-', so it becomes 'Tele-APDS9960#Up' and so on.

artafakhari, 2019-06-03:
It Worked !!

closed 2019-06-03 by artafakhari

ascillato2, 2019-06-03:
@artafakhari 

Hi, 

Please, can you add this example to the wiki? It will be very useful for other users too. Thanks

artafakhari, 2019-06-03:
I checked and see that staars did it
Without his advice it was so confusing.

Could you please help me a way to dim the light smoothly?

Staars, 2019-06-04:
I think, you want to use the FADE command. But as this is unrelated to the gesture sensor,  please consider to ask for help at the Discord-channel.

artafakhari, 2019-06-04:
@Staars I used Fade 1 command and it worked.
But faced another issue:
```
13:42:32 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:34 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:34","APDS9960":{"Long":1}}
13:42:34 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:34 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:42:34 RSL: stat/sonoff1/POWER2 = OFF
13:42:34 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:36 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:38 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:38","APDS9960":{"Long":1}}
13:42:38 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:38 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:42:38 RSL: stat/sonoff1/POWER2 = ON
13:42:38 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:41 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:43 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:43","APDS9960":{"Long":1}}
13:42:43 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:43 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:42:43 RSL: stat/sonoff1/POWER2 = OFF
13:42:43 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:45 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:47 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:47","APDS9960":{"Long":1}}
13:42:47 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:47 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:42:47 RSL: stat/sonoff1/POWER2 = ON
13:42:47 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:50 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:51 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:51","APDS9960":{"Long":1}}
13:42:51 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:51 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:42:51 RSL: stat/sonoff1/POWER2 = OFF
13:42:51 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:54 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:42:56 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:42:56","APDS9960":{"Long":1}}
13:42:56 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:42:56 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:42:56 RSL: stat/sonoff1/POWER2 = ON
13:42:56 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:42:58 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:00 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:00","APDS9960":{"Long":1}}
13:43:00 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:00 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:43:00 RSL: stat/sonoff1/POWER2 = OFF
13:43:00 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:43:02 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:04 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:04","APDS9960":{"Long":1}}
13:43:04 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:04 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:43:04 RSL: stat/sonoff1/POWER2 = ON
13:43:04 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:43:07 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:09 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:09","APDS9960":{"Long":1}}
13:43:09 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:09 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:43:09 RSL: stat/sonoff1/POWER2 = OFF
13:43:09 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:43:11 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:13 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:13","APDS9960":{"Long":1}}
13:43:13 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:13 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:43:13 RSL: stat/sonoff1/POWER2 = ON
13:43:13 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:43:16 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:17 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:17","APDS9960":{"Long":1}}
13:43:17 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:17 RSL: stat/sonoff1/RESULT = {"POWER2":"OFF"}
13:43:17 RSL: stat/sonoff1/POWER2 = OFF
13:43:17 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
13:43:20 RSL: tele/sonoff1/RESULT = {"Gesture":"On"}
13:43:22 RSL: tele/sonoff1/SENSOR = {"Time":"2019-06-04T13:43:22","APDS9960":{"Long":1}}
13:43:22 RUL: TELE-APDS9960#LONG performs "power2 toggle"
13:43:22 RSL: stat/sonoff1/RESULT = {"POWER2":"ON"}
13:43:22 RSL: stat/sonoff1/POWER2 = ON
13:43:22 RSL: tele/sonoff1/RESULT = {"Gesture":"Off"}
```

the sensor continuously goes into LONG state
I thought my sensor is damaged and replaced it with a new one. Also changed the cabling. But still I'm have this problem.

When I reboot the devices it works normal but after a while it happnes again

artafakhari, 2019-06-04:
I think I could solve the problem with this Command
```Sensor27 2```

Staars, 2019-06-04:
Yes, the sensor can be tricky and I tried to provide a solution with the special command, that you already found.

Good luck and best regards