arendst/Tasmota #5906: Compilation Error dev f61f095 and d0502c1

DavidFW1960, 2019-06-03:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
_A clear and concise description of what the bug is._
Compilation errors in last 2 dev versions 
I reported this already in the Discord chat but no response.
```
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development d0502c1/sonoff/settings.ino: In function 'void SettingsDefaultSet2()':
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development d0502c1/sonoff/settings.ino:826:43: error: 'IR_RCV_MIN_UNKNOWN_SIZE' was not declared in this scope
Settings.param[P_IR_UNKNOW_THRESHOLD] = IR_RCV_MIN_UNKNOWN_SIZE;
^
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development d0502c1/sonoff/settings.ino: In function 'void SettingsDelta()':
C:/Users/david/Documents/Sonoff-Tasmota/Sonoff-Tasmota-development d0502c1/sonoff/settings.ino:1184:47: error: 'IR_RCV_MIN_UNKNOWN_SIZE' was not declared in this scope
Settings.param[P_IR_UNKNOW_THRESHOLD] = IR_RCV_MIN_UNKNOWN_SIZE;Compiling .pioenvs\sonoff\lib0a5\ESP8266WiFi\ESP8266WiFiSTA.cpp.o

^
Compiling .pioenvs\sonoff\lib0a5\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pioenvs\sonoff\lib0a5\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pioenvs\sonoff\lib0a5\ESP8266WiFi\WiFiClientSecure.cpp.o
*** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1
```

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ ] Device used (i.e. Sonoff Basic) : _____
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Compiling with core 2.3.0

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expect to compile. No problem with commit 4d070bf

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

DavidFW1960, 2019-06-03:
Tried compiling with core 2.5.2 and error:
```
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff\firmware.elf] Error 1
```
I cleaned out the .platformio/platforms and .platformio/packages of all 8266 stuff as well.

DavidFW1960, 2019-06-03:
Tried compiling with stage and got this:
```
Generating .pioenvs\sonoff\core\core_version.h
fatal: No names found, cannot describe anything.
```
and then
```
xtensa-lx106-elf-g++: error: sonoff\sonoff.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pioenvs\sonoff\libf06\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pioenvs\sonoff\libf06\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pioenvs\sonoff\lib12c\Adafruit_SSD1306-1.1.2\Adafruit_SSD1306.cpp.o
*** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1
Compiling .pioenvs\sonoff\lib9cb\ESP8266SdFat\FatLib\StdioStream.cpp.o
========================== [ERROR] Took 86.69 seconds ==========================
```

ascillato, 2019-06-03:
Hi, please try again. Theo has done a fix. Thanks.

closed 2019-06-03 by ascillato2

DavidFW1960, 2019-06-04:
Compiled successfully core 2.3.0 - 2.5.2 is still too unreliable.