arendst/Tasmota #5907: POW R2 Serial Logging leads to crash at Status 0

tammo0, 2019-06-03:
### BUG DESCRIPTION
_Having Serial Logging enabled, Sonoff POW R2 crashes by software watchdog on asking for Status 0 in console_
_this will persist even when you disable Serial Logging until you reboot the device. A reboot/crash by watchdog ignores(?) the disabled serial logging_

### REQUESTED INFORMATION

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : Sonoff POW R2
- [X] Tasmota binary firmware version number used : 6.5.0 / pre-compiled
- [X] Development IDE - Compiler / Upload tools used : OTA
- [X] Provide the output of command ``status 0`` : just can provide it with Serial Log disabled.
```
STATUS 0 OUTPUT HERE:
11:19:19 RSL: stat/sonoff/STATUS = {"Status":{"Module":43,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:19:19 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"2019-06-03T09:19:07","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":120,"SaveAddress":"F6000"}}
11:19:19 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.5.0(sonoff)","BuildDateTime":"2019-03-19T12:47:29","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
11:19:19 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HiTheo",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000001"]}}
11:19:19 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
11:19:19 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2870","IPAddress":"192.168.1.164","Gateway":"192.168.1.3","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.3","Mac":"CC:50:E3:54:CB:36","Webserver":2,"WifiConfig":4}}
11:19:19 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_54CB36","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:19:19 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 03 09:19:19 2019","Local":"Mon Jun 03 11:19:19 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:50","Sunset":"21:46"}}
11:19:19 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:19:19 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-06-03T11:19:19","ENERGY":{"TotalStartTime":"2019-05-31T20:11:52","Total":0.696,"Yesterday":0.230,"Today":0.081,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
11:19:19 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-06-03T11:19:19","Uptime":"0T00:00:12","Vcc":3.458,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HiTheo","BSSId":"34:32:C3:84:76:C7","Channel":11,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### TO REPRODUCE
Set Serial log level to > 0.
Ask for `Status 0` at the console.
Software Watchdog crashes the POW.
This is also discribed here #3313 and #3482
_extended version:_
Disable the Serial Logging.
Ask for `Status 0` in console.
Software Watchdog crashes the POW.
Checking the Logging State of Serial Logging in the Web Interface, shows the logging is disabled.
But Status 0 keeps crashing the device.
This will persist until you restart the POW by your own. So is logging still enabled but not viewable in the interface?

### EXPECTED BEHAVIOUR
Disable the ability to enable Serial Logging at POWs or write a comment about it somewhere in the wiki.
Find out why Serial Logging seems to be disabled in the web interface but the POW is still crashing. Is the setting not saved or undefined? Workaround: Reboot the device on changing the log levels?

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-06-03:
As Pow R2 uses the serial interface to communicate with the energy monitoring chip seriallogging is disabled by the driver.

If user enables it chances are it will fail someday.

The wiki can be updated by you so I suggest You'll update it.

tammo0, 2019-06-03:
Thanks. I edited the POW R2 wiki.
What about the strange behaviour showing that logging is disabled but still leads to a crash until you reboot it? Is a reboot by watchdog different then a reboot triggered by the user?

ascillato2, 2019-06-03:
Thanks for adding this information.

closed 2019-06-03 by ascillato2

tammo0, 2019-06-03:
The question remains why a reboot by watchdog seems to be different to a reboot by user and why the problem persist even when serial logging seems to be disabled.

meingraham, 2019-06-03:
I can confirm that a PZEM component connected to an HW-655 running 6.5.0.12/2.5.2 still occasionally WDT's, i.e., the same problematic behavior reported by @tammo0. Now that I have been able to completely eliminate the exception reboots (when running 6.4.1.19/2.4.2), the WDT reboots happen infrequently, but do happen.

Mike