arendst/Tasmota #5909: Inverted PulseTime

KristinaOlson, 2019-06-03:
I have yet to find a way to configure a Sonoff Basic (*via the web interface) such that the switch is normally ON and toggling the switch via the web interface will cause it to shut OFF - then after a period of time expires - turn the switch back ON.  (Essentially a reverse PulseTime function).  PulseTime appears to be hard coded to go from OFF to ON to OFF - only.

I'm using it to power a multi-port USB power adapter supplying power to run a Raspberry PI server farm and want to be able to bounce the farm remotely with a single toggle.  I'm aware of how to use PulseTime with garage door opener applications with Tasmota - but just want the inverse.

Jason2866, 2019-06-03:
It is there. Invert Pulsetime with
`PowerOnState 5` = after a PulseTime period turn relay(s) ON (acts as inverted PulseTime mode)
See wiki section Commands

KristinaOlson, 2019-06-03:
Perfect, I will go back to looking at that.  Since it was a server farm, I wanted it to power up at application of power - so I used a different "PowerOnState" - the one that defaulted the device to ON - not last state.  We can close this one ;)

Jason2866, 2019-06-03:
You have opened you can close :-)

closed 2019-06-03 by ascillato2