arendst/Tasmota #5911: Toggle Switch switches

CleberMartins, 2019-06-04:
### Missing _Toggle Switch_ switches
_Bouncing logic failing to capture switches_

- [ ] Device used (i.e. Sonoff Basic) : Generic (or new added to template)
- [ ] Tasmota binary firmware version number used : 6.4.1 / self-compiled
- [ ] Development IDE - Arduino IDE
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

03:01:12 MQT: stat/LunchRoom/STATUS = {"Status":{"Module":61,"FriendlyName":["LunchRoom","LunchRoom2","LunchRoom3","LunchRoom4"],"Topic":"LunchRoom","ButtonTopic":"0","Power":8,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
03:01:12 MQT: stat/LunchRoom/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"HSH/tasmota","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:24:10","StartupUTC":"2019-06-04T01:37:02","Sleep":50,"BootCount":8,"SaveCount":55,"SaveAddress":"3F7000"}}
03:01:12 MQT: stat/LunchRoom/STATUS2 = {"StatusFWR":{"Version":"6.4.1(sonoff)","BuildDateTime":"2019-06-03T20:17:58","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
03:01:12 MQT: stat/LunchRoom/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HomeSweetHome",""],"TelePeriod":300,"SetOption":["00008009","558180C0","00000000"]}}
03:01:12 MQT: stat/LunchRoom/STATUS4 = {"StatusMEM":{"ProgramSize":558,"Free":2512,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashMode":2,"Features":["00000809","0FFAE794","000383A0","23B617CE","00003BC0"]}}
03:01:12 MQT: stat/LunchRoom/STATUS5 = {"StatusNET":{"Hostname":"LunchRoom-1030","IPAddress":"192.168.1.192","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:3A:E8:07:04:06","Webserver":2,"WifiConfig":4}}
03:01:12 MQT: stat/LunchRoom/STATUS6 = {"StatusMQT":{"MqttHost":"*****","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_070406","MqttUser":"*****","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:01:12 MQT: stat/LunchRoom/STATUS7 = {"StatusTIM":{"UTC":"Tue Jun 04 02:01:12 2019","Local":"Tue Jun 04 03:01:12 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:49","Sunset":"20:47"}}
03:01:12 MQT: stat/LunchRoom/STATUS10 = {"StatusSNS":{"Time":"2019-06-04T03:01:12","Switch1":"ON","Switch2":"ON","Switch3":"ON"}}
03:01:12 MQT: stat/LunchRoom/STATUS11 = {"StatusSTS":{"Time":"2019-06-04T03:01:12","Uptime":"0T00:24:10","Vcc":3.418,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"*****","BSSId":"B4:FB:E4:17:BC:25","Channel":1,"RSSI":100}}}
03:02:00 MQT: tele/LunchRoom/UPTIME = {"Time":"2019-06-04T03:02:00","Uptime":"0T00:24:58"}
03:02:16 MQT: tele/LunchRoom/STATE = {"Time":"2019-06-04T03:02:16","Uptime":"0T00:25:14","Vcc":3.435,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"*****","BSSId":"B4:FB:E4:17:BC:25","Channel":1,"RSSI":100}}
03:02:16 MQT: tele/LunchRoom/SENSOR = {"Time":"2019-06-04T03:02:16","Switch1":"ON","Switch2":"ON","Switch3":"ON"}

### The following logic was working perfectly, before migrating to Tasmota, and works again if I re-flash the device:

void checkSwitches()
{
  for (int i = 0; i < sizeof(swtch) / sizeof(swtch[0]); i++)
  {
    if (swtch[i].value != digitalRead(swtch[i].pin))
    {
      int cnt = 0;
      for (int j = 0; j < 10; j++)
      {
        if (swtch[i].value != digitalRead(swtch[i].pin))
        {
          cnt = cnt + 1;
        }
        delay(5);
      }
      if (cnt == 10)
      {
        swtch[i].value  = digitalRead(swtch[i].pin);
        myESP.publish(SwitchesTopic, swtch[i].name, false);
//        client.print((String)swtch[i].name); client.println(" switched!");

        for (int j = 0; j < sizeof(link) / sizeof(link[0]); j++)
        {
          if (strcmp(link[j].swtch, swtch[i].name) == 0)
          {
            for (int k = 0; k < sizeof(relay) / sizeof(relay[0]); k++)
            {
              if (strcmp(link[j].relay, relay[k].name) == 0)
              {
                digitalWrite(relay[k].pin, !digitalRead(relay[k].pin));
                relay[k].value = digitalRead(relay[k].pin);
                EEPROM.put(0, relay);
                delay(10);
                EEPROM.commit();
                relay[k].report = 1;
              }
            }
          }
        }

//        client.println("Writing to EEPROM");
        if (debugOn) {Serial.println((String)swtch[i].name); Serial.println(" switched!");}
        EEPROM.put(0, relay);
        delay(10);
        EEPROM.commit();
      }
    }
  }
}
`````
Troubleshooting:
- I've tried all the scale on the bouncing configuration, no difference;
- Same behavior on any Switch, I have 3
- Very rarely I can get the relay to "blink" if I keep playing with the toggle switches, not able to consistently repeat it
- Other than the physical switches, everything else works perfectly; mqtt & webui
- This same Tasmota bin file and configuration get Switches to work in another equal device (LinkNode R4), the difference is the environment, the problematic one is installed near electricity (not possible to move) and the others have less interference
- I'm not experienced enough to understand the way Tasmota is handling the bouncing, but it seems to me that it doesn't reach the threshold yet it records the inverted status without executing the action because of the heavy bouncing; my previous logic (amaturely written), is re-seting the count while the bouncing is happening, so only inverts the status once it finds it to be different than previously recorded 10 successful times in the sequence within 5 ms of each other (I believe :)).

Thanks a lot for your support, except for this one device I have everything on Tasmota now and I'm loging it :)

-Cleber.

wongnam, 2019-06-04:
your FW is too old("Version":"6.4.1(sonoff)"). Try the latest FW.

CleberMartins, 2019-06-04:
:O just did, with the pre-compiled 6.5.0 bin available as the latest one... The device is now dead, relay2 (GPIO14) blinking forever (yes, my luchroom like a disco), not sure if that is the only relay, no more remove access... I'll need to open the ceiling and get my hands on it... just tested in another device I had in hands, very same behavior when I upgrade the FW to 6.5.0... Wish me luck now :)

Jason2866, 2019-06-04:
See wiki commands. For switch debounce the command is `SwitchDebounce` 
User control over switch debounce timing 40..1000 = set switch debounce time in ms. Default is 40

Please close since your question is answered and the needed feature is already implemented 
Thx

ascillato2, 2019-06-04:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-04 by ascillato2

CleberMartins, 2019-06-04:
Thanks. I've found the issue, nothing related to configuration, it was a hardware issue.
Yet, upgrading the FW to 6.5.0 pre-built from 6.4.1 killed two of my devices, easily repeatable. I'll do some further tests and get another case open if needed. Really thanks, -Cleber.