arendst/Tasmota #5914: [Help] access tasmota with iPhone

Niggolo, 2019-06-04:
Hello,
today i just flashed a S20 smart plug but i just figured out the ewelink app is not working anymore so i start trying out how to turn on/off the device without this app, after few hours on wasting time trying to undestand this MQTT i just realize it need an external device for making a MQTT server ?!
Im new to tasmona so, there are something i can do for an easy way to connect from ios/web to the tasmona web ui or anything else so i can turn it on/off?
thank you
Sorry for my english!

joba-1, 2019-06-04:
You are on the right track. Use safari web browser. Read this: https://github.com/arendst/Sonoff-Tasmota/wiki/initial-configuration
Access over internet is possible (vpn or home automation software) but advanced stuff. Start within your home wlan.

meingraham, 2019-06-04:
Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-04 by ascillato2