arendst/Tasmota #5915: ButtonDeBounce disable function for Touch Switches

sheppy99, 2019-06-04:
The minimum time I can set ButtonDeBounce to is 40ms, can this be changed to zero for capacitive switches such as the SOnOff Touch which presumably don't bounce by design? Installing ESPEasy on a Touch makes it more responsive than Tasmota so I'm guessing the DeBounce setting is the reason. With ESPEasy the relay clicks instantly making the switch appear like a click of a standard light switch which is more intuitive.

ascillato, 2019-06-05:
Sorry, the delay is due to multipress feature. Just disable multipress to instant reaction. See setoption commands in the wiki. Thanks.

ascillato2, 2019-06-05:
Closing this issue as it has been answered. If you need further assistant, please find us in the Tasmota support chat. Thanks. 

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-05 by ascillato2

joba-1, 2019-06-05:
try SetOption13