arendst/Tasmota #5916: 2P WiFi Smart Circuit Breaker with Energy monitoring and meter function

maumauns, 2019-06-05:
Hi all, searching Aliexpress i found this circuit breaker. As u can see, it use ewelink to monitoring and control the circuit breaker.
Does anyone know if i can install tasmota on this??

[image]

Jason2866, 2019-06-05:
Short answer, NO.  There are so many devices....

Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

blakadder, 2019-06-05:
There is a high probability you can install Tasmota on it. The question is whether it will work with it or not.

closed 2019-06-05 by blakadder

arihantdaga, 2019-10-14:
Hi @maumauns did you find any succees ? I want to do the same, i couldnt understand its circuit and how to control the motor to on and off etc. Would be glad if you could share some of your findings.

OfferLa, 2019-10-18:
I'm also very interested in controlling circuit breaker device with Tasmota.
The main question is if the motor is controlled by a relay that's connected to one of the GPIO pins of a 8266 chip or similar.
If so - Tasmota will be able to control it as a charm.
If you actually own such a device and can confirm the presence of 8266 chip and a GPIO connection to the motor - it will be a big step towards a solution.

arihantdaga, 2019-10-20:
@OfferLa I bought one earlier, but it had only one gpio to rotate the motor in both directions, also i couldn't figure out, how it detects the current state of the circuit breaker (On or Off), it seems like there is no other gpio then gpio12(going to an ic, with no name on it. That looks like a child controller) and 13(going to led) connected anywhere.

I ordered one more from a different brand. Will receive it soon, then i'll share more details.

OfferLa, 2019-10-20:
@arihantdaga Thanks for that information.
I'm looking now on this device:
https://www.aliexpress.com/item/32979662165.html

Let's hope a compatible device will be found at last, enabling smart-home automation much easier and standard.

Looking forward to your updates!

ShafiqUrRehman, 2019-10-26:
@arihantdaga I confirm there is a child controller that takes care of moter function via gpio 12 and gpio 13 is connected to purple LED.

I have successfully install esphome on it and controlling it via home assistant.

I am struggling to find how esp8285 will get to knows that breaker has been turned on or off manually. In the factory firmware, there is such feed back.

Please share if you get to know the feedback GPIO.

Thank,

Thanks

arihantdaga, 2019-11-07:
@ShafiqUrRehman Thats great. I am sorry for replying late. I have few doubts here. As i could see, that GPIO0 is not connected to anything. How does the esp get current status of the Circuit breaker. From my findings i saw that there are few hall effect sensors, kept on key positions, to detect current status of the Circuit breaker. But all these hall effect sensors are connected to the child microcontroller only. 

What exact;y you meant by `Gpio0 start toggling` ? (Like Blinking ?)
Can you share your code which is working ?

ShafiqUrRehman, 2019-11-07:
Try following, it is working for me.

esphome:
  name: 40yga_01
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "myssid"
  password: *****

logger:
api:
ota:

status_led:
  pin: GPIO13
  
switch:
  - platform: gpio
    name: Switch
    restore_mode: ALWAYS_OFF
    id: relay
    pin:
      number: GPIO12

binary_sensor:
  - platform: gpio
    pin: GPIO0
    name: "GPIO0"
    on_multi_click:
    - timing:
        - ON for at most 2s
        - OFF for at most 2s
      then:
        - switch.toggle: relay

maumauns, 2019-11-08:
Ow very good this post is alive yet. In my case I only want the option to monitor the energy, not toggle the relay.

@arihantdaga Did you bought the same as i mentioned??

arihantdaga, 2019-11-09:
@maumauns noh, the one i bought is not exactly the same.. But i think underlying firware should br almost similar in all of these circuit breakers..

mbt45, 2020-03-17:
@arihantdaga Do you have photos of the esp soldered on the board? it would help to determine what does what.

arihantdaga, 2020-03-18:
@mbt45  I am sorry, currently i dont have it.

iliyang, 2020-03-31:
Very interesting thread! Can anyone provide a photo or link to a breaker that has been confirmed to work with Tasmota (or another firmware)?

(FWIW, I'm also only interested in the power metering function.)

mbt45, 2020-03-31:
@iliyang 
I bought 2 different types

1P and 2P
[image]
[image]

The 1P was flashed it on Tasmota with tuya-convert no problem and then switched to ESPHOME. 
I don't think Tasmota has any config of these. 
I had to search each pins one buy one to find the right pins. I actually didn't look for the power consumption pin probably the ADC pin but I'll have a look later.

The 2P is still not flashed because it has the new Tuya firmware and Tuya-convert cannot flash it yet. 
Since these are riveted for security purpose I didn't crack-it open to access the chip. I'll have to wait for the tuya-convert update.
( https://github.com/ct-Open-Source/tuya-convert/issues/483)  

This is my code for the 1P 

`esphome:
  name: disjoncteur_1p
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: "my network"
    password: "my password"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    name: Switch
    restore_mode: ALWAYS_OFF
    id: relay
    pin: GPIO12
    on_turn_on:
      then:
      - switch.turn_on: red_led
    on_turn_off:
      then:
      - switch.turn_off: red_led
      
  - platform: gpio
    name: Switch Red LED
    id: red_led
    internal: true
    pin:
      number: 4
      inverted: yes

  - platform: restart
    name: "Disjoncteur_1P Restart"
    
    
status_led:
    pin:
      number: 5
      inverted: yes
      

binary_sensor:
  - platform: gpio
    pin: GPIO13
    internal: true
    name: "quick/comp"
    on_multi_click:
      - timing:
        - OFF for at most 2s
        - ON for at most 2s
        then:
        - switch.toggle: relay`

iliyang, 2020-03-31:
Thanks a lot for the prompt reply @mbt45, very informative!!

mbt45, 2020-03-31:
@iliyang no problem.

foxel, 2020-04-03:
Did anybody crack this open? What does the switch under the green tape do?

mbt45, 2020-04-07:
@foxel 
I didn't crack one open, but that switch will just put the breaker on and off multiple times. As stated in the manual it is used for testing. Once flashed, this switch still does the same pattern. It is likely this is not linked to the ESP since I did not see and changes on the pins will put in input

kevinblo, 2020-06-25:
> EDITED:
> 
> @arihantdaga I confirm there is a child controller that takes care of motor function via gpio 12 and gpio 13 is connected to purple LED.
> 
> If you have turned the breaker on by turning gpio 12 high, and manually turned off the circuit breaker, then gpio 0 will start toggle. Similarly, if you turn the breaker off by turning gpio 12 low, and manually turned on the circuit breaker, then gpio 0 will start toggling.
> 
> I have successfully install esphome on it and controlling it via home assistant. (I yet have to include feedback in my esphome logic)
> 
> Thanks

How you can flash firmware?

jd1900, 2020-09-11:
> How you can flash firmware?

Did you manage to do it?

ramponis, 2024-08-26:
> @iliyang I bought 2 different types
> 
> 1P and 2P [image] [image]
> 
> The 1P was flashed it on Tasmota with tuya-convert no problem and then switched to ESPHOME. I don't think Tasmota has any config of these. I had to search each pins one buy one to find the right pins. I actually didn't look for the power consumption pin probably the ADC pin but I'll have a look later.
> 
> The 2P is still not flashed because it has the new Tuya firmware and Tuya-convert cannot flash it yet. Since these are riveted for security purpose I didn't crack-it open to access the chip. I'll have to wait for the tuya-convert update. ( [ct-Open-Source/tuya-convert#483](https://github.com/ct-Open-Source/tuya-convert/issues/483))
> 
> This is my code for the 1P `esphome: name: disjoncteur_1p platform: ESP8266 board: esp01_1m
> 
> wifi: networks:
> 
> * ssid: "my network"
>   password: "my password"
> 
> captive_portal:
> 
> logger:
> 
> api:
> 
> ota:
> 
> switch:
> 
> * platform: gpio
>   name: Switch
>   restore_mode: ALWAYS_OFF
>   id: relay
>   pin: GPIO12
>   on_turn_on:
>   then:
>   
>   * switch.turn_on: red_led
>     on_turn_off:
>     then:
>   * switch.turn_off: red_led
> * platform: gpio
>   name: Switch Red LED
>   id: red_led
>   internal: true
>   pin:
>   number: 4
>   inverted: yes
> * platform: restart
>   name: "Disjoncteur_1P Restart"
> 
> status_led: pin: number: 5 inverted: yes
> 
> binary_sensor:
> 
> * platform: gpio
>   pin: GPIO13
>   internal: true
>   name: "quick/comp"
>   on_multi_click:
>   
>   * timing:
>     
>     * OFF for at most 2s
>     * ON for at most 2s
>       then:
>     * switch.toggle: relay`

mbt45, I have the same 1P device, can you tell me how did you put it into EZ pairing mode to flash it with tuya-convert?