arendst/Tasmota #5917: T1 2 & 3CH - Configuring a syslog server @ log level 4 reverts the switch to a sonoff basic

kyemacdonald, 2019-06-05:
### BUG DESCRIPTION

A Sonoff T1 2ch or 3ch US (not tested on 1ch) switch that is configured with a syslog server reporting log level 4 with reset at some point to a sonoff basic module.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ x] Device used (i.e. Sonoff Basic) : Sonoff T1 2ch & Sonoff T1 3ch US
- [x ] Tasmota binary firmware version number used : 6.5.0.14(d0502c1-sonoff) / (pre-compiled )
- [x ] Development IDE - Compiler / Upload tools used : Webgui direct upload
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE: **note this is after reconfiguring the switch WITHOUT the syslog server running.  
03:10:21 CMD: status 0
03:10:21 SRC: WebConsole from 10.1.1.45
03:10:21 RSL: Received Topic /status, Data Size 1, Data 0
03:10:21 RSL: Group 0, Index 1, Command STATUS, Data 0
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS = {"Status":{"Module":30,"FriendlyName":["Sonoff Master Bedroom 1","Sonoff Master Bedroom 2","Sonoff Stairwell Light Upper"],"Topic":"sonoffmasterbedroom","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:22","StartupUTC":"2019-06-05T02:03:59","Sleep":50,"CfgHolder":4617,"BootCount":272,"SaveCount":544,"SaveAddress":"F7000"}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0.14(d0502c1-sonoff)","BuildDateTime":"2019-06-02T19:01:37","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skyenet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA029","280500000100060000000000000000000000","00000200"]}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE394","001783A0","23B617CC","01003BC0"]}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoffmasterbedroom-6947","IPAddress":"10.1.1.52","Gateway":"10.1.1.254","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.254","Mac":"C4:4F:33:9E:DB:23","Webserver":2,"WifiConfig":4}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.99","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9EDB23","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 05 02:10:21 2019","Local":"Wed Jun 05 03:10:21 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:48","Sunset":"20:48"}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS10 = {"StatusSNS":{"Time":"2019-06-05T03:10:21"}}
03:10:21 MQT: sonoffmasterbedroom/stat/STATUS11 = {"StatusSTS":{"Time":"2019-06-05T03:10:21","Uptime":"0T00:06:22","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"Skyenet","BSSId":"1C:F2:9A:CA:E3:DE","Channel":6,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:04"}}}
03:10:23 WIF: Checking connection...
03:10:23 WIF: Connected

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE

Provide remote syslog server and set syslog level to 4.
Switch will reset to a sonoff basic at some point in the next hour.

My syslog server is rsyslog running on a ubuntu server on the local network.

### EXPECTED BEHAVIOUR
For it to stay as the correct switch type

### SCREENSHOTS

### ADDITIONAL CONTEXT

[ESP-APP.log](https://github.com/arendst/Sonoff-Tasmota/files/3255242/ESP-APP.log)
[ESP-CFG.log](https://github.com/arendst/Sonoff-Tasmota/files/3255243/ESP-CFG.log)
[ESP-HTP.log](https://github.com/arendst/Sonoff-Tasmota/files/3255244/ESP-HTP.log)
[ESP-LOG.log](https://github.com/arendst/Sonoff-Tasmota/files/3255245/ESP-LOG.log)
[ESP-MOD.log](https://github.com/arendst/Sonoff-Tasmota/files/3255246/ESP-MOD.log)
[ESP-RSL.log](https://github.com/arendst/Sonoff-Tasmota/files/3255247/ESP-RSL.log)
[ESP-SRC.log](https://github.com/arendst/Sonoff-Tasmota/files/3255248/ESP-SRC.log)
[ESP-WIF.log](https://github.com/arendst/Sonoff-Tasmota/files/3255249/ESP-WIF.log)

Sonoff T1 3CH Module
Sonoff Master Bedroom 1
Program Version	6.5.0.14(d0502c1-sonoff)
Build Date & Time	2019-06-02T19:01:37
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	0T00:03:23
Flash write Count	231 at 0xF8000
Boot Count	20
Restart Reason	Software/System restart
Friendly Name 1	Sonoff Master Bedroom 1
Friendly Name 2	Sonoff Master Bedroom 2
Friendly Name 3	Sonoff Stairwell Light Upper
 
AP1 SSId (RSSI)	Skyenet (100%)
Hostname	sonoffmasterbedroom-6947
IP Address	10.1.1.52
Gateway	10.1.1.254
Subnet Mask	255.255.255.0
DNS Server	10.1.1.254
MAC Address	C4:4F:33:9E:DB:23
 
MQTT Host	
MQTT Port	1883
MQTT User	mqtt
MQTT Client	DVES_9EDB23
MQTT Topic	sonoffmasterbedroom
MQTT Group Topic	sonoffs
MQTT Full Topic	sonoffmasterbedroom/cmnd/
MQTT Fallback Topic	cmnd/DVES_9EDB23_fb/
 
Emulation	None
mDNS Discovery	Disabled
 
ESP Chip Id	10410787
Flash Chip Id	0x144051
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	524kB
Free Program Space	476kB
Free Memory	14kB

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-06-05:
Nothing strange in the logs other than (too) many normal restarts. No indications of exceptions.

Considering your experience with ghost switching I suspect a button press has been detected over a longer amount of time resulting in a reset to default module, being Sonoff Basic.

If you do not use multipress you might want to consider using commands SetOption1 (disable inadvertant reset due to long press) and SetOption13 (Disable multipress detection and speed up button recognition).

kyemacdonald, 2019-06-05:
Setoption1 and setoption13 are set on all my switches.

Also the change to sonoff basic will happen in less than 30 minutes and on all the switches.  My ghost switching happens once every 6-10 hours with some switches being way worse than others.  

I will do it now to one switch and refresh it till it changes.

arendst, 2019-06-05:
You might also want to change sleep to 0 as this will also influence buttondebounce time when set below sleep time.

kyemacdonald, 2019-06-05:
Thanks.

At this point I'm wondering is buttondebounce 400 is enough to fix all my problems. I've set one switch to use a remote syslog over an hour ago and it's not reset yet. Previously this would have happened easily by now.

It's coming up on midnight here, if I get through the night with no ghost switching that would be the first time.

kyemacdonald, 2019-06-05:
Ok I'm closing this issue.  Sorry for taking up your time.  Switch is still outputting to log file and has not reset.  Also I've had no ghost switches on any switch for the past 12 hours.

closed 2019-06-05 by kyemacdonald