arendst/Tasmota #5918: Support for Sonoff TX Series

stevengohss, 2019-06-05:
iTead had recently launched a new touch switch ([Sonoff TX Series](https://www.itead.cc/sonoff-tx-series-wifi-smart-wall-switches.html))

Not sure if Tasmota supports this (using existing Sonoff T1 XCH) or a new module type needs to be created.

Will there be any advise?

Jason2866, 2019-06-05:
We dont have this device. It needs someone who does the research.
If you do, your findings are highly welcome.

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

arendst, 2019-06-05:
I have some in the mail but it is verrrrry slow.

Once received I'll check if it can be implemented.

stevengohss, 2019-06-05:
Thanks so much and appreciate if you could post once checked so I can place my order too! The TX series looks great.

dandowftm, 2019-06-11:
Hi, just received a number of TX UK models to add to my home when i get it rewired.
Struggled on for a bit, but found some help from a UK automation forum on Facebook.
Anyway, I've managed to flash tasmota to 1 and 2 channel devices and i suspect the 3 channel will be the same, just havent got round to it yet.
The board is different to any I've found when searching for information so I found a pin out and located the GPIO0 pin on the chip. I bridged this with ground whilst plugging in the FTDI module and got it into boot mode. From here I was able to flash using Atom.
[image]
[image]

alex-sever-h, 2019-06-15:
Hi,
I just got sonoff TX devices for all my apartment, so I'm interested in making them work reliably.
Thanks @dandowftm for finding GPIO0. In the meanwhile, I've found an easier location (R19) to get to it on the PCB. Just measured it, because I don't have too much time right now to test it out.
[image]

stevengohss, 2019-06-23:
Thanks @alex-server-h and @dandowftm for the great work and help on the new TX series. Can I ask if the flashing process is the same as the T1 after bridging GPIO0 or R19 to ground?

dandowftm, 2019-06-23:
Yes flashes the same way as any other sonoff device. Personally I always use Atom

stevengohss, 2019-06-23:
Thanks @dandowftm, shall order mine and try

shafzhr, 2019-07-02:
Is it possible to be done without soldering or it's too tiny?

timbru31, 2019-07-05:
An interesting new feature seems the controllable LED light - could this be Tasmota controllable in the future? (IMHO the LED is way to bright in the night)

meingraham, 2019-07-05:
@timbru31 A combination of a "fake" LedLink to be able to have direct control of the LED GPIO and configure them as PWM? That's as far as I'm going to take it since I don't have one of these to test with. But now I've assigned the science experiment ;-)

arendst, 2019-07-15:
Received my Tx Eu 3 gang and flashed it with Tasmota. Selected T1 3CH and off I went. No issues what so ever. It perfoms exactly like the T1 EU's I had so I think we can say Tasmota supports TX too.

Regarding flashing I used the testpoint (TP2 GPIO0) on the back of the touch board (That's what testpoints are for) and soldered a small wire to it. Flashing is done with the board disconnected from the base anyway.

[image]

ascillato2, 2019-07-15:
Closing this issue as support for this device has been added to Tasmota by Theo. Thanks

closed 2019-07-15 by ascillato2

zomfg, 2019-08-16:
quick question: are your TXs as quick as T1? mine has like half second delay after button press, and since I flashed right away I'm not sure if it's the same with stock firmware :( my T1s are almost instant though

shafzhr, 2019-08-16:
@zomfg Mine response instantly.

ascillato, 2019-08-16:
@zomfg 

See commands: Setoption1 and setoption13 for inmediate response.

zomfg, 2019-08-16:
SetOption13 was it, thank you very much.

mithoog, 2019-08-23:
@dannygohks Connect TP2 with Ground on boot to enter into Flash mood.

dannygohks, 2019-08-23:
> Received my Tx Eu 3 gang and flashed it with Tasmota. Selected T1 3CH and off I went. No issues what so ever. It perfoms exactly like the T1 EU's I had so I think we can say Tasmota supports TX [...]

can you show the wiring using TP2 to flash the device?

dannygohks, 2019-08-24:
My T2UK3C TP2 dont have GPIO0 in the board. How can i boot into flash mode?

[image]
[image]

meingraham, 2019-08-24:
The Sonoff 4Ch Pro R1 has a similar issue. The trick is to have a lead connected to ground with a very small pin (e.g., a needle) on the other end. You then hold the pin onto GPIO0 of the ESP chip itself and then apply power to the board. Or you could ohm out the traces to see if GPIO0 is broken out to somewhere on the PCB (e.g., to a pull-up resistor).

GaseGasovic, 2019-08-29:
Guys, what do u use for flashing, source code or .bin file, because, I cannot flash it with .bin file dunno why.... And another question, does TX 2 channels uses ESP8266 or ESP8265? Thanks in advance

Berkinka, 2019-12-26:
Hello, I bought Sonoff TX T2 EU (2 gang and 3 gang) version. Yesterday they came and i just tried flashing by using this link https://github.com/arendst/Tasmota/wiki/Sonoff-TX-Series but i saw that the person who did this did not connected ground to serial connector. He just bridged ground and GPIO0 pins. When i did that, did not get any response from device. (I'm using CH340G) Then I tried by connecting ground to serial connector, and then using the other ground on the PCB and bridge it with GPIO0. This time I got 3 blue lights always on. I installed drivers and Tasmotizer able to see my device but when i try to flash unfortunately it can't connect ESP and throw an error. 

I tried on Windows and Mac, same results. Then i accidentally connected TX to TX and RX to RX. This time, Tasmotizer able to connect ESP, erase and write flash successfully. Program said power cycle and did. But when i gave power, all 3 blue lights was just blinking and hardware plugged in/plugged out tone was repeating itself. 

So I cant able to flash this thing. I don't know what i did wrong. How can i solve this? Thanks