arendst/Tasmota #5919: NodeMCU with MHZ19B fatal exception after command sensor15 0

sfromis, 2019-06-05:
Persistent fatal exception reboots issuing command "sensor15 0"

Running pre-compiled Tasmota 6.5.0 (sonoff.bin) on a Geekcreit NodeMCU v4 with a MHZ19B CO2 sensor attached over the serial interface, I experienced bad calibration after getting well above 1000 ppm, not going below around 600 ppm even after hours in draft near an open bedroom window, rural area without nearby CO2 sources. Attempting to disable the automatic baseline calibration using "status15 0" persistently results in rebooting with reason fatal exception. Same scenario with "sensor15 1" or just "sensor15" commands. No crash with other variants of "sensor15 xx".

{"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x4026b921 depc:0x00000000"}

Module configured as Generic(18) and:
TX GPIO1 Serial Out: MHZ Tx (60)
RX GPIO3 Serial In: MHZ Rx (61)
D2 GPIO4: I2C SDA (6)
D1 GPIO5: I2C SCL (5)
Serial Tx/Rx appropriately cross-connected to Rx/Tx on MHZ19B.

A quick look in xsns_15_mhz19.ino MhzCommandSensor seems consistent with a speculation of the code having trouble accessing the flag "Settings.SensorBits1.mhz19b_abc_disable", but I'm not able to do real debugging.

Searching revealed no similar issues (fatal exception with mhz19/mhz19b). I've found no way *not* to reproduce the problem. Not having the I2C connected BME280 did not change things. I've seen no other anomalies running the sensor for a few weeks, with lots of readings, except for occasional (not repeatable) missing time (1970-01-01) on first MQTT message after boot, like after testing now:
nodemcu2/tele/STATE {"Time":"1970-01-01T00:00:15","Uptime":"0T00:00:15","Vcc":2.726,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":11,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:05"}}

```
10:41:12 CMD: status 0
10:41:12 SRC: WebConsole from 192.168.86.33
10:41:12 RSL: Received Topic /status, Data Size 1, Data 0
10:41:12 RSL: Group 0, Index 1, Command STATUS, Data 0
10:41:12 MQT: nodemcu2/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["nodemcu2"],"Topic":"nodemcu2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:41:13 MQT: nodemcu2/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"nodemcu","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:00:27","StartupUTC":"2019-06-05T09:40:45","Sleep":50,"CfgHolder":4617,"BootCount":52,"SaveCount":63,"SaveAddress":"F5000"}}
10:41:13 MQT: nodemcu2/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:41:13 MQT: nodemcu2/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"Resolution":"598180C0","SetOption":["00008009","280500000100000000000000000000000000","00000000"]}}
10:41:13 MQT: nodemcu2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
10:41:13 MQT: nodemcu2/stat/STATUS5 = {"StatusNET":{"Hostname":"nodemcu2","IPAddress":"192.168.86.152","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.86.1","Mac":"xxxx","Webserver":2,"WifiConfig":4}}
10:41:13 MQT: nodemcu2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.86.80","MqttPort":1883,"MqttClientMask":"nodemcu2c","MqttClient":"nodemcu2c","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:41:13 MQT: nodemcu2/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 05 09:41:13 2019","Local":"Wed Jun 05 10:41:13 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:48","Sunset":"20:48"}}
10:41:13 MQT: nodemcu2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-06-05T10:41:13","BME280":{"Temperature":26.8,"Humidity":51.8,"Pressure":1010.60,"SeaPressure":1012.29},"MHZ19B":{"Model":"B","CarbonDioxide":471,"Temperature":28.0},"PressureUnit":"hPa","TempUnit":"C"}}
10:41:13 MQT: nodemcu2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-06-05T10:41:13","Uptime":"0T00:00:28","Vcc":2.726,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":6,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```

```
00:00:00 CFG: Loaded from flash at F6, Count 62
00:00:00 SRC: Restart
00:00:00 Project sonoff nodemcu2 Version 6.5.0(release-sonoff)-2_3_0
00:00:00 SHT: Sensor did not ACK command
00:00:00 I2C: BME280 found at 0x76
00:00:00 SNS: Hardware Serial
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 xxxx in mode 11N as nodemcu2...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 DMP:
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 DMP:
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
00:00:05 DMP:
00:00:06 WIF: Checking connection...
00:00:06 WIF: Connected
00:00:06 HTP: Web server active on nodemcu2 with IP address 192.168.86.152
00:00:07 DMP:
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: nodemcu2/tele/LWT = Online (retained)
00:00:07 MQT: nodemcu2/cmnd/POWER = 
00:00:07 MQT: Subscribe to nodemcu2/cmnd/#
00:00:07 MQT: Subscribe to nodemcu/cmnd/#
00:00:07 MQT: Subscribe to cmnd/nodemcu2c_fb/#
00:00:07 MQT: nodemcu2/tele/INFO1 = {"Module":"Generic","Version":"6.5.0(release-sonoff)","FallbackTopic":"cmnd/nodemcu2c_fb/","GroupTopic":"nodemcu"}
00:00:07 MQT: nodemcu2/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"nodemcu2","IPAddress":"192.168.86.152"}
00:00:07 MQT: nodemcu2/tele/INFO3 = {"RestartReason":"Fatal exception:3 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x4026b921 depc:0x00000000"}
00:00:08 DMP:
00:00:10 APP: Boot Count 52
00:00:10 APP: (UTC) Wed Jun 05 09:40:55 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
10:40:56 CFG: Saved to flash at F5, Count 63, Bytes 3584
10:40:56 DMP: FF 86 01 90 52 00 00 00 97
10:41:00 MQT: nodemcu2/tele/STATE = {"Time":"2019-06-05T10:41:00","Uptime":"0T00:00:15","Vcc":2.739,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"xxxx","BSSId":"xxxx","Channel":6,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:05"}}
10:41:01 MQT: nodemcu2/tele/SENSOR = {"Time":"2019-06-05T10:41:00","BME280":{"Temperature":26.8,"Humidity":51.7,"Pressure":1010.66,"SeaPressure":1012.35},"MHZ19B":{"Model":"B","CarbonDioxide":400,"Temperature":42.0},"PressureUnit":"hPa","TempUnit":"C"}
10:41:04 DMP: FF 86 02 05 47 00 00 00 2C
10:41:11 WIF: Checking connection...
10:41:11 WIF: Connected
10:41:12 DMP: FF 86 01 E4 44 00 00 00 51

```

ascillato, 2019-06-05:
Try erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py.

Also, use the precompiled bins with core 2.5.2

arendst, 2019-06-05:
Fixed in dev release. 

Duplicate issue. Can't find the original one. Caused by unallowed flash access.

sfromis, 2019-06-05:
Thanks, will await update.

arendst, 2019-06-05:
No need to wait. Just download the dev release from the link in the README.md file

sfromis, 2019-06-05:
Ok, flashed http://thehackbox.org/tasmota/020502/sonoff-4m.bin and reconfigured
Result: {"Sensor15":"ABC is Disabled"}
:-)

closed 2019-06-05 by sfromis