arendst/Tasmota #5920: Wemos D1 mini turn off LCD 2004 with momentary button, how?

Giro-S, 2019-06-05:
Hi to all and a BIG thank to Arends for his firmware!
I'm trouble with a button, I would like to turn off (or on) my LCD 2004 but I don't know how :(
The button is on D6 with pull_down resistore.
I've configured LCD with D1 and D2 (scl, sda) ...it works fine, after I've added a momentary button (button1) for first on D3 and after on D6; in all cases the button works! 
I've added a PZME004T for energy monitor on TX and RX....it works well but when I press the button I have 2 cases:
1- nothig happen
2- wemos reset itself and lose configuration

How to solve it? With a rule? 
I've seen on main page that there is a switch, if I click on it, I can turn on/off the LCD, can I porting this "internal switch" to external momentary button?

Thank you very much

meingraham, 2019-06-05:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Giro-S, 2019-06-05:
Ok, thank you.

closed 2019-06-05 by ascillato2