arendst/Tasmota #5921: Flashing Sonoff Basic R3

patjenl, 2019-06-05:
I just received two Sonoff Basic R3 modules, but I could not find any info on the pinout for flashing.
As I just successfully flashed both modules with the latest Tasmota version, I thought I would share the pinout I used.
The new R3 modules seem to behave as expected with the normal Sonoff Basic module configuration.

[image]

lalo-uy, 2019-06-05:
May you provide more pictures of the boards? Both sides.
The upper board is removable or soldered?
Thanks in advance

patjenl, 2019-06-05:
The upper board is soldered onto the bottom board.

I soldered a print header upside down into the four PCB holes and removed the plastic spreader afterwards. Then used normal dupont wires to attach the programmer.

[image]
[image]
[image]

lalo-uy, 2019-06-05:
Thanks.
It is labeled OTA_SW where you hook up GND.
May be this one can be flashed via OTA  with this jumper on place.

joba-1, 2019-06-05:
thanks for sharing.

Stange: the pcb design looks more messy and more complicated. Wonder what the improvements are :)

thxthx0, 2019-06-05:
Jumper for DIY Mode ;)

Sonoff REST-API
"Tasmotizing Sonoff days may be nearing end" 😕 🤣 

https://github.com/itead/Sonoff_Devices_DIY_Tools

use earbuds:
https://www.youtube.com/watch?v=fRS-ukCgD_I

fhassine, 2019-06-05:
Is GPIO14 accessible on this new R3 board? You have set the pins on VCC, RX, TX, and GND, but , should I look for GPIO14, do you see where we can access it?

arendst, 2019-06-06:
So using the above links find a way to OTA upload tasmota to it :-)

thxthx0, 2019-06-06:
@patjenl: 
I'm pretty sure that the marked pin 'GPIO14' is **GND** too, 
maybe you could check...

patjenl, 2019-06-06:
> @patjenl:
> I'm pretty sure that the marked pin 'GPIO14' is **GND** too,
> maybe you could check...

You are right ... I changed the first picture.
When measuring I must have falsely measured from the ESP8285 bottom-right corner pad (which is gnd) instead of the bottom data pins. Thank you.

lalo-uy, 2019-06-06:
Since there is the place of external eprom I wonder if the signals of the SPI bus may be used as IO?

arendst, 2019-06-06:
@lalo-uy good point. 

As the ESP8285 uses DOUT SPI flash comms, GPIO9 (pin 18) and GPIO10 (pin 19) are available for user connections.

ascillato2, 2019-06-07:
@patjenl 

Thanks for sharing all this info, very appreciated :+1:

Can you add this to the wiki?

Thanks

closed 2019-06-07 by ascillato2

georgia088, 2019-06-10:
Has anyone figured out how to switch between one of the gpios to gnd to use an existing wall switch to also be able to control device? 
Like the old basics, you could flash  tasmota set gpio 3 to switch 1 in settings and solder wires from gnd and rx on the board. These wires would also switch the sonoff on/off. 

Does the r3 allow this? Maybe make it simpler?

Thanks!

thxthx0, 2019-06-10:
Yes, RX and TX work in the same way here.
You can also use GPIO9 and 10, which are a better choice.

**Maybe someone can tell us, which GPIO is connected to J1 (OTA_SW)**
GND and ??? 
Could be an easy connection (including pullup resistor).

lalo-uy, 2019-06-10:
GPIO16

georgia088, 2019-06-10:
Just curious why gpio 9-10 are better choices? Also, what to you mean ota-sw? Is that some how to flash tasmota ota? I’m just trying to set it so two pins can connect together to switch the sonoff.

thxthx0, 2019-06-11:
RX is ok, if you don't need serial logging...
TX is a bad choice for switch, ESP won't boot if set to GND.

Jumper J1 is labeled OTA_SW, see post and links about DIY Mode above.

You might give GPIO16 @ J1 a try.

dkossman, 2019-06-22:
it would be really helpful if you could post the dimensions (LxWxH) of the R3 PCB.  i checked the specs and the R3 plastic enclosure is a bit bigger than the R2, which means it won't fit into the IP66 water resistant case that Itead sells for the R2.  However i need to figure out if the R3 PCB will fit into this, or if it will fit into a standard receptacle box.

amaze11, 2019-06-24:
Any chanche to recover this R3 firmware and flashing it on other devices (say D1 mini or other ESP8266 boards)? Or are there any hardware limitations?

auriux, 2019-07-15:
Hi, could someone help me.. I messed up. Added the jumper to activate DIY mode and used DIY Tool to flash tasmota minimal.bin, tried normal version of tasmota, but it say the filesize was too big. After the flash complete my R3 is not responding, no lights flashing, nothing. No sign that it works. Did I made a big mistake? Or I just could order serial to USB adapter and try to flash it in the old way?

easyyu, 2019-07-24:
Now I'm confused a bit how to flash Sonoff basic r3. From the page: [https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic](url) there is a message:
`This version of the device supports the new Itead DIY architecture which allows OTA firmware upload. The DIY OTA procedure won't work with Tasmota. Tasmota expects execution from address 0. DIY OTA expects espressif images starting at either 1000 or 8100. You must flash Tasmota on this this device using the serial interface. The device was reviewed by DigiblurDIY in this video.`

That means that I still need to do it manualy with serial-to-USB convertor.
If yes is the flashing of r3 same as it was at r1? 
Putting in prerequisites mode and using one of the tool esptool.py or NodeMCU PyFlasher?

What confusing me is this video: [https://youtu.be/BUfWytrzrJ4?t=271](url)
Then I do not need at the end to use serial-to-USB and I can do it over web interface?

And last one, maybe offtopic: is it posible to use at once two sensors: BH1750 and BME280 on this device?
Thx

meingraham, 2019-07-24:
Apologies, things changed in the last week. I need to update that bit about the R3

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY

easyyu, 2019-07-24:
Ok understand, there are no need for serial-to-USB convertor.
This article is really confusing:

`Select a pre-compiled Tasmota binary or your own self-compiled binary. It must be smaller than 500k. Do NOT use the sonoff-minimal pre-compiled binary as it does not allow to change any setting.`

From the posted link every file have bigger value than 500kB.

But if I would like the process is same as i was at R1?
Thx

arendst, 2019-07-24:
> From the posted link every file have bigger value than 500kB.

sonoff-basic.bin | http://thehackbox.org/tasmota/sonoff-basic.bin | 444k | 20190724 14:08 GMT+2 | sonoff-basic.map.gz | c2413a2
-- | -- | -- | -- | -- | --

The above is compiled for core 2.4.2. You can also use core 2.3.0 images which are even smaller.

umibozu83, 2019-08-21:
hi I also bought Sonoff r3 I flashed the sonoff-basic.bin firmware but the device no longer responds. I soldered the serial pins but I don't know how to shorten the gpio0. do you know where to find it on the circuit?

auriux, 2019-08-21:
> 
> 
> hi I also bought Sonoff r3 I flashed the sonoff-basic.bin firmware but the device no longer responds. I soldered the serial pins but I don't know how to shorten the gpio0. do you know where to find it on the circuit?

Hi,
Hold the button before powering with serial adapter, this will take you to flashing mode.
The button is the Gpio0.

GilDev, 2019-08-31:
Hi,
I successfully flashed Tasmota onto my Basic R3 using [Sonoff DIY OTA](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY) but I think I made a mistake and launched a firmware upgrade from the web interface. I guess the firmware in the URL field was `sonoff.bin` instead of `sonoff-basic.bin`. After a few seconds, the relay was indefinitely switching on and off. I unplugged the Sonoff but when plugging it back it does nothing, no LEDs light up at all.
Should I now look into flashing via USB to serial interface to unbrick my device?
Thanks!

mrcasablr, 2019-09-09:
@GilDev would you mind sharing details on the steps to flash tasmota on sonoff basic r3?

easyyu, 2019-09-10:
> 
> 
> @GilDev would you mind sharing details on the steps to flash tasmota on sonoff basic r3?

I'm interested too!

meingraham, 2019-09-10:
@mrcasablr @easyyu  - https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY

auriux, 2019-09-10:
Has anyone tried other Sonoff devices running eWelink 3.1 or higher to use DIY mode?

GilDev, 2019-09-10:
@mrcasablr @easyyu The link with instructions is in my message and it’s the one @meingraham also gave. When it failed I had to flash with an FTDI adapter.

drodiger, 2019-09-24:
Few points from me about DIY process:
- You need to add your Basic R3 and Mini DIY into the eWeLink first and update to latest 3.3 firmware from Sonoff. This version is compatible with tool from Itead which is also V3.3
- Download Tool from Itead Git. It works on Windows only. When you first start it, it will ask you to allow it on Firewall.
- Download sonoff-basic.bin and sonoff.bin from Tasmota Git
- On you mobile phone or AP, create sonoffDiy access point with password 20170618sn
- Disconnect Sonoff Basic/Mini from power, put jumper and connect power back. It will connect to this AP. You should see it on your phone
- In Windows, you should stop Firewall, because Tool is creating http server to send firmware, and not to interfere with that process
- With Windows laptop connect to that same AP (sonoffDiy). You should see 2 devices connected to that AP
- Start the Tool. It will find the Sonoff Device. First try to put relay On/Off to test if the connection is working correctly. If it responds with json and relay switches, than everything is OK. If it says, wait before you send new requests, than connection is not working correctly. Sometimes you need to restart laptop, because wireless didn't connect or you didn't stop firewall on private/public/Domain. On your phone, you should see the IP address in the connected devices list for your Sonoff Device and your laptop. You can try to ping Sonoff IP from laptop command line.
- Click Flash Firmware, select sonoff-basic.bin (smaller than sonoff.bin), select your Sonoff device and click start
- It will start in a few seconds, be patient. If percentage doesn't go from 0 to 100%, you can restart with the process (ONLY if it didn't start at all = 0%)
- When flashing finishes, close the tool, device will restart and than you should connect with your laptop to new AP which should aper, which will be something like sonoff-216537
- Open web page on http://192.168.4.1. Set your router Access Point and password. If you enter wrong AP or password, press and hold for 40 seconds, the button on Sonoff which changes relay state. This will reset everything. You should then reconnect to new sonoff-xxx AP, open 192.168.4.1 and set correct password.
- If your Sonoff connects to your router AP successfully, than you should find IP address (in router devices list or using nmap/zenmap or some other tool. If it didn't connect, you can probably try to reset it with 40sec hold.
- Connect with your laptop to Router AP and open web page from that IP, something like http://192.168.1.128 (depending on DHCP range you have setup on your router)
- Now, you can flash sonoff.bin from Flash Firmware, and set everything from name, MQTT, template...
- There is no need to solder anything with this process or use USBtoSerial...

Thanks

meingraham, 2019-09-24:
@drodiger 

Note that this is fairly well documented in the [Tasmota wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY). For example it notes that the device has to be on Itead's 3.1 version or greater. It also covers a couple of key points about the second firmware update missing from your procedure that are critical. In addition, it provides alternate information for OTA DIY flashing without a Windows computer.

dedors, 2019-10-06:
I just bricked 3 sonoff basic r3 devices following every step from the tasmota wiki. At the step where tasmota was running fine and I wanted to upgrade from "sonoff-basic" to "sonoff".
To be honest, on the first one I forgot the reset 5. But the same stuff happened with the other 2 as well. I did upload via browser, not via URL.

I tried flashing with serial but failing so far.

meingraham, 2019-10-06:
@dedors 

This appears to be hit or miss unfortunately. You did try to upgrade by downloading sonoff.bin to your PC and then performing the upgrade via File Upload (the second option on the Firmware Upgrade web page) and selecting the file you downloaded? I just want to confirm because if this method also failed, then there seems to be no foolproof means to get a successful upgrade after the initial DIY flash.

I guess after the first one didn't upgrade successfully, I might have been gun shy to try two more.

Persistence on serial flashing will pay off... eventually.

Mike

mrcasablr, 2019-10-06:
I bricked one sonoff R3 by not doing Reset 5 and doing web ui upgrade. The other 2 are doing quite well when I followed the instructions to the dot. 

I want to recover the bricked sonoff r3. Does anyone have specific instructions to serial flash the r3?

meingraham, 2019-10-06:
@mrcasablr https://www.youtube.com/watch?v=A-WcIz47Lco&t=167

dedors, 2019-10-07:
It's pretty much the same to flash via serial as the old basics. The layout can be found here:
https://forum.creationx.de/forum/index.php?thread/1940-info-sonoff-basic-r3/
(If the holes are on the right, from up to down its GND, TX, RX, 3.3V).
Press the button when connecting the usb. You dont need the DIY pins here.

I revived all my 3 basics now, to be sure I first blank flashed with 1MB first (http://forum.creationx.de/index.php?attachment/2269-blank-1mb-2mb-4mb-mit-esptool-zip/ ). It needs a reboot after that.

I did flash my own compiled sonoff.bin 6.5 (2.4.2 core). On 2 devices I did a direct OTA web update to sonoff.bin 6.6 (2.5.2 core) which worked despite both firmwares are slighly above 512KB.
On the third I first flashed sonoff-minimal, and was not able to update to sonoff.bin (several tries). After I did a reset 5 I could update like the others via OTA web.

@meingraham It happened after I uploaded the firmware via "file upload". BUT I clicked on Web OTA before, but the upgrade didn't happen. I can't remember if I clicked that on all 3 devices or just one tho.

bovirus, 2020-05-21:
@dedors

Then you can flash a Sonoff basic R3 in teh same way you can flash Sonoff Basic R1/R2.

Using USB serial TTL converter and tasnota flash program using tasmota firmware?

What is maxuim size for tasmota firmware that I can flash?

rolandking, 2022-07-13:
Super useful thread. For anyone reading it, I used the usual UART flashing method to flash RavenSystem's homekit-compatible firmware. I did dump off the flash first incase I needed to go back but erase, flash and followed the boot up instruction and (after 10 minutes .. what is it doing during that time) it was up on the network and available to homekit. OTA may indeed work but it was SO EASY to do it with a UART that anyone who wants to do that, I'd suggest just do that. 

oh and for pins I just found a long pin header, snipped it off the short side of the plastic the pins go thru, fiddled it in from the back and soldered it on top. That left just enough pin sticking out to fairly easily connect the UART to. I figured I'd only have to do it a few times. 

So thanks threadstarter for the picture with the pin designations on, that helped a lot.