arendst/Tasmota #5922: Can I connect an RS232/serial device to ESP8266 (Sonoff devices) ?

sofakng, 2019-06-05:
I'm using a Sonoff POW R2 and Sonoff Basic devices which are ESP8266 based.

Is it possible to use these devices as relays (which is what they are), but also connect a serial/rs232 device to them? (i.e. I want to both control the power on/off AND use it for remotely communicating with an rs232/serial device)

meingraham, 2019-06-05:
POW uses the serial comms itself. Besides, it's not recommended to connect external components to power monitoring devices like the POW.

With the Basic, yes, it is possible.

meingraham, 2019-06-05:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-05 by ascillato2