arendst/Tasmota #5924: Sonoff Pow R2 & Backlog power 2 off; delay 20; power 2 on;

dmitriytim, 2019-06-06:
I want to reboot the entire network, including the router, using Sonoff Pow R2 & Backlog,
<code>
http://192.168.1.2/cm?user=admin&password=password&cmnd=Backlog%20power%202%20off;%20delay%2020;%20power%202%20on
</code>
only leads to shutdown, :(
everything works fine with Sonoff-basic 

version Sonoff-Tasmota 6.5.0

meingraham, 2019-06-06:
You have extra spaces in your command

`Backlog power 2 off; delay 20; power 2 on`

Should be
`Backlog power2 off; delay 20; power2 on`

Also, you may need to "escape" the semicolons (`%3B`) in addition to the spaces.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

dmitriytim, 2019-06-06:
Thanks, helped

closed 2019-06-06 by dmitriytim