arendst/Tasmota #5925: Support sonoff mini

leschNUX, 2019-06-06:
Hi,

Itead new sonoff mini.

https://www.itead.cc/sonoff-mini.html

Jason2866, 2019-06-06:
Since the device is not available yet it cant be supported at the moment.

ascillato2, 2019-06-06:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-06 by ascillato2

Derkades, 2019-07-31:
It is available now: https://www.itead.cc/smart-home/sonoff-mini.html

arendst, 2019-07-31:
See #6022

meingraham, 2019-07-31:
@leschNUX @Derkades 

See also - https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-DIY