arendst/Tasmota #5928: Shelly H&T sensor support

JoergTiedemann, 2019-06-07:
Hi
does Tasmota supports the Shelly H&T sensor device and if not is it planned to add the support for this device to Tasmota
Regards Joerg

meingraham, 2019-06-07:
Tasmota does not currently have a deep sleep mode. Your H&T battery would be drained very quickly. Shelly has an MQTT mode that you can use to so you can use the H&T without the Shelly Cloud. If you want additional "Tasmota" features, take a look at the sterfanbode Tasmota fork. It has deep sleep capability that may function on the H&T.

Please close this issue. Address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-07 by JoergTiedemann

DarkSunOne, 2021-05-05:
Theirs a usb power supply now, so tasmota would make sense again ;-)

thfrei, 2022-11-25:
Can I revisit that? It looks like it has been merged? 
At least Stefan Bode is on the contributors list along with his features: 

[image]

thfrei, 2022-11-25:
Ok, I got it, it's all good: https://tasmota.github.io/docs/DeepSleep/#lm1
Sorry if I bothered anyone :-)