arendst/Tasmota #5929: Deprecated configs

Psychokiller1888, 2019-06-08:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Some configurations are deprecated but still present in the user_setting file

```
// -- MQTT topics ---------------------------------
  // Example "tasmota/bedroom/%topic%/%prefix%/" up to 80 characers
#define MQTT_FULLTOPIC         "%prefix%/%topic%/" // [FullTopic] Subscribe and Publish full topic name - Legacy topic
```
Especially the example. On discord I have been forced into not changing the full topic, **especially** not like the exemple because it reverses topic and prefix. **I have been banned from Discord daring to ask the reason not to change it** and challenging the sayers to give me the reason instead of telling me "because" and "we are the almighty"

**Describe the solution you'd like**
Remove deprecated configs or unban me from discord

**Describe alternatives you've considered**
Using micro python. I just need some actions to trigger on pir/ switch/ button etc

**Additional context**
[image]

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-06-08:
If you do not use HAss you can change the fulltopic as you like.

HAss autodiscovery changes the fulltopic for you to better handle it's communication.

jziolkowski, 2019-06-08:
We didn't tell you that you can't add things like bedroom or kitchen to full topic. We told you you shouldn't remove %prefix% from full topic, as clearly explained in the wiki.

Read this: https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT#mqtt-topic-definition

Psychokiller1888, 2019-06-08:
I Never said I'd remove topic, I got flamed because my full topic was wrong and that I shouldn't touch it. Read your discord history

meingraham, 2019-06-08:
Seems that you have chosen your online presence username perfectly.

Psychokiller1888, 2019-06-08:
Even out of discord the trolling continues?

jziolkowski, 2019-06-08:
Read what I said. Prefix, not topic. You got flamed because instead on focusing on the issue at hand you got all butthurt because you've been told to keep the full topic default so we can narrow down the problem at this moment without receiving an essay on tasmota design and why it is constructed as it is. Not to mention that it is already clearly described in the wiki. When someone is starting a sentence with "I'm not some child..." when receiving a direct (even if not thoroughly explained) response on why something is wrong, then don't expect people to treat you seriously if you take it personally. I'm done with this thread, and wasting time on people who a) clearly don't bother to read at least the basics in the wiki, and b) have reading comprehension issues.

arendst, 2019-06-08:
Closing as the question has been answered

closed 2019-06-08 by arendst