arendst/Tasmota #5932: Using Flow Sensor Causing YF-S201 causing "Exceptional Restart" v6.5.0

asking23, 2019-06-09:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
I am using NodeMCU v2. with Tasmota 6.5.0(release-sonoff), 

Build Date & Time | 2019-03-19T12:24:10
-- | --

Core/SDK Version | 2_3_0/1.5.3(aec24ac9)
-- | --

I am using 1 Nos. of Ultrasonic Sensor for Tank Water Level and Using 1 Relay for Motor Control. It works just fine without introduction of YF-S201 Flow Sensor. As Soon as Flow Sensor was configured for D5 Pin as Counter. Module Started Rebooting Unexpectedly and Randomely. "Restart Reason : Exception"

[image]

[image]

```
STATUS 0 OUTPUT HERE:
09:54:33 MQT: stat/ultrasonic/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"ultrasonic","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[3,1,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:54:33 MQT: stat/ultrasonic/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T11:29:05","StartupUTC":"2019-06-08T16:55:28","Sleep":250,"CfgHolder":4617,"BootCount":1187,"SaveCount":1483,"SaveAddress":"FA000"}}
09:54:33 MQT: stat/ultrasonic/STATUS2 = {"StatusFWR":{"Version":"6.5.0(release-sonoff)","BuildDateTime":"2019-03-19T12:24:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:54:33 MQT: stat/ultrasonic/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Superx",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008008","280500000100000000000000000000000000","00000000"]}}
09:54:33 MQT: stat/ultrasonic/STATUS4 = {"StatusMEM":{"ProgramSize":507,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDAE394","000783A0","23B617CE","00003BC0"]}}
09:54:33 MQT: stat/ultrasonic/STATUS5 = {"StatusNET":{"Hostname":"ultrasonic-4771","IPAddress":"192.168.1.234","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:1E:12:A3","Webserver":2,"WifiConfig":5}}
09:54:33 MQT: stat/ultrasonic/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.188","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1E12A3","MqttUser":"openhabian","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:54:33 MQT: stat/ultrasonic/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 09 04:24:33 2019","Local":"Sun Jun 09 09:54:33 2019","StartDST":"Sun Oct 27 02:00:00 2019","EndDST":"Sun Oct 27 02:00:00 2019","Timezone":99,"Sunrise":"06:04","Sunset":"19:32"}}
09:54:33 MQT: stat/ultrasonic/STATUS10 = {"StatusSNS":{"Time":"2019-06-09T09:54:33","Switch1":"ON","Switch2":"OFF","SR04":{"Distance":21.544}}}
09:54:34 MQT: stat/ultrasonic/STATUS11 = {"StatusSTS":{"Time":"2019-06-09T09:54:33","Uptime":"0T11:29:05","Vcc":2.740,"SleepMode":"Dynamic","Sleep":250,"LoadAvg":29,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Superx","BSSId":"60:A4:4C:ED:3C:30","Channel":1,"RSSI":38,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
As soon as D5 Pin is configured as Counter and When YF-S201 Starts giving pulses. Module Resets itself and Dip is observed in connected relay. i tried changing Active low to active high configuration for relay, but behavior remains same.

### EXPECTED BEHAVIOUR
Module Resets Randomly and within few seconds as soon as D5 Pin is configured as Counter.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-06-09:
You cant use a switch on Gpio 0. In wiki is explained why. 
Connect it to a different Gpio

asking23, 2019-06-09:
Actually its only Push to ON Button configured. Actually its just configured. I use it rarely once in a month. Do you mean configuring it will cause this weird behavior ?

ascillato2, 2019-06-10:
Hi, 

I'm using also the same flow meter and I don't have any restart. Zero since it is installed more than a year ago. I have been updating my device with every Tasmota update.

So, please, check:

+ that your device receives enough power
+ your flash was erased before flashing (if not, use command reset 5)
+ update to latest Tasmota version from: http://thehackbox.org/tasmota/020300/sonoff.bin

ascillato2, 2019-06-10:
Hi, any news on this?

ascillato2, 2019-06-11:
Closing this issue as there is no feedback. Please, when you have time, tell us if you could solve your issue. Thanks

closed 2019-06-11 by ascillato2

asking23, 2019-06-13:
Hi i did try removing GPIO 0 from switch configuration. Still as soon as flow sensor is connected via 3.3v to 5v logic converter. And D5 pin configured as counter. Module still restarts exceptional.