arendst/Tasmota #5933: Sonoff Basic R2, 5V out

waldorffsmama, 2019-06-09:
I have found a few people here asking if there is a 5V out pin on the Sonoff Basic R2 but in the threads I have seen, nobody seems to have found it. Well, there is a 5V out on the board and it works like a charme. I have tested it powering the RCWL-0516 microwave movement detector:

[image]

Jason2866, 2019-06-09:
Please close.
We dont want to publish mod infos which can be dangerous to people which have not the skills to do it safe. 
Connecting additional stuff to the poor power supply of sonoff basic is a bad idea anyway. The power supply is on its design limits already. So there is great chance that the device will have a defect soon.
Hopefully it doesnt die with burning your apartment or house down. No insurance will pay anything in this case

ascillato2, 2019-06-09:
Closing this issue as it is not related to Tasmota Software.

Besides a sonoff hardware is not designed for that use.

closed 2019-06-09 by ascillato2