arendst/Tasmota #5934: Password input suggestion

WayneManion, 2019-06-09:
The current method of input for the Wi-Fi password is a single input field that is invisible to the user. It is pretty easy for the user to enter an incorrect password, particularly if entering the password on a touchscreen. 

If a device is already inside a wall box or requires disassembly or soldering to gain access to the programming pins, this can be extremely aggravating. 

I suggest that one or both of these features be added to reduce the likelihood of entering an incorrect password: 

1. Add a button to the password fields that will allow the user to see the contents of the password field.  https://www.w3schools.com/howto/howto_js_toggle_password.asp
2. Require that the password be entered twice, so that a simple one-time typo doesn't cause the device to accept an incorrect password. 

I think that if either of these changes are incorporated into Tasmota, that it should probably also apply to the MQTT password field.

Thank you for your consideration.

Jason2866, 2019-06-09:
Thanks for your suggestion. 
There is already a easy way getting to the safe side to recover.
See [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Initial-Configuration#using-web-ui)

arendst, 2019-06-11:
I love this.

Once, a long time ago, Tasmota provided all passwords in readable format. Was great. I liked it...

Then people came along complaining about readable passwords which seems to be a no go (for them). Over time the passwords readability has been reduced to zero resulting in what you experience now.

Your suggestions will lead to extra code taking up space which could have been used for more useful items. See https://stackoverflow.com/questions/51121625/place-mask-unmask-button-inside-a-password-input-elementfor what is needed.

For now, won't change.

arendst, 2019-06-11:
But as always I couldn't resist and will release the visibility toggle on most passwords anyway ;-)

arendst, 2019-06-11:
Give it a try...

Jason2866, 2019-06-12:
Tested, works

closed 2019-06-12 by ascillato2