arendst/Tasmota #5935: Sonoff T1 US-3C (3 gangs) double press is working improperly

wongnam, 2019-06-09:
### BUG DESCRIPTION
Sonoff T1 US-3C double press is working improperly on version 6.5.0.15

- If i 2 press on button-1 then the relay-2 toggle
- and if i 2 press on button-2 then relay-3 toggle,
- 2 press on button-3 then relay-1 toggle.
- it also cannot perform correctly by rule to publish a cmnd mqtt topic.

I revert it back to version 6.5.0.14 that everything will back to normal.

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
21:48:55 MQT: stat/groundfloorlight1/STATUS = {"Status":{"Module":30,"FriendlyName":["Ground Floor Light 1","Front Yard Light","Ground Floor Balcony Light"],"Topic":"groundfloorlight1","ButtonTopic":"","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:48:55 MQT: stat/groundfloorlight1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:02","StartupUTC":"2019-06-09T14:45:53","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":709,"SaveAddress":"F6000"}}
21:48:55 MQT: stat/groundfloorlight1/STATUS2 = {"StatusFWR":{"Version":"6.5.0.15(sonoff)","BuildDateTime":"2019-06-07T14:15:57","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
21:48:55 MQT: stat/groundfloorlight1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-1","Wong"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0000800B","0A0500000100060000000000000000000000","00000001"]}}
21:48:55 MQT: stat/groundfloorlight1/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F0AA384","041101A0","020000CC","000008C0"]}}
21:48:55 MQT: stat/groundfloorlight1/STATUS5 = {"StatusNET":{"Hostname":"groundfloorlight1-5108","IPAddress":"192.168.12.117","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"DC:4F:22:DB:F3:F4","Webserver":2,"WifiConfig":4}}
21:48:55 MQT: stat/groundfloorlight1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBF3F4","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:48:55 MQT: stat/groundfloorlight1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 09 14:48:55 2019","Local":"Sun Jun 09 21:48:55 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00","Sunrise":"05:29","Sunset":"18:14"}}
21:48:55 MQT: stat/groundfloorlight1/STATUS10 = {"StatusSNS":{"Time":"2019-06-09T21:48:55"}}
21:48:55 MQT: stat/groundfloorlight1/STATUS11 = {"StatusSTS":{"Time":"2019-06-09T21:48:55","Uptime":"0T00:03:02","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
21:49:57 APP: Button3 multi-press 1
21:49:58 WIF: Checking connection...
21:49:58 WIF: Connected
21:49:58 APP: Button3 multi-press 2
21:49:58 SRC: Button
21:49:58 MQT: stat/groundfloorlight1/RESULT = {"POWER1":"ON"}
21:49:58 MQT: stat/groundfloorlight1/POWER1 = ON
21:49:58 SRC: MQTT
21:49:58 RSL: Received Topic cmnd/groundfloorlight1/POWER1, Data Size 12, Data PowerOnState
21:49:58 RSL: Group 0, Index 1, Command POWER, Data PowerOnState
21:49:58 MQT: stat/groundfloorlight1/RESULT = {"POWER1":"ON"}
21:49:58 MQT: stat/groundfloorlight1/POWER1 = ON
21:49:59 CFG: Saved to flash at F5, Count 710, Bytes 3584
21:50:02 APP: Button2 multi-press 1
21:50:03 APP: Button2 multi-press 2
21:50:03 SRC: Button
21:50:03 MQT: stat/groundfloorlight1/RESULT = {"POWER3":"ON"}
21:50:03 MQT: stat/groundfloorlight1/POWER3 = ON
21:50:03 SRC: MQTT
21:50:03 RSL: Received Topic cmnd/groundfloorlight1/POWER3, Data Size 12, Data PowerOnState
21:50:03 RSL: Group 0, Index 3, Command POWER, Data PowerOnState
21:50:03 MQT: stat/groundfloorlight1/RESULT = {"POWER3":"ON"}
21:50:03 MQT: stat/groundfloorlight1/POWER3 = ON
21:50:03 CFG: Saved to flash at F4, Count 711, Bytes 3584
21:50:07 APP: Button1 multi-press 1
21:50:07 APP: Button1 multi-press 2
21:50:07 MQT: tele/groundfloorlight1/STATE = {"Time":"2019-06-09T21:50:07","Uptime":"0T00:04:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"iot-1","BSSId":"B0:C7:45:36:85:E0","Channel":7,"RSSI":96,"LinkCount":1,"Downtime":"0T00:00:04"}}
21:50:07 SRC: Button
21:50:07 MQT: stat/groundfloorlight1/RESULT = {"POWER2":"ON"}
21:50:07 MQT: stat/groundfloorlight1/POWER2 = ON
21:50:08 SRC: MQTT
21:50:08 RSL: Received Topic cmnd/groundfloorlight1/POWER2, Data Size 12, Data PowerOnState
21:50:08 RSL: Group 0, Index 2, Command POWER, Data PowerOnState
21:50:08 MQT: stat/groundfloorlight1/RESULT = {"POWER2":"ON"}
21:50:08 MQT: stat/groundfloorlight1/POWER2 = ON
21:50:08 CFG: Saved to flash at FB, Count 712, Bytes 3584

```

### TO REPRODUCE
- use Sonoff T1-US 3C (3 gangs )
- flash it to fw 6.5.0.15
- double press on each button to try

### EXPECTED BEHAVIOUR
- when double press on the button that it should toggle the same relay or publish mqtt topic by rule correctly.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-06-09:
Pls update to latest dev release https://github.com/arendst/Sonoff-Tasmota/commit/1d32011596c4fa07d367a1b4be4911f741f1e78b should solve your issue.

wongnam, 2019-06-09:
latest FW is still the same issue.

```
22:27:45 CMD: status 0
22:27:45 MQT: stat/groundfloorlight1/STATUS = {"Status":{"Module":30,"FriendlyName":["Ground Floor Light 1","Front Yard Light","Ground Floor Balcony Light"],"Topic":"groundfloorlight1","ButtonTopic":"","Power":4,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:27:45 MQT: stat/groundfloorlight1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:21","StartupUTC":"2019-06-09T15:26:24","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":15,"SaveAddress":"F5000"}}
22:27:45 MQT: stat/groundfloorlight1/STATUS2 = {"StatusFWR":{"Version":"6.5.0.15(sonoff)","BuildDateTime":"2019-06-09T22:20:24","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
22:27:45 MQT: stat/groundfloorlight1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":1,"LogHost":"192.168.12.155","LogPort":514,"SSId":["iot-2","Wong"],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0000800B","0A0500000100060000000000000000000000","00000000"]}}
22:27:45 MQT: stat/groundfloorlight1/STATUS4 = {"StatusMEM":{"ProgramSize":490,"Free":512,"Heap":28,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0E0AA390","04110000","020000CC","000008C0"]}}
22:27:45 MQT: stat/groundfloorlight1/STATUS5 = {"StatusNET":{"Hostname":"groundfloorlight1-5108","IPAddress":"192.168.12.117","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"210.245.31.220","Mac":"DC:4F:22:DB:F3:F4","Webserver":2,"WifiConfig":4}}
22:27:45 MQT: stat/groundfloorlight1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.155","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_DBF3F4","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:27:45 MQT: stat/groundfloorlight1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 09 15:27:45 2019","Local":"Sun Jun 09 22:27:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+07:00"}}
22:27:45 MQT: stat/groundfloorlight1/STATUS10 = {"StatusSNS":{"Time":"2019-06-09T22:27:45"}}
22:27:45 MQT: stat/groundfloorlight1/STATUS11 = {"StatusSTS":{"Time":"2019-06-09T22:27:45","Uptime":"0T00:01:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","Wifi":{"AP":1,"SSId":"iot-2","BSSId":"38:17:C3:F1:90:01","Channel":1,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

ascillato, 2019-06-09:
@wongnam

Hi,

Theo has just made another fix. Cam you test it? Thanks

wongnam, 2019-06-09:
Confirm, the bug is fixed.
Now, the T1-US 3C works as expected.
Thanks you.

closed 2019-06-09 by wongnam