arendst/Tasmota #5936: Suggestion: Simulating long press

killjoy2505, 2019-06-09:
Recently installed Dooya blind motors. Grabbed the rfraw codes and converted them so I can use them with RF Bridge/Homeassistant.

There is a half open limit, but I need to hold the stop button for 3 seconds on the 2702 remote. Which basically spams the stop button rfrawcode.

So I figured I could use the backlog command for executive commands. However due to the lengthy codes I am maxing out the console command limit (i think). It works for 3 commands and if I add a 4th it just does not do anything, nothing in console and no codes are transmitted.

```
rfraw AA B0 35 05 04 12CA 05FA 015E 02E4 230A B481A3B2B2A3A3A3B2A3A3A3B2B2A3B2A3A3B2A3B2B2B2B2B2B2B2A3B2A3A3A3B2B2A3B2A3B2A3B2A3 55
```

Could one please remove the max limit for console commands?

Or either add an option to continuously transmit rfraw commands for an X amount of seconds?

For example
repeat 3 rfraw AA B0 35 05 04 12CA 05FA 015E 02E4 230A B481A3B2B2A3A3A3B2A3A3A3B2B2A3B2A3A3B2A3B2B2B2B2B2B2B2A3B2A3A3A3B2B2A3B2A3B2A3B2A3 55

Which would then in quick succession transmit the rfraw code for 3 seconds?

I would greatly appreciate it.

Jason2866, 2019-06-09:
Have you tried to send with rfraw 0xB0? This rfraw variant has the possibility to repeat code sending.
See https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/0xB0

killjoy2505, 2019-06-09:
This can be closed now, thanks a lot Jason2866

closed 2019-06-09 by killjoy2505