arendst/Tasmota #5938: Pow R2 not loading Webinterface

Wibias, 2019-06-09:
Soo today my POW-R-2 started becoming really unresponsive and almost can never be reached. I found out that it just becomes unresponsive when I set the module as `Sonoff Pow R2`, but when I set it as `Sonoff Pow` it works just fine. Can anyone help me with that?

Jason2866, 2019-06-09:
This is just a coincidence. If it worked before it seems it is hardware defective.
Do you have changed something in your wifi (hardware) setup?

Wibias, 2019-06-09:
Well. It works fine now. When it's set to Sonoff Pow or to Sonoff Basic. When I change it to Sonoff Pow R2 it doesnt work anymore. On Both Sonoff Pow-R-2. One of them was bought just 3 months ago. So not thinking about hardware there. Not changed anything on wifi side.

Jason2866, 2019-06-09:
What device is it? It wont work correct if you dont choose the right module.

Wibias, 2019-06-09:
It is a Sonoff Pow-R2 as I've already said. But when i select the right module: `Sonoff Pow R2` the webpage cant be reached anymore and loads for ever. It can be reached every now and then but with like 2 minutes loading time. When I select other modules the webpage loads just fine and I can atleast use the power toggle.

closed 2019-06-09 by ascillato2