arendst/Tasmota #5941: Topics on Sonoff 4C

georgelza, 2019-06-10:
on the 4C
would really prefer if each of the 4 load channels can have a topic of it's own, ie:

home/sonoff_4c/oven
home/sonoff_4c/pool
home/sonoff_4c/geyser-kitchen
home/sonoff_4c/geyser-bathroom

at the moment it comes through as 

home/sonoff_4c/power1
home/sonoff_4c/power2
home/sonoff_4c/power3
home/sonoff_4c/power4

and have the topics be independent of the hostname (which is mostly just used for network purposes/identification.

G

Jason2866, 2019-06-10:
Flash space is very rare. Will not be done because this change would need to much of it.
You can close this.

georgelza, 2019-06-10:
there is already under the "Other" section 4 x friendly name fields, pretty much actually just need these to be mapped to the topic names for the 4 load/channels.
G

Jason2866, 2019-06-10:
Not possible by design.

meingraham, 2019-06-10:
Friendly Names are "free form" text which is not compatible with MQTT topic names.

Tasmota design considers the physical device and the device contains multiple components. Even a "simple" device can have a relay, an LED, a button, etc. If you define a different topic for relays, then why not for the other components. The configuration quickly gets very complicated if you go down this route. Not to mention that the coding complexity and memory consumption increases.

georgelza, 2019-06-10:
Agree
But like on the sonof basic, you have the ability to give it a %topic% that makes sense, independent of the hostname.

We basically have here 4 x sonoff basics... I. One package which we can only refer to as Power1-4

Last I check this was a feature request...  thread, and normal logic follows we should be allowed to request and then state/justify why... and well not just be told sorry no/close topic...

closed 2019-06-10 by georgelza

meingraham, 2019-06-10:
I think what went unsaid is that the folks responding are taking into account many of the previous 5,940 issues and countless thousands of comments, including Theo's. So it's somewhat of a "been there, done that" shortcut to set expectations quickly. Understand that with Tasmota, it is not only the relays that have a state and need to be accounted for when considering how to transmit the information about each component. If using individual topics for each component, a Sonoff Basic would be

- home/sonoff/oven
- home/sonoff/oven_button
- home/sonoff/oven_led

A 4Ch would be:

- home/sonoff_4c/oven
- home/sonoff_4c/oven_button
- home/sonoff_4c/oven_led
- home/sonoff_4c/pool
- home/sonoff_4c/pool_button
- home/sonoff_4c/pool_led
- home/sonoff_4c/geyser-kitchen
- home/sonoff_4c/geyser-kitchen_button
- home/sonoff_4c/geyser-kitchen_led
- home/sonoff_4c/geyser-bathroom
- home/sonoff_4c/geyser-bathroom_button
- home/sonoff_4c/geyser-bathroom_led

The code would have to allow for unique naming for each component. There would be no leveraging for a class of device with enumeration (e.g., `Power<x>`). Now, add to this the fact that users can add components to practically any device, you would have to account for all possible configurations (13 GPIO plus A0). Sure, it's possible, but it's messy.

Thus the answer - highly unlikely to happen. Please close this topic.

Regards.

Mike