arendst/Tasmota #5944: Sonoff Pow reports 0 volts when relay off

SimoneBnc, 2019-06-11:
Hi. Is there a way to prevent Tasmota on sonoff pow to report 0 volts when relay is turned off? If it could still provide a voltage reading too it would be nice.

thxthx0, 2019-06-11:
SetOption21 1

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#SetOption21

SimoneBnc, 2019-06-11:
thanks, I was searching for "voltage"..... :(

closed 2019-06-11 by arendst

localhost61, 2019-06-11:
@SimoneBnc 
While you are at it, could you check consistency of Energy display?
I just received a Neo Coolcam 16A that shares the same driver.
It is calibrated now and shows proper Voltage/Power/Current but Energy is totally wrong.
When I power on the plug with a 2kW resistive load running 2mn, i should get 2kW*(2/60)=0.067kWh and I get 0.6kWh which is 10 time more.
`
16:20:54 CMD: Backlog energyreset1 0; power on;delay 20;status 10;delay 1180;power off;energyreset
16:20:54 MQT: stat/Spare/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
16:20:55 MQT: stat/Spare/RESULT = {"POWER":"ON"}
16:20:55 MQT: stat/Spare/POWER = ON
16:20:55 MQT: stat/Spare/RESULT = {"Delay":20}
16:20:57 MQT: stat/Spare/STATUS10 = {"StatusSNS":{"Time":"2019-06-11T16:20:57","ENERGY":{"TotalStartTime":"2019-06-11T16:20:54","Total":0.007,"Yesterday":0.000,"Today":0.007,"Power":1995,"ApparentPower":1995,"ReactivePower":0,"Factor":1.00,"Voltage":220,"Current":9.054}}}
16:20:57 MQT: stat/Spare/RESULT = {"Delay":1180}
16:22:55 MQT: stat/Spare/RESULT = {"POWER":"OFF"}
16:22:55 MQT: stat/Spare/POWER = OFF
16:22:56 MQT: stat/Spare/RESULT = {"EnergyReset":{"Total":0.598,"Yesterday":0.000,"Today":0.598}}`

arendst, 2019-06-11:
Xavier, pls open a new issue for this.

I also found some miscalculations but I'm still not sure if it's wrong or not.

localhost61, 2019-06-11:
OK for the issue. :-/