arendst/Tasmota #5945: Wrong energy calculation

localhost61, 2019-06-11:
I just received my new NEO CoolCam 16A, flashed it using [Tuya-convert](https://www.heise.de/ct/artikel/Tuya-Convert-Escaping-the-IoT-Cloud-no-solder-need-4284830.html) method.
I applied blakadder's template of [this](https://blakadder.github.io/templates/neocoolcam_16a.html) page to initialize GPIOs. The voltage readout was wrong, I got 750V instead of about 220V. 
So, I calibrated it with a 40W resistive load and confirmed with a 2kW load, using an Eastron SDM230 (class 1.0) as reference.
```
voltageset 228.5
-> {"VoltageSetCal":599}

currentset 174
-> {"CurrentSetCal":2589}
```
I didn't use PowerSet command, the readout was consistent and thought it would alter the result of the previous ones.

### BUG DESCRIPTION
The issue concerns Energy calculation, it seems about 10 times too high. It becomes obvious with a 2kW load. If I power a 2kW load during 2mn (0.0333h) it consumes 2*0.0333=0.067kWh but I get 0.6kWh
This is what I realize with the next Backlog command (I repeat what I already posted [here](https://github.com/arendst/Sonoff-Tasmota/issues/5944#issuecomment-500869917))
```
16:20:54 CMD: Backlog energyreset1 0; power on;delay 20;status 10;delay 1180;power off;energyreset
16:20:54 MQT: stat/Spare/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
16:20:55 MQT: stat/Spare/RESULT = {"POWER":"ON"}
16:20:55 MQT: stat/Spare/POWER = ON
16:20:55 MQT: stat/Spare/RESULT = {"Delay":20}
16:20:57 MQT: stat/Spare/STATUS10 = {"StatusSNS":{"Time":"2019-06-11T16:20:57","ENERGY":{"TotalStartTime":"2019-06-11T16:20:54","Total":0.007,"Yesterday":0.000,"Today":0.007,"Power":1995,"ApparentPower":1995,"ReactivePower":0,"Factor":1.00,"Voltage":220,"Current":9.054}}}
16:20:57 MQT: stat/Spare/RESULT = {"Delay":1180}
16:22:55 MQT: stat/Spare/RESULT = {"POWER":"OFF"}
16:22:55 MQT: stat/Spare/POWER = OFF
16:22:56 MQT: stat/Spare/RESULT = {"EnergyReset":{"Total":0.598,"Yesterday":0.000,"Today":0.598}}
```
[image]

### REQUESTED INFORMATION
- [X] Device used (i.e. Sonoff Basic) : [NEO CoolCam 16A](https://fr.aliexpress.com/item/-/32963796295.html)
- [X] Tasmota binary firmware version number used : 6.5.0.15 build  09dcb93 / self-compiled
- [X] Development IDE - Compiler / Upload tools used : PlatformIO /OTA with Tuya-convert on RPi3B
```
19:16:11 CMD: status 0
19:16:11 MQT: stat/Spare/STATUS = {"Status":{"Module":0,"FriendlyName":["Neo Coolcam 16A"],"Topic":"Spare","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:16:11 MQT: stat/Spare/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://rpi.local:5000/sonoff-FR.bin","RestartReason":"Power on","Uptime":"0T03:12:18","StartupUTC":"2019-06-11T14:03:53","Sleep":50,"CfgHolder":4617,"BootCount":13,"SaveCount":47,"SaveAddress":"F5000"}}
19:16:11 MQT: stat/Spare/STATUS2 = {"StatusFWR":{"Version":"6.5.0.15(sonoff)","BuildDateTime":"2019-06-09T18:13:12","Boot":4,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
19:16:11 MQT: stat/Spare/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["freebox",""],"TelePeriod":300,"Resolution":"418180C0","SetOption":["00208008","140500000100060000000000000000000000","00000000"]}}
19:16:11 MQT: stat/Spare/STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["0000040C","0F882190","04008001","040000C6","000080C0"]}}
19:16:11 MQT: stat/Spare/STATUS5 = {"StatusNET":{"Hostname":"Spare-5727","IPAddress":"192.168.4.4","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"BC:DD:77:77:77:77","Webserver":2,"WifiConfig":4}}
19:16:11 MQT: stat/Spare/STATUS6 = {"StatusMQT":{"MqttHost":"pi.local","MqttPort":1883,"MqttClientMask":"DVES_%04X","MqttClient":"DVES_9B765F","MqttUser":"pi","MqttCount":20,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:16:11 MQT: stat/Spare/STATUS7 = {"StatusTIM":{"UTC":"Tue Jun 11 17:16:11 2019","Local":"Tue Jun 11 19:16:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:06","Sunset":"22:05"}}
19:16:11 MQT: stat/Spare/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
19:16:11 MQT: stat/Spare/STATUS10 = {"StatusSNS":{"Time":"2019-06-11T19:16:11","ENERGY":{"TotalStartTime":"2019-06-11T16:20:54","Total":0.742,"Yesterday":0.000,"Today":0.742,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":228,"Current":0.000}}}
19:16:11 MQT: stat/Spare/STATUS11 = {"StatusSTS":{"Time":"2019-06-11T19:16:11","Uptime":"0T03:12:18","Vcc":3.468,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"POWER":"OFF","Wifi":{"AP":1,"SSId":"freebox","BSSId":"E8:65:44:44:44:44","Channel":11,"RSSI":86,"LinkCount":15,"Downtime":"0T00:01:30"}}}
```
### TO REPRODUCE
Issue that should affect Sonoff POW too, it shares same energy meter chip HLW8012

localhost61, 2019-06-11:
Did 2 other tests measuring  40W during 6mn (delay 3600) and got 33Wh instead of 4Wh.

arendst, 2019-06-12:
Just did a test on a BlitzWolf SHP6 using this template:
```
{"NAME":"BlitzWolf SHP5","GPIO":[56,255,158,255,0,134,0,0,131,17,132,21,0],"FLAG":0,"BASE":45}
```
Using a 150W bulb it correctly reports 5W over two minutes.
```
19:33:30 CMD: backlog status 8;power 1;delay 1200;power 0;status 8
19:33:30 MQT: stat/shp6/STATUS8 = {"StatusSNS":{"Time":"2019-06-12T19:33:30","ENERGY":{"TotalStartTime":"2018-11-25T17:57:09","Total":2.392,"Yesterday":0.004,"Today":0.010,"Power":0.40,"ApparentPower":21.43,"ReactivePower":21.40,"Factor":0.02,"Voltage":225.6,"Current":0.095}}}
19:33:30 MQT: stat/shp6/RESULT = {"POWER":"ON"}
19:33:30 MQT: stat/shp6/POWER = ON
19:33:31 MQT: stat/shp6/RESULT = {"Delay":1200}
19:35:31 MQT: stat/shp6/RESULT = {"POWER":"OFF"}
19:35:31 MQT: stat/shp6/POWER = OFF
19:35:31 MQT: stat/shp6/STATUS8 = {"StatusSNS":{"Time":"2019-06-12T19:35:31","ENERGY":{"TotalStartTime":"2018-11-25T17:57:09","Total":2.397,"Yesterday":0.004,"Today":0.015,"Power":68.20,"ApparentPower":152.71,"ReactivePower":136.60,"Factor":0.45,"Voltage":224.9,"Current":0.679}}}
```
It uses the same driver as your Neo Coolcam.

I'll do some further tests on different hardware.

arendst, 2019-06-12:
As the Neo is configured for HLW8012 I would suggest you change it's config to BL0937 by changing GPIO04 from 133 to 134.

I haven't seen any HLW8012 lately in new devices. It's mostly BL0937.

You may need to recalibrate.

localhost61, 2019-06-12:
My latest readings tended to confirm what you tell me now. HLW8012 was for POW R1 and my further tests on LED lamps show no active power :-/ 
I'll check right now.

localhost61, 2019-06-12:
@arendst  you got it right! Many thanks!
 I get 82Wh instead of 67Wh now, the measure is not especially accurate but it's consistent.

I close the post, the error was mine :-D

closed 2019-06-12 by localhost61

localhost61, 2019-06-12:
After removing the inner sticker, and the plastic safety cover with a 2mm Hex key, the digital PCB appears and I can confirm there is a Belling BD0937 chip in my device as U5.

[image]

[image]

RezzZ, 2019-09-25:
@localhost61 I noticed the same issues as you with my new 16a wifi switches so I switched gpio4 to 134 and it looks better but not great.
Voltage hangs around 280...., and Watt usage is around 15 to 20 watts higher that actual

localhost61, 2019-09-25:
Yes, it absolutely needs [calibration](https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration). For me that was quite easy, I own everything needed for the operation.