arendst/Tasmota #5946: Opening Closing as a percentage of time

matteo2019, 2019-06-11:
I thank all those who answer me in advance. I have a dual R2 sonoff with stefanbode tasmota firmware. My problem is to open / close the shutter at a certain time with a certain percentage. Example: I set the 7 o'clock timer and the shutter is raised with the seconds established by the shutteropenduration function, I open it 100%. At 8 o'clock I want to lower the blinds by 20% of the total opening, always setting the timer. I await your answers, please explain in detail.

meingraham, 2019-06-11:
Support for Stefan's firmware fork should be directed to his GitHub.

Please close this issue.

matteo2019, 2019-06-11:
how can I refer you to your GitHub?

ascillato2, 2019-06-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-11 by ascillato2

localhost61, 2019-06-11:
@matteo2019 
Michael was talking about [stefan's fork](https://github.com/stefanbode/Sonoff-Tasmota)
For your usage you'll have to enable and use [Rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) together with Clock#Timer triggers.
Current Stefanbode fork is almost up to date with Tasmota, so you can use Tasmota wiki for your information.

matteo2019, 2019-06-11:
localhost61. I need some examples to understand it well. Could you help me?

matteo2019, 2019-06-11:
examples of rules to apply

localhost61, 2019-06-11:
These are some of the commands I enter in the console to rule my rollershutters which are
- currently using Tasmota (not a fork)
- wired with Relay 1=Down, Relay2=Up
- all wired with wall switches 
- opening with sunrise but one after the other with a 2" delay to avoid current rush if all of them were launched together.
- at 100% open they wait 200ms and roll back during 200ms to release the pressure.

This is for 4th rollershutter, delay is sun+6":
```
poweronstate 0			// Keep relay(s) off after power on
ledstatus 1			// LED open when a relay is ON
setoption0 off			// PowerOn relays state is Off
setoption14 on			// Relay interlock
setoption26 on			// Postfix on Power msg 
switchmode1 2			// Relay follow inverted wall switch state (ON when pin grounded)
switchmode2 2			// Relay follow inverted wall switch state (ON when pin grounded)
pulsetime1 120			// Power1 will be released after 20s
pulsetime2 120			// Power2 will be released after 20s
timezone 99			// Enable Sunset/Sunrise computation
backlog latitude 48.11; longitude -1.68		// geolocation for sun elevation computation
Timer1 {"Arm":1,"Mode":2,"Window":0,"Days":"1111111","Repeat":1,"Output":1,"Action":3}	// Mode2=Sunset
Timer2 {"Arm":1,"Mode":1,"Window":0,"Days":"1111111","Repeat":1,"Output":2,"Action":3}	// Mode1=Sunrise
rule1 on Clock#Timer=1 do backlog delay 60; power1 1 endon On Clock#Timer=2 do backlog delay 60; power2 1;delay 220;power2 0;delay 2;power1 1;delay 2;power1 0 endon // Define both  rules launched at sunset & sunrise
rule on				// Enable rules
```
It doesn't fit your need but can be of little help. Good luck

Jason2866, 2019-06-12:
@matteo2019 You know what a wiki is?
It is absolute no go to change or add something which is a personal question!!!
https://github.com/arendst/Sonoff-Tasmota/wiki/Opening---Closing-as-a-percentage-of-time/8c6ac8c8f38dab29849e4da66ae92e205e39b5b4
Page will be reverted.
Take this as a warning. Disregard the rules one more time and you will be locked out from here.

Please read and respect [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

matteo2019, 2019-06-12:
@ Jason2866 do not get upset. I don't know how to move to ask for information. I have to learn where I can ask for advice and information hoping to be helped. I'm sorry you turned to threats.

matteo2019, 2019-06-12:
@Jason2866 if I have made some false steps, forgive me. I was not motivated by arrogance and indifference.

localhost61, 2019-06-12:
@matteo2019 Tasmota is a collaborative work and you are **destroying** it. It should be read only but in order to enable everybody to give help you are allowed to add information. I personally don't consider @Jason2866 warning as offending, he is preserving the collaborative job and I think everybody here should thank him for that.

matteo2019, 2019-06-12:
Quite right! I apologize it was not my intention to ruin his work. I would like to return to the chat where meingraham invited me

Jason2866, 2019-06-12:
Most Wiki updates the last months are from @blakadder and @meingraham 
I try to support ;-)

matteo2019, 2019-06-12:
Thanks for the support.  I would like to participate in the chat.  Why do I get written that I was banned?

Ottieni Outlook per Android<https://aka.ms/ghei36>