arendst/Tasmota #5947: Console update after a non-activated timer.

matteo2019, 2019-06-11:
I thank all those who answer me in advance. I have a double R2 sonoff with stefanbode tasmota firmware. When there is no power in the house the sonoff is off, when it is reset the timers have not been activated. I need to tell the sonoff to activate all the timers that were not executed when he was turned off.

meingraham, 2019-06-11:
Support for Stefan's firmware fork should be directed to his GitHub.

Please close this issue.

matteo2019, 2019-06-11:
if I opened this request it means that I could not find what I need

ascillato2, 2019-06-11:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-11 by ascillato2

meingraham, 2019-06-11:
A. You neglected to complete the information requested in the template.
B. Had you read that, you would have been directed to:
   1. Read the wiki thoroughly first
   2. Ask any questions unanswered by the wiki on the Discord chat. You may have gotten some assistance there even for Stefan's fork.