arendst/Tasmota #5948: how to make sonoff work without internet connection?

matteo2019, 2019-06-11:
I have sonoffs that carry out set commands, I don't interact with them from the outside world (internet) they carry out all the commands set manually. How to automatically update daylight saving time and standard time without internet? if the current goes away how to keep the time in the sonoff memory? (with a buffer battery?)
I ask for your help

meingraham, 2019-06-11:
You can point your devices to an NTP server (e.g., a Raspberry Pi) which you set up on your network. That server gets its time from the Internet when a connection is available. You can also attach a real-time clock to your server rather than requiring an Internet connection to retrieve the time. Your server is available all the time to service time requests from your devices.

You can attach an [RTC component](https://github.com/arendst/Sonoff-Tasmota/wiki/DS3231---External-RTC) to each of your devices.

No way to keep the time in the device if it loses power. I don't think having each of your devices plugged in to a UPS for battery back-up is feasible.

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-06-11:
@meingraham  explained how Tasmota gets his time to function correctly without a hardware add one.
One point is not 100% correct, with adding the linked RTC as hardware addon it is possible to have
the correct time after a power outage without internet connection.
The device needs ONE time a internet connection to get correct time from a NTP server.
With this the RTC is set to correct time and date. After that Tasmota gets his time from this RTC when
no NTP server is reachable.

closed 2019-06-11 by ascillato2