arendst/Tasmota #5950: Not able to send large ir values

sreeprem, 2019-06-12:
We have come across a problem where we are unable to send large ir values ie, not above 120
through tasmota mqtt . So this caused a major pause to the work flow .when we send less than 100 raw values its working.

(WORKING CMD)
08:17:00 CMD: irsend 0,4350,4350, 550,1650, 500,600, 450,1650, 550,1650, 500,550, 550,550, 500,1650, 550,550, 550,550, 500,1650, 500,550, 550,550, 500,1650, 550,1650, 500,600, 500,1600, 550,600, 500,1600, 550,1600, 550,1650, 500,1650, 550,550, 500,1650, 550,1600, 550,1600, 550,550, 550,550, 500,600, 500,600, 500,1650, 500,600, 500,550, 550,1600, 550,1650, 500,1650, 500,600, 500,550, 500,600, 500,600, 500,600, 500,550, 500,600, 500,600, 450,1650, 550,1650, 500,1650, 500,1650, 500,1650, 550
08:17:00 RSL: stat/sonoff/RESULT = {"IRSend":"Done"}

But when we try to send more than that no feedback(NOT WORKING CMD) 
irsend 0,4350,4350, 550,1650, 500,600, 450,1650, 550,1650, 500,550, 550,550, 500,1650, 550,550, 550,550, 500,1650, 500,550, 550,550, 500,1650, 550,1650, 500,600, 500,1600, 550,600, 500,1600, 550,1600, 550,1650, 500,1650, 550,550, 500,1650, 550,1600, 550,1600, 550,550, 550,550, 500,600, 500,600, 500,1650, 500,600, 500,550, 550,1600, 550,1650, 500,1650, 500,600, 500,550, 500,600, 500,600, 500,600, 500,550, 500,600, 500,600, 450,1650, 550,1650, 500,1650, 500,1650, 500,1650, 550,4350,4350, 550,1650, 500,600, 450,1650, 550,1650, 500,550, 550,550, 500,1650, 550,550, 550,550, 500,1650, 500,550, 550,550, 500,1650, 550,1650, 500,600, 500,1600, 550,600, 500,1600, 550,1600, 550,1650, 500  

We are using generic ac remote in India which transmits raw value nearly 230. We also tried changing Mqtt Packet size but it didnt work,

Thank you  
[image]

Jason2866, 2019-06-12:
You have to compile your own version with increased buffer size.
Details in this thread https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-440716483

sreeprem, 2019-06-12:
> You have to compile your own version with increased buffer size.
> Details in this thread [#2116 (comment)](https://github.com/arendst/Sonoff-Tasmota/issues/2116#issuecomment-440716483)

It is for rec ,we want to increase size for ir send .

Thankyou

Jason2866, 2019-06-12:
You cant send such a long string via console (console buffer is not big enough).
It is only possible via mqtt. Have you tried?
Please set log level to debug and post outputs when command is sended via mqtt.
You have to increase mqtt buffer size as described

ascillato2, 2019-06-12:
Closing this issue as it is **duplicated**. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-12 by ascillato2

Jason2866, 2019-06-12:
@ascillato2  @arendst  It is not possible to send a such long string via mqtt and comand irsend.
Tried to send 
`irsend 0,4350,4350,550,1650,500,600,450,1650,550,1650,500,550,550,550,500,1650,550,550,550,550,500,1650,500,550,550,550,500,1650,550,1650,500,600,500,1600,550,600,500,1600,550,1600,550,1650,500,1650,550,550,500,1650,550,1600,550,1600,550,550,550,550,500,600,500,600,500,1650,500,600,500,550,550,1600,550,1650,500,1650,500,600,500,550,500,600,500,600,500,600,500,550,500,600,500,600,450,1650,550,1650,500,1650,500,1650,500,1650,550,4350,4350,550,1650,500,600,450,1650,550,1650,500,550,550,550,500,1650,550,550,550,550,500,1650,500,550,550,550,500,1650,550,1650,500,600,500,1600,550,600,500,1600,550,1600,550,1650,500`
results in
```

00:00:00 CFG: Loaded from flash at F9, Count 268
00:00:00 Project sonoff WeMini 192-168-2-150 Version 6.5.0.16(sonoff)-2_5_2
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-17DBAE-7086...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on sonoff-17DBAE-7086 with IP address 192.168.2.150
12:59:18 MQT: Attempting connection...
12:59:18 MQT: Connected
12:59:18 MQT: tele/sonoff-17DBAE/LWT = Online (retained)
12:59:18 MQT: cmnd/sonoff-17DBAE/POWER = 
12:59:18 MQT: tele/sonoff-17DBAE/INFO1 = {"Module":"Generic","Version":"6.5.0.16(sonoff)","FallbackTopic":"cmnd/sonoff-17DBAE_fb/","GroupTopic":"sonoffs"}
12:59:18 MQT: tele/sonoff-17DBAE/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-17DBAE-7086","IPAddress":"192.168.2.150"}
12:59:18 MQT: tele/sonoff-17DBAE/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4027ff40 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000044 depc:0x00000000"}

```
Status 0
```
13:03:04 CMD: status 0
13:03:04 MQT: stat/sonoff-17DBAE/STATUS = {"Status":{"Module":18,"FriendlyName":["WeMini 192-168-2-150"],"Topic":"sonoff-17DBAE","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.2.20/web/tasmota/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:03:53","StartupUTC":"2019-06-12T10:59:11","Sleep":50,"CfgHolder":4617,"BootCount":214,"SaveCount":269,"SaveAddress":"F8000"}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS2 = {"StatusFWR":{"Version":"6.5.0.16(sonoff)","BuildDateTime":"2019-06-12T11:41:10","Boot":31,"Core":"2_5_2","SDK":"3.0.0-dev(c0f7b44)"}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Jason_Home_WLAN",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00400009","280500000100060000000000000000000000","00000101"]}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS4 = {"StatusMEM":{"ProgramSize":479,"Free":524,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0F8AA184","040103A1","00000496","000000C0"]}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS5 = {"StatusNET":{"Hostname":"sonoff-17DBAE-7086","IPAddress":"192.168.2.150","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"B4:E6:2D:17:DB:AE","Webserver":2,"WifiConfig":5}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.114","MqttPort":1883,"MqttClientMask":"sonoff-17DBAE","MqttClient":"sonoff-17DBAE","MqttUser":"openhabian","MqttCount":1,"MAX_PACKET_SIZE":2000,"KEEPALIVE":30}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 12 11:03:04 2019","Local":"Wed Jun 12 13:03:04 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:09","Sunset":"21:15"}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS10 = {"StatusSNS":{"Time":"2019-06-12T13:03:04","AM2301":{"Temperature":24.6,"Humidity":47.5},"BME280":{"Temperature":28.2,"Humidity":43.4,"Pressure":968.5,"SeaPressure":1011.8},"PressureUnit":"hPa","TempUnit":"C"}}
13:03:04 MQT: stat/sonoff-17DBAE/STATUS11 = {"StatusSTS":{"Time":"2019-06-12T13:03:04","Uptime":"0T00:03:53","Vcc":3.093,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:1E:69:50","Channel":4,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:05"}}}
```
Settings:

```
#define USE_IR_REMOTE
  #define USE_IR_RECEIVE
    #define IR_RCV_BUFFER_SIZE      1000 
    #define IR_RCV_TIMEOUT          50   
    #define IR_RCV_MIN_UNKNOWN_SIZE 6   

```
and in platformio.ini

```
[common]
framework                 = arduino
board                     = esp01_1m
board_build.flash_mode    = dout

platform                  = ${core_active.platform}
build_flags               = ${core_active.build_flags}
                            -DMQTT_MAX_PACKET_SIZE=2000
```

Jason2866, 2019-06-12:
MAP file of used firmware
[map.zip](https://github.com/arendst/Sonoff-Tasmota/files/3281033/map.zip)

arendst, 2019-06-12:
@Jason2866 as your irsend string is only 609 bytes long there is no need to increase any parameter. If I try to send your string using mqtt-spy it is absorbed by tasmota and reported as being send. No exception noticed.

```
17:55:02 SRC: MQTT
17:55:02 RSL: Group 0, Index 1, Command IRSEND, Data 0,4350,4350,550,1650,500,600,450,1650,550,1650,500,550,550,550,500,1650,550,550,550,550,500,1650,500,550,550,550,500,1650,550,1650,500,600,500,1600,550,600,500,1600,550,1600,550,1650,500,1650,550,550,500,1650,550,1600,550,1600,550,550,550,550,500,600,500,600,500,1650,500,600,500,550,550,1600,550,1650,500,1650,500,600,500,550,500,600,500,600,500,600,500,550,500,600,500,600,450,1650,550,1650,500,1650,500,1650,500,1650,550,4350,4350,550,1650,500,600,450,1650,550,1650,500,55
17:55:03 MQT: stat/wemos4/RESULT = {"IRSend":"Done"}
```

Same for the original issue:
```
17:59:07 SRC: MQTT
17:59:07 RSL: Group 0, Index 1, Command IRSEND, Data 0,4350,4350, 550,1650, 500,600, 450,1650, 550,1650, 500,550, 550,550, 500,1650, 550,550, 550,550, 500,1650, 500,550, 550,550, 500,1650, 550,1650, 500,600, 500,1600, 550,600, 500,1600, 550,1600, 550,1650, 500,1650, 550,550, 500,1650, 550,1600, 550,1600, 550,550, 550,550, 500,600, 500,600, 500,1650, 500,600, 500,550, 550,1600, 550,1650, 500,1650, 500,600, 500,550, 500,600, 500,600, 500,600, 500,550, 500,600, 500,600, 450,1650, 550,1650, 500,1650, 500,1650, 500,1650, 550,43
17:59:07 MQT: stat/wemos4/RESULT = {"IRSend":"Done"}
```

I'll investigate the original issue where tasmota reports Done but the user experiences no IR signal send.

arendst, 2019-06-12:
I currently cannot test if IR signals are actually being send but...

When I try to send a string up to 974 bytes (= 204 values) using mqtt-spy, tasmota reports done. If you want to send 230 values you will need to increase MQTT_MAX_PACKET_SIZE in both lib/PubSubClient-EspEasy-2.6.09/PubSubClient.h AND sonoff_post.h to 1200. Raising it to 2000 will probably lead to exceptions as noted above as the array of values to send is reserved on the stack which only allows max 4k and is probably used by others too.

Now I think of it I will add a check and if there are more than 200 values I'll reserve space for the array on the heap. This will also fail in the future due to memory fragmentation but you have been warned.

Jason2866, 2019-06-12:
Thx, for looking into. Yes crashing is coming from too big mqqt packet size.
Knowing the limits helps to fiddle out the maximum (if next user is asking...)

sreeprem, 2019-06-12:
Thank you so much friends ,its working perfectly.Now i am able to send the ir raw value to wemos from cloudmqtt .

gastonc, 2019-07-17:
> Thank you so much friends ,its working perfectly.Now i am able to send the ir raw value to wemos from cloudmqtt .

hi sreeprem, 
I have the same problem. What changes did you make? Thank you

Saniksanik, 2019-09-08:
Please give me compiled firmware, that can read and write raw ir codes with big size!

gastonc, 2019-09-11:
hello
try with this bin

Saniksanik, 2019-09-11:
Thanks, but i don,t see file link

miquel-simo, 2020-05-04:
Hello, would be possible to have a precompiled binary with increased buffer size?. I tried to compile it by myself but everytime I try to flash it my device cracks. Thank you so much.