arendst/Tasmota #5952: next release? when?

riddik14, 2019-06-13:
for me it's a real problem, :-)) I'm waiting for the next release for the inclusion of the latest IR library for hvac control. when the next release is scheduled. thank you very much for your work.

Jason2866, 2019-06-13:
There is no need to wait for the next release. Use latest development build. It is very very stable and
improved to latest dev. Grab it from the official OTA Tasmota server
http://thehackbox.org/tasmota/020300/

ascillato2, 2019-06-13:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-13 by ascillato2

riddik14, 2019-06-13:
I have try but command from consolle { "Vendor":"PANASONIC","Power":1,"Mode":Cold, "FanSpeed":5,"Temp":16} AS EXAMPLE not work, from consolle the error is “invalid json”, Who? Tasmota release is  6.5.0.16

riddik14, 2019-06-13:
CMD: Status 0
18:33:31 MQT: nodemcu/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"nodemcu","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:33:31 MQT: nodemcu/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-sensors.bin","RestartReason":"Software/System restart","Uptime":"0T01:04:15","StartupUTC":"2019-06-13T16:29:16","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":12,"SaveAddress":"F8000"}}
18:33:31 MQT: nodemcu/stat/STATUS2 = {"StatusFWR":{"Version":"6.5.0.16(e43aa0b-sensors)","BuildDateTime":"2019-06-12T23:05:50","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:33:31 MQT: nodemcu/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","280500000100060000000000000000000000","00000200"]}}
18:33:31 MQT: nodemcu/stat/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":440,"Heap":11,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","0FDEE394","0017A3A4","B7FFBFCC","01DABBC0"]}}
18:33:31 MQT: nodemcu/stat/STATUS5 = {"StatusNET":{"Hostname":"nodemcu-5271","IPAddress":"192.168.1.201","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:6F:D4:97","Webserver":2,"WifiConfig":4}}
18:33:31 MQT: nodemcu/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.247","MqttPort":1883,"MqttClientMask":"nodemcu","MqttClient":"nodemcu","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:33:32 MQT: nodemcu/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 13 17:33:31 2019","Local":"Thu Jun 13 18:33:31 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"04:46","Sunset":"20:54"}}
18:33:32 MQT: nodemcu/stat/STATUS10 = {"StatusSNS":{"Time":"2019-06-13T18:33:32","ANALOG":{"A0":387}}}
18:33:32 MQT: nodemcu/stat/STATUS11 = {"StatusSTS":{"Time":"2019-06-13T18:33:32","Uptime":"0T01:04:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"Wifi":{"AP":1,"SSId":"wlan2","BSSId":"5C:49:79:94:76:B7","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:13"}}}
18:34:05 MQT: nodemcu/tele/HASS_STATE = {"Version":"6.5.0.16(e43aa0b-sensors)","BuildDateTime":"2019-06-12T23:05:50","Core":"2_3_0","SDK":"1.5.3(aec24ac9)","Module":"Generic","RestartReason":"Software/System restart","Uptime":"0T01:04:49","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:13","MqttCount":1,"BootCount":5,"SaveCount":12,"IPAddress":"192.168.1.201","RSSI":"100","LoadAvg":38}
18:34:28 MQT: nodemcu/tele/STATE = {"Time":"2019-06-13T18:34:28","Uptime":"0T01:05:12","SleepMode":"Dynamic","Sleep":50,"LoadAvg":67,"Wifi":{"AP":1,"SSId":"wlan2","BSSId":"5C:49:79:94:76:B7","Channel":3,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:13"}}
18:34:28 MQT: nodemcu/tele/SENSOR = {"Time":"2019-06-13T18:34:28","ANALOG":{"A0":384}}

Jason2866, 2019-06-13:
Panasonic AC is not supported in Tasmota. It isnt enough that the latest IR library is used.
A driver for Tasmota is needed too. Someone has to make...