arendst/Tasmota #5953: Has anyone ever tried flashing a shelly OTA?

HenniC4, 2019-06-13:
I stumbled over this unrelated post: https://forum.creationx.de/forum/index.php?thread/1561-firmware-update-1-4-6/ Which basically says: "Oh, yeah, fix that by installing a special beta, you do that by just entering http://192.168.x.x/ota?url=http://repo.shelly.cloud/firmware/review/1-4-6-wifireconfix/switch1-esp8266-last.zip into your browser and shelly will update itself"

I am not a speciallist for these espressif architectures, but this seems to be so simple, (4 binaries and one self-explaining json with memory locations and SHA1 hashes for the bins) what stops me from trying the obvious is i don't want to brick a shelly because it was hellish to get them inplace inside my walls.

Has someone ever tried out this way of flashing a shelly?

Jason2866, 2019-06-13:
It is easy just flash as described in [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisites)
Shelly is fully supported and need NO special firmware. Just use sonoff.bin

ascillato2, 2019-06-13:
Closing this issue as it has been answered.

If you need further assistance, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-13 by ascillato2

jtoledo1974, 2019-12-22:
I don't believe Jason did actually answer the original question. The fact is that original Shelly devices are upgradable OTA. Would it be possible to upgrade them feeding them a tasmota firmware?

ascillato, 2019-12-22:
No, it is not possible. Different memory map (2MB vs 1MB) and propietary OTA Cloud protocols.

jtoledo1974, 2019-12-22:
Thank you for the prompt reply!

idallen, 2023-07-25:
Yes, you can flash Tasmoda into a Shelly device OTA via an intermediate:  https://github.com/arendst/mgos-to-tasmota
(I just flashed Tasmota 13 into my Shelly UNI using the Web URL.)