arendst/Tasmota #5961: Press, double press and hold via MQTT

JumpmanJunior, 2019-06-17:
I would like to send all button actions via MQTT without effect on the local device/relay. If I configure the button as a switch with according SwitchMode and a rule for sending MQTT message I can only distinguish between press and hold. If I configure it as a button I can distinguish between press, double press and hold, but one of these three is always triggering the local relay and not sending an MQTT message.

I think the simplest solution would be to add an option which would enable double presses for switches configured as pushbuttons. Instead of another option further switchmodes could be introduced.

It would be even better if Tasmota could recognize triple and quadruple presses and send hold and release messages for button presses.

meingraham, 2019-06-17:
Button presses have pre-defined functions in Tasmota. You can alter the function associated with single/double press and hold. As the default double-press action is to act on the second relay, the associated command is `Power2`, but the other actions are "hard-coded".

https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

Please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

JumpmanJunior, 2019-06-17:
I am aware of the predefined actions for buttons. That's why I suggested to introduce an additional option or switchmode. Following are the answers to the questions from the template:

**Have you look for this feature in other issues and in the wiki?**
Yes. There are no solutions but statements that this is not possible at the moment.

**Is your feature request related to a problem? Please describe.**
I have a Shelly1 with two connected push buttons. One is for switching the integrated relay. I want to send the actions on the other one to my home automation system. This is working for single press and hold, but not for double press.

**Describe the solution you'd like**
Add an additional option or switchmode in order to enable double press for switches. Additional nice-to-have features: triple and quadruple presses as well as hold and release messages via MQTT.

**Describe alternatives you've considered**
I have tired to achieve this with a Button instead of a Switch. Buttons support double press as well as hold, but because of the pre-defined functions they cannot send all events via MQTT.

I hope I have provided now all relevant information which are needed.

epandi, 2019-06-18:
Cant you use rules for this scenario? Where you trigger a custom event based in the button single/double press/ hold and also trigger these custom events using mqtt cmnd?

JumpmanJunior, 2019-06-18:
That's what I tried. But because of the pre-defined functions it is not possible to send MQTT messages for all events of a button.

closed 2019-07-21 by stale[bot]