arendst/Tasmota #5967: Support for BlitzWolf® BW-SHP8 3680W

WladyX, 2019-06-20:
I know it's early, but opening this just to let you know there is a new BW socket and when somebody gets their hands on one maybe post some pictures/instructions for flashing.
Thanks.

https://www.banggood.com/BlitzWolf-BW-SHP8-3680W-16A-Smart-WIFI-Wall-Outlet-EU-Plug-Socket-Timer-Remote-Control-Power-Monitor-Work-with-Alexa-Google-Assistant-IFTTT-p-1502402.html

arendst, 2019-06-20:
Try https://blakadder.github.io/templates/smanergy_KA10.html as it looks to me a perfect copy...

closed 2019-06-20 by arendst

mbellido, 2019-08-27:
I have tried to flash BW-SHP8 and configure it as Smanergy KA10 and it works correctly.
Thank you!

jujoparo, 2019-08-28:
> I have tried to flash BW-SHP8 and configure it as Smanergy KA10 and it works correctly.
> Thank you!

Could you explain how have you flash the bw-shp8? Have you needed to connect wires or through web interface?

mbellido, 2019-08-28:
I need to connect wires. Visit [https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S)](url) 

Flashing:
**GPIO0 Needs to be grounded in order to get into flashing mode.**
1) erase_flash
2) flash tasmota sonoff_sensor_6.6.0

When flashed, configure it as KA10 (64) via the web interface.
Finally you might need to calibrate your power monitoring device:[https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration](url)

drtsb, 2019-09-25:
Some photo
[image]
[image]

WladyX, 2019-10-03:
I was able to flash tasmota OTA on this using https://github.com/ct-Open-Source/tuya-convert.

iondulgheru, 2019-10-07:
I also tried https://github.com/ct-Open-Source/tuya-convert and it was a success. Easier that soldering on those small pins.

iondulgheru, 2019-10-23:
The only thing I don't like about this device is that it shows wrong voltage and amps. It shows 270V instead of 230 and when measuring amps it seems to have a deviation between 10% and 20%. Does it need any calibration? Or maybe the KA10 settings need any adjustments?

ascillato, 2019-10-23:
in the comment above it is explained that it requires calibration:

> Finally you might need to calibrate your power monitoring device:https://github.com/arendst/Sonoff-Tasmota/wiki/Power-Monitoring-Calibration

tiagofreire-pt, 2020-01-15:
Tested with the template for the KA-10.

I share my one, customized, and based on the KA-10:

`{"NAME":"BW-SHP8 16A","GPIO":[0,56,0,17,134,132,0,0,131,53,21,0,0],"FLAG":0,"BASE":64}`

schilea, 2020-01-22:
I have installed Tasmota without any problem using tuya-convert and it is working fine. But today when trying to integrate it into HA, I noticed it is not turning off anymore the power, even if the relay seems to be working. Wifi is fine, MQTT is fine, the indications in Tasmota and HA are fine, but the power is always on.
[image]
[image]
[image]

Have I done something wrong in the configuration? Did I manage to turn it into a dummy outlet?

ascillato, 2020-01-22:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

dajpereira, 2020-02-08:
> I have installed Tasmota without any problem using tuya-convert and it is working fine. But today when trying to integrate it into HA, I noticed it is not turning off anymore the power, even if the [...]

Whitch tutorial have you followed? I have bought 2 SHP8 and want to flash them.

schilea, 2020-02-09:
follow this one from digiblurDYI https://www.youtube.com/watch?v=dt5-iZc4_qU

mediarath, 2021-03-30:
[image]

Does anyone know of an alternative GPIO0?
Unfortunately it was not usable when new. The metal plate was not present. Could not solder anything on. Can I still save the connector somehow?

piotrpawelptak, 2022-03-28:
Hello, yesterday I've tried to tuya-convert the SPH8 but with no success. I have followed this tutorial https://www.youtube.com/watch?v=dt5-iZc4_qU and everything went fine untill the actual tasmotizing of the device. After many attempts to reconnect the syntax said that I probably have the new  version of firmware installed on the SPH8 and this can't be tasmotized. There was also a link related to workaround: https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02.

My question is: could someone specify which model exactly of SPH8 is supported by tuya-convert. I assume this would have to be produced before the change to the firmware. Where could I buy the supported model?

BR, Piotr

barbudor, 2022-03-28:
@piotrpawelptak 
no way to know
There are less and less units that can be converted by tuya convert
If you want to be on the safe side, purchase one with tasmota preloaded
https://templates.blakadder.com/preflashed.html

raptorecki, 2022-03-30:
> Hello, yesterday I've tried to tuya-convert the SPH8 but with no success

Same here. I gave it a try with the exact same setup for donor device (D1 Mini with ESP8266). It seems that OTA flashing is no longer working with the new SPH8 devices. I did, however, successfully flash it over the serial connection. SPH8 is quite easy to disassemble and tap into the TX/RX pins. It's way easier that way than looking for some old stock...

piotrpawelptak, 2022-03-30:
> Some photo
> 
> [image]
> 
> [image]
> 
> 

Thank you both for the quick replies. Could you please confirm if the order of the pins matches the one from the above photo?

raptorecki, 2022-03-30:
> Could you please confirm if the order of the pins you have used for serial flashing matches the one from the above photo?

Yes. I followed that exact pinout and it worked. GPIO0 needs to be connected to GND for flashing. 3.3V VCC from Pi Zero was suficient to power the ESP for flashing. Good luck!

ThomDietrich, 2022-12-10:
Hey guys,
the provided template above suggests that the red LED is connected to GPIO1. However this LED seems to be hardwired with the relay. I did not check the PCB but tried different template configurations. Can anyone confirm? If I am not utterly mistaken, the better template is therefore 

```
{"NAME":"SHP8","GPIO":[0,0,0,32,2720,2656,0,0,2624,576,224,0,0,0],"FLAG":0,"BASE":64}
```