arendst/Tasmota #5968: Historical power data incorrect on Shelly 2.5 device

guiumb, 2019-06-20:
### BUG DESCRIPTION

Dear All,

I have flashed one new Shelly 2.5 device with self compiled software version 6.5.0.14 and Shelly 2.5 buttons template.

The device is attached on a HomeAssistant RPI without any problem, but it seems that an issue regarding historical data was detected:
usualli the load of device is very poor (5 W medium) and just only for a few minutes in a day the load is 1500 W, but the historical values counted by device and his firmware are very hi!
In attach you can see 2 historical graph for instant power and historical daily power and a screenshot from device web status.

I have also a PZEM energy module and several Sonoff POW R2 connecetd on HA and flashed with 6.5.0 firmware and they works perfectly.

Should be exist and issue on historical data on Shelly 2.5?

Today I have update the device with last 6.5.0.16 firmware but the issue it is not fixed.

Many thanks in advance
Umberto

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [ X] Device used (i.e. Sonoff Basic) : Shelly 2.5
- [x ] Tasmota binary firmware version number used : 6.5.0.16 self-compiled
- [x ] Development IDE - Compiler / Upload tools used : Arduino 1.8.9 with 2.5.2
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
22:00:34 CMD: status 0
22:00:34 MQT: stat/bagnosala/STATUS = {"Status":{"Module":0,"FriendlyName":["BagnoSala","Sonoff2"],"Topic":"bagnosala","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
22:00:34 MQT: stat/bagnosala/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:41","StartupUTC":"2019-06-20T19:56:53","Sleep":50,"CfgHolder":4617,"BootCount":15,"SaveCount":86,"SaveAddress":"1FA000"}}
22:00:34 MQT: stat/bagnosala/STATUS2 = {"StatusFWR":{"Version":"6.5.0.16(sonoff)","BuildDateTime":"2019-06-20T21:55:07","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
22:00:34 MQT: stat/bagnosala/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["guilan",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["0000880B","010500000100060000000000000000000000","00000000"]}}
22:00:34 MQT: stat/bagnosala/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":1456,"Heap":22,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":255,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
22:00:34 MQT: stat/bagnosala/STATUS5 = {"StatusNET":{"Hostname":"sensore14","IPAddress":"10.0.0.164","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"24:62:AB:00:AE:10","Webserver":2,"WifiConfig":4}}
22:00:34 MQT: stat/bagnosala/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.150","MqttPort":1883,"MqttClientMask":"bagnosala","MqttClient":"bagnosala","MqttUser":"umbyha","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:00:34 MQT: stat/bagnosala/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 20 20:00:34 2019","Local":"Thu Jun 20 22:00:34 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"05:33","Sunset":"21:16"}}
22:00:34 MQT: stat/bagnosala/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
22:00:34 MQT: stat/bagnosala/STATUS10 = {"StatusSNS":{"Time":"2019-06-20T22:00:34","ANALOG":{"Temperature":61.2},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":123.323,"Yesterday":23.078,"Today":22.127,"Power":0,"ApparentPower":4,"ReactivePower":5,"Factor":0.07,"Voltage":206,"Current":0.022},"TempUnit":"C"}}
22:00:34 MQT: stat/bagnosala/STATUS11 = {"StatusSTS":{"Time":"2019-06-20T22:00:34","Uptime":"0T00:03:41","SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:04"}}}

guiumb, 2019-06-20:
[image]
[image]
[image]

arendst, 2019-06-21:
Cannot reproduce but made a change for energy calculation.

With logging set to 3 you can now see the values read from the energy measuring device. Perhaps it gives more information when you experience this bahaviour.

You might want to setup a syslog server to log these messages for further analysis.

ascillato2, 2019-06-23:
@guiumb 

Hi, any news on this?

guiumb, 2019-06-25:
Sorry for the delay but I have an internet outage in my home ! As soon as the providet fix it I will set up a syslog server in order to collect more detail logs.
I will keep you in touch.
Many thanks

guiumb, 2019-06-26:
I am back.
Using syslog I can not see much more data then web console:
SYSLOG:
10.0.0.164	Jun 26 20:49:31	sensore14			ESP-APP	Restarting
10.0.0.164	Jun 26 20:49:37	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 26 20:49:37	sensore14			ESP-HTP	Web server active on sensore14 with IP address 10.0.0.164
10.0.0.164	Jun 26 20:49:38	sensore14			ESP-MQT	Attempting connection...
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	Connected
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	tele/bagnosala/LWT = Online (retained)
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	cmnd/bagnosala/POWER = 
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	Subscribe to cmnd/bagnosala/#
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	Subscribe to cmnd/sonoffs/#
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	Subscribe to cmnd/bagnosala_fb/#
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	tele/bagnosala/INFO1 = {"Module":"Shelly 2.5 bts","Version":"6.5.0.16(sonoff)","FallbackTopic":"cmnd/bagnosala_fb/","GroupTopic":"sonoffs"}
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	tele/bagnosala/INFO2 = {"WebServerMode":"Admin","Hostname":"sensore14","IPAddress":"10.0.0.164"}
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	tele/bagnosala/INFO3 = {"RestartReason":"Software/System restart"}
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	stat/bagnosala/RESULT = {"POWER1":"ON"}
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	stat/bagnosala/POWER1 = ON
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	stat/bagnosala/RESULT = {"POWER2":"ON"}
10.0.0.164	Jun 26 20:49:40	sensore14			ESP-MQT	stat/bagnosala/POWER2 = ON
10.0.0.164	Jun 26 20:49:41	sensore14			ESP-HTP	Main Menu
10.0.0.164	Jun 26 20:49:45	sensore14			ESP-APP	Boot Count 19
10.0.0.164	Jun 26 20:49:45	sensore14			ESP-CFG	Saved to flash at 1F6, Count 114, Bytes 3584
10.0.0.164	Jun 26 20:49:49	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:14","SleepMode":"Dynamic","Sleep":50,"LoadAvg":40,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
10.0.0.164	Jun 26 20:49:49	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"1970-01-01T00:00:16","ANALOG":{"Temperature":66.9},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":245.170,"Yesterday":15.088,"Today":20.870,"Period":0,"Power":0,"ApparentPower":4,"ReactivePower":4,"Factor":0.07,"Voltage":204,"Current":0.021},"TempUnit":"C"}
10.0.0.164	Jun 26 20:50:04	sensore14			ESP-APP	(UTC) Wed Jun 26 18:50:10 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
10.0.0.164	Jun 26 20:50:51	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-26T20:50:58","Uptime":"0T00:01:16","SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
10.0.0.164	Jun 26 20:50:51	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-26T20:50:58","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":245.205,"Yesterday":15.088,"Today":20.906,"Period":36,"Power":0,"ApparentPower":4,"ReactivePower":4,"Factor":0.07,"Voltage":205,"Current":0.022},"TempUnit":"C"}
10.0.0.164	Jun 26 20:51:54	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-26T20:52:00","Uptime":"0T00:02:18","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
10.0.0.164	Jun 26 20:51:54	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-26T20:52:00","ANALOG":{"Temperature":67.4},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":245.217,"Yesterday":15.088,"Today":20.918,"Period":12,"Power":0,"ApparentPower":4,"ReactivePower":5,"Factor":0.09,"Voltage":207,"Current":0.022},"TempUnit":"C"}
10.0.0.164	Jun 26 20:52:54	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-26T20:53:01","Uptime":"0T00:03:19","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
10.0.0.164	Jun 26 20:52:54	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-26T20:53:01","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":245.253,"Yesterday":15.088,"Today":20.954,"Period":36,"Power":0,"ApparentPower":5,"ReactivePower":5,"Factor":0.06,"Voltage":211,"Current":0.023},"TempUnit":"C"}

Web Console:
00:00:00 SRC: Restart
00:00:00 Project sonoff BagnoSala Version 6.5.0.16(sonoff)-2_5_2
00:00:00 SHT: Sensor did not ACK command
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 guilan in mode 11N as sensore14...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sensore14 with IP address 10.0.0.164
00:00:06 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/bagnosala/LWT = Online (retained)
00:00:08 MQT: cmnd/bagnosala/POWER = 
00:00:08 MQT: Subscribe to cmnd/bagnosala/#
00:00:08 MQT: Subscribe to cmnd/sonoffs/#
00:00:08 MQT: Subscribe to cmnd/bagnosala_fb/#
[... cut ...]
20:53:01 MQT: tele/bagnosala/STATE = {"Time":"2019-06-26T20:53:01","Uptime":"0T00:03:19","SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:53:01 MQT: tele/bagnosala/SENSOR = {"Time":"2019-06-26T20:53:01","ANALOG":{"Temperature":67.1},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":245.253,"Yesterday":15.088,"Today":20.954,"Period":36,"Power":0,"ApparentPower":5,"ReactivePower":5,"Factor":0.06,"Voltage":211,"Current":0.023},"TempUnit":"C"}
I am doing some mistake?

guiumb, 2019-06-26:
This is logs device setting:
[image]

meingraham, 2019-06-26:
@guiumb 

Since the Shelly 2.5 uses the serial interface to exchange energy data, you need to set SerialLog to 0. In fact, this may be the source of some of your problems.

You may as well set WebLog to 2 since you have syslog turned on.

To capture more verbose logging you can set SysLog to 4.

ascillato, 2019-06-27:
@guiumb

Hi, any news on this? Have you set seriallog to 0?

guiumb, 2019-06-27:
Hi, no more plus data from traces with seriallog set to 0:
SYSYLOG:
10.0.0.164	Jun 26 20:54:26	sensore14			ESP-HTP	Configure Logging
10.0.0.164	Jun 27 20:12:37	sensore14			ESP-LOG	SerialLog 0, WebLog 4, SysLog 4, LogHost 10.0.0.6, LogPort 514, TelePeriod 60
10.0.0.164	Jun 27 20:12:37	sensore14			ESP-HTP	Configuration
10.0.0.164	Jun 27 20:12:37	sensore14			ESP-CFG	Saved to flash at 1F9, Count 119, Bytes 3584
10.0.0.164	Jun 27 20:12:48	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:12:48	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:12:49	sensore14			ESP-HTP	Main Menu
10.0.0.164	Jun 27 20:12:51	sensore14			ESP-HTP	Console
10.0.0.164	Jun 27 20:12:53	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-27T20:13:02","Uptime":"0T23:23:20","SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":84,"LinkCount":13,"Downtime":"0T00:01:05"}}
10.0.0.164	Jun 27 20:12:53	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-27T20:13:02","ANALOG":{"Temperature":61.2},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":268.936,"Yesterday":24.080,"Today":20.557,"Period":12,"Power":0,"ApparentPower":4,"ReactivePower":4,"Factor":0.07,"Voltage":197,"Current":0.021},"TempUnit":"C"}
10.0.0.164	Jun 27 20:13:09	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:13:09	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:13:29	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:13:29	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:13:50	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:13:50	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:13:54	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-27T20:14:03","Uptime":"0T23:24:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":13,"Downtime":"0T00:01:05"}}
10.0.0.164	Jun 27 20:13:55	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-27T20:14:03","ANALOG":{"Temperature":63.5},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":268.972,"Yesterday":24.080,"Today":20.592,"Period":36,"Power":1,"ApparentPower":4,"ReactivePower":4,"Factor":0.12,"Voltage":203,"Current":0.021},"TempUnit":"C"}
10.0.0.164	Jun 27 20:14:10	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:14:10	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:14:31	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:14:31	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:14:51	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:14:51	sensore14			ESP-WIF	Connected
10.0.0.164	Jun 27 20:14:55	sensore14			ESP-MQT	tele/bagnosala/STATE = {"Time":"2019-06-27T20:15:04","Uptime":"0T23:25:22","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":82,"LinkCount":13,"Downtime":"0T00:01:05"}}
10.0.0.164	Jun 27 20:14:55	sensore14			ESP-MQT	tele/bagnosala/SENSOR = {"Time":"2019-06-27T20:15:04","ANALOG":{"Temperature":64.2},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":269.020,"Yesterday":24.080,"Today":20.640,"Period":48,"Power":0,"ApparentPower":4,"ReactivePower":4,"Factor":0.02,"Voltage":197,"Current":0.022},"TempUnit":"C"}
10.0.0.164	Jun 27 20:15:11	sensore14			ESP-WIF	Checking connection...
10.0.0.164	Jun 27 20:15:11	sensore14			ESP-WIF	Connected

WebLOG:
20:12:45 LOG: SerialLog 0, WebLog 4, SysLog 4, LogHost 10.0.0.6, LogPort 514, TelePeriod 60
20:12:45 HTP: Configuration
20:12:45 CFG: Saved to flash at 1F9, Count 119, Bytes 3584
20:12:57 WIF: Checking connection...
20:12:57 WIF: Connected
20:12:58 HTP: Main Menu
20:12:59 HTP: Console
20:13:02 MQT: tele/bagnosala/STATE = {"Time":"2019-06-27T20:13:02","Uptime":"0T23:23:20","SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":84,"LinkCount":13,"Downtime":"0T00:01:05"}}
20:13:02 MQT: tele/bagnosala/SENSOR = {"Time":"2019-06-27T20:13:02","ANALOG":{"Temperature":61.2},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":268.936,"Yesterday":24.080,"Today":20.557,"Period":12,"Power":0,"ApparentPower":4,"ReactivePower":4,"Factor":0.07,"Voltage":197,"Current":0.021},"TempUnit":"C"}
20:13:17 WIF: Checking connection...
20:13:17 WIF: Connected
20:13:38 WIF: Checking connection...
20:13:38 WIF: Connected
20:13:58 WIF: Checking connection...
20:13:58 WIF: Connected
20:14:03 MQT: tele/bagnosala/STATE = {"Time":"2019-06-27T20:14:03","Uptime":"0T23:24:21","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":78,"LinkCount":13,"Downtime":"0T00:01:05"}}
20:14:03 MQT: tele/bagnosala/SENSOR = {"Time":"2019-06-27T20:14:03","ANALOG":{"Temperature":63.5},"ENERGY":{"TotalStartTime":"2019-06-08T16:17:14","Total":268.972,"Yesterday":24.080,"Today":20.592,"Period":36,"Power":1,"ApparentPower":4,"ReactivePower":4,"Factor":0.12,"Voltage":203,"Current":0.021},"TempUnit":"C"}
20:14:18 WIF: Checking connection...
20:14:18 WIF: Connected

Do you suggest more actions?

meingraham, 2019-06-27:
@guiumb Please paste the template you are using. Do the `GPIO assignments` match this?

{"NAME":"Shelly 2.5","GPIO":[`56,255,17,255,21,83,0,0,6,82,5,22,156`],"FLAG":2,"BASE":18}

guiumb, 2019-06-27:
This is the template:
{"NAME":"Shelly 2.5 bts","GPIO":[56,255,17,255,21,127,0,0,6,128,5,22,156],"FLAG":2,"BASE":18}

I have used 1 template suggested on a wiki web page because I am not interested in switch command using HW button.

meingraham, 2019-06-27:
https://blakadder.github.io/templates/shelly_25.html

This is the template to use if you do not use externally connected devices:

`{"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}`

You have actually selected the template for connecting external components

```
If you connect momentary switches and want press/double press/hold functionality, use the following template:

{"NAME":"Shelly 2.5 (buttons)","GPIO":[56,255,17,255,21,127,0,0,6,128,5,22,156],"FLAG":2,"BASE":18}
```

guiumb, 2019-06-27:
I have confused the two template strings. I will reconfigure the device immediatly and I forword a result ASAP.

guiumb, 2019-06-27:
No differences: I reset the total counter 10 minutes ago:
[image]

guiumb, 2019-06-27:
Should be the issue correlated with device temperauture? Is it seems quite hight.

meingraham, 2019-06-27:
I don't have one of these so cannot comment on what nominal device temperatures are.

When you flashed Tasmota initially, did you completely erase the flash first before uploading Tasmota?

I'm not sure if [`SetOption39`](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption39) is applicable. For readings below 5W, maybe the Shelly is sending lots of readings that may actually not be valid.

guiumb, 2019-06-27:
I have flashed the device with Arduino IDE tools and I have set the option to erase before write new firmware.
Actually device runs with release 6.5.0.16 that should be support the option. I will set it up.

guiumb, 2019-06-27:
no way... the total power increases continuously.
Reading the info on SetOption39 it seems that works only for chips like the HLW8032 (Blitzwolf SHP5) and CSE7766 (Sonoff S31, Sonoff POW R2), but Shelly 2.5 use ADE7953, right?

joba-1, 2019-06-27:
I have the device with original firmware. It shows the temperature in the same region (55-65 Celsius). Seems normal. It is not ambient, but on pcb temperature for detecting overheating, so it should be near warm parts by design.
Energy history with original fw is only possible using the cloud. It is not an onchip feature. Good, because it can be „fixed“ in tasmota :)

meingraham, 2019-06-29:
@guiumb 

Did you perform a calibration on your Shelly 2.5?

Also, seems that up to 90C is OK for internal temp - https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/2155006084598744/?comment_id=2155025641263455&comment_tracking=%7B%22tn%22%3A%22R%22%7D

arendst, 2019-06-30:
FYI Tasmota overtemp is set to 73C which will power off relays when reached.

arendst, 2019-06-30:
You might want to increase value resolution using commands ``wattres 2`` and ``voltres 1`` to see that there is actually load being taken by the device itself.

[image]

Frigorio, 2019-06-30:
I've read that original fw has dual power metering (for each channel). 
Is there any way to enable dual metering in Tasmota as well?

arendst, 2019-06-30:
Currently Tasmota accumulates both measurements into one. Would need a change in representation for both GUI and JSON leading to larger code size. Just a matter of code size vs functionality.

guiumb, 2019-06-30:
@meingraham
> Did you perform a calibration on your Shelly 2.5?

I didn't. I will do that and I'll update you.

@arendst
> Currently Tasmota accumulates both measurements into one. Would need a change in representation for both GUI and JSON leading to larger code size. Just a matter of code size vs functionality.

I think that should be a great improvement!
Could be used by default "single" summarized view with no impact on code size and "dual split" viewed as an option in a source code compiling.

> FYI Tasmota overtemp is set to 73C which will power off relays when reached.
Is there a settable parameter in source or is it "hardcoded"?

guiumb, 2019-07-01:
@meingraham 
>     Did you perform a calibration on your Shelly 2.5?
> I didn't. I will do that and I'll update you.

I performed power calibration as described in wiki page [https://github.com/arendst/Sonoff-Tasmota/wiki/Power-monitoring-calibration](url)  using current clamp.

Historical values are currently wrong and increases in a bad way with very low load (5 W induction charger); if I connect medium/hi load values become true and there is no fault increment.

Setting SetOption39 to 1 it seems that have no result on this device.

No way to have right historical values with low load !

meingraham, 2019-07-01:
"No way to have right historical values with low load"

What kind of load are you wanting to monitor?

What was the load you used to calibrate your device?

Mike

guiumb, 2019-07-01:
I have used resistive load device (100 Watt incandescent light bulb) as suggested on a wiki.

I have to monitor 2 wall sockets: in 1st there is always connected a toothbrush induction charger and more times during a day was replaced by hairdryer; the 2nd is used for power and iron.
I have to monitor these device because when the home global power reach max values (measured with PZEM power meter) that electical provider grant to me I disconnect this 2 devices using automation script in my Home Assistant.

meingraham, 2019-07-01:
So, you want to record total power used (toothbrush charger plus whenever someone uses the hairdryer and iron)? And, if you reach the limit set by your provider, you want to turn the relays of the Shelly 2.5 off?

My guess is that these devices may not have the accuracy for the toothbrush charger. Is the charger's daily draw that large? Perhaps you can have a good estimate and adjust your automation script to trigger at a lower value taking into account the approximate power that the charger would have drawn?

Mike

guiumb, 2019-07-01:
The main idea is monitor istant power: in case that main house power (on PZEM device) reach maximum trigger values and there is power charge on Shelly device the automation will open relays on device itself.
Because Tasmota provide also historical data my idea was to store them on Home Assistant DB for reports and consultation using same strategy as Sonoff POW R2 on other electical main devices connected on HA.
The trouble is that 1st wall socket located on bathroom is shared from toothbrush charger and hairdryer !

meingraham, 2019-07-01:
I think for your purposes, even if you "double count" your toothbrush charger, you can ignore that possibility. It is the iron and the hair dryer that are going to account for the vast majority of the power you need to track.

In any case, I'm going to assume people aren't going to leave the iron or the hair dryer on. Are you intending to turn them off even while they are in use if the power limit is reached? I hope it's not your spouse that's using those when that happens ;-)

Mike

guiumb, 2019-07-02:
You are right!
The main power drain is not toothbrush charger but usually is connected on wall socket and it fails historical data collection on Shelly. How can I ignore it? 
Obviously iron and hair dryer are connected and disconnected from wall socket just only when they needed.

As you say, I would automate disconnection the 2 loads on Shelly when the power limit is reached. In case my wife is using iron or hairdryer and relays goes off I have a little display with alarm messaging and buzzer that inform her !!!

meingraham, 2019-07-02:
@arendst Does SetOption39 apply to the power monitoring chip in the Shelly 2.5?

arendst, 2019-07-02:
Nope. It only applies to CSE7766/HLW8032 chips as used in Sonoff 32/Pow R2 and BlitzWolf SHP5

meingraham, 2019-07-02:
It seems that the Shelly 2.5 also suffers from low load inaccuracy issues. Is something similar to what you've done for the others possible for the Shelly power monitoring device(s)?

arendst, 2019-07-02:
I'll give it a try.

Frigorio, 2019-07-02:
A lot of smart plug have problems with low power metering.
I've got a Gosund SP111 (very similar to Blitzwolf SHP6) and when there is no load it sometimes returns false measurements. 
[image]

meingraham, 2019-07-02:
@Frigorio Yes. See [SetOption39](Commands#setoption39)

Frigorio, 2019-07-03:
> @Frigorio Yes. See [SetOption39](../wiki/Commands#setoption39)

Theo said:

> Nope. It only applies to CSE7766/HLW8032 chips as used in Sonoff 32/Pow R2 and BlitzWolf SHP5

Apparently not my case.

meingraham, 2019-07-03:
@Frigorio Did you try using the setting to see if it had an effect? It won't hurt to try it out.

The devices Theo mentioned and that are listed in the wiki are examples and not a comprehensive list of devices with these types of energy monitoring chips. In other words, the SHP5 is mentioned, but it may apply to the SHP6. For instance, the Blitzwolf SHP6 is basically the same device as the Gosund SP111. It is difficult if not impossible to list every variant without having a sample of each one. It's compounded because the markings on the chips aren't readily searchable. For example, some SHP6 come with a chip with `HJL-01, J1749CYH, D797480E` markings. You won't easily find these with an Internet search. One may be able to deduce based on the parts around that chip and the wiring, that its a HLW8012. Again though, the HLW8012 is not mentioned in the `SetOption39` writeup... perhaps because the code truly is specific to the HLW8032... or perhaps the HLW line of chips operate similarly and `SetOption39` may indeed have effect on devices with an HLW8012 such as the SHP6. Also, the SHP6 may have gone through a redesign and may now use an HLW8032.

Regards.

Mike

arendst, 2019-07-03:
For the Shelly 2.5 the sensor used is the ADE7953. This is connected using I2C GPIOs 12 and 14. The data read from the sensor directly relates to voltage, current and power. 

I see no reason to implement SetOption39 as the sensor correctly reports nothing for low loads and starts reporting valid data for loads above 2W.

guiumb, 2019-07-03:
So basically power chip sould not be send "corrupted" data when the power load is very low, right?
Reading datasheet on pag. 40 there is a section named "NO-LOAD DETECTION" that explain how chip works. 
Do you think that registry AP_NOLOADA, VAR_NOLOADA and VA_NOLOADA should be used to detect when chip is in "no load status" and they could be usable in your frimware?

Could I provide some other trace/dump using my device?

Many thanks for you investigation on the issue.

arendst, 2019-07-03:
No need. I did some reseach today. I cannot find why your period power is so high as it uses the same measured value as can be seen at power. Difference is that it is an accumulation of all measured power values. For some reason it becomes way too high.

I will investigate further but it's dificult as my shelly does no ghost reads.

You might want to use the latest dev version, keep teleperiod at 60 and enable logging 4. This will show measued data every second. Log during two teleperiod moments and provide it here. There must be a very large measurement at some second.

arendst, 2019-07-03:
weblog 3 should be fine.

arendst, 2019-07-03:
Also try a pre compiled version as I noticed your log contains a SHT message which probably interferes with the ade i2c signal.

guiumb, 2019-07-03:
I have download the source code and after compiling I have load the new firmware.
Cleared historical values.
Set up wblog 3.
this is the log:

22:48:42 LOG: SerialLog 0, WebLog 3, SysLog 2, LogHost , LogPort 514, TelePeriod 60
22:48:42 HTP: Configuration
22:48:42 ADE: U 5630047, I 2172 + 0 = 2172, P 136 + 0 = 136
22:48:43 CFG: Saved to flash at 1F8, Count 200, Bytes 3584
22:48:43 ADE: U 5627573, I 2205 + 0 = 2205, P 121 + 0 = 121
22:48:44 ADE: U 5635723, I 2205 + 0 = 2205, P 117 + 0 = 117
22:48:45 ADE: U 5635444, I 2205 + 0 = 2205, P 151 + 0 = 151
22:48:46 ADE: U 5633756, I 2205 + 0 = 2205, P 102 + 0 = 102
22:48:47 ADE: U 5645275, I 2205 + 0 = 2205, P 123 + 0 = 123
22:48:48 ADE: U 5659907, I 2172 + 0 = 2172, P 107 + 0 = 107
22:48:49 ADE: U 5654875, I 2172 + 0 = 2172, P 73 + 0 = 73
22:48:50 ADE: U 5646459, I 2172 + 0 = 2172, P 94 + 0 = 94
22:48:51 ADE: U 5646192, I 2172 + 0 = 2172, P 138 + 0 = 138
22:48:52 ADE: U 5642736, I 2172 + 0 = 2172, P 144 + 0 = 144
22:48:53 ADE: U 5640063, I 2172 + 0 = 2172, P 135 + 0 = 135
22:48:54 ADE: U 5635597, I 2172 + 0 = 2172, P 111 + 0 = 111
22:48:55 ADE: U 5644343, I 2172 + 0 = 2172, P 88 + 0 = 88
22:48:56 ADE: U 5641723, I 2172 + 0 = 2172, P 147 + 0 = 147
22:48:57 ADE: U 5652584, I 2172 + 0 = 2172, P 142 + 0 = 142
22:48:58 ADE: U 5651913, I 2172 + 0 = 2172, P 82 + 0 = 82
22:48:59 ADE: U 5634157, I 2172 + 0 = 2172, P 120 + 0 = 120
22:49:00 ADE: U 5628929, I 2172 + 0 = 2172, P 148 + 0 = 148
22:49:01 ADE: U 5626759, I 2172 + 0 = 2172, P 95 + 0 = 95
22:49:02 ADE: U 5636852, I 2172 + 0 = 2172, P 114 + 0 = 114
22:49:03 ADE: U 5638978, I 2172 + 0 = 2172, P 93 + 0 = 93
22:49:04 ADE: U 5638171, I 2172 + 0 = 2172, P 183 + 0 = 183
22:49:05 ADE: U 5627368, I 2172 + 0 = 2172, P 106 + 0 = 106
22:49:06 ADE: U 5625989, I 2172 + 0 = 2172, P 94 + 0 = 94
22:49:07 ADE: U 5630554, I 2172 + 0 = 2172, P 114 + 0 = 114
22:49:08 ADE: U 5631052, I 2172 + 0 = 2172, P 125 + 0 = 125
22:49:09 ADE: U 5637439, I 2172 + 0 = 2172, P 60 + 0 = 60
22:49:10 ADE: U 5638542, I 2172 + 0 = 2172, P 91 + 0 = 91
22:49:11 ADE: U 5635881, I 2172 + 0 = 2172, P 118 + 0 = 118
22:49:12 ADE: U 5634515, I 2172 + 0 = 2172, P 127 + 0 = 127
22:49:13 ADE: U 5633323, I 2172 + 0 = 2172, P 121 + 0 = 121
22:49:14 ADE: U 5633302, I 2172 + 0 = 2172, P 99 + 0 = 99
22:49:15 ADE: U 5644227, I 2172 + 0 = 2172, P 101 + 0 = 101
22:49:16 ADE: U 5576646, I 2172 + 0 = 2172, P 86 + 0 = 86
22:49:17 ADE: U 5639812, I 2172 + 0 = 2172, P 120 + 0 = 120
22:49:18 ADE: U 5644243, I 2172 + 0 = 2172, P 90 + 0 = 90
22:49:19 ADE: U 5646085, I 2172 + 0 = 2172, P 132 + 0 = 132
22:49:20 ADE: U 5629829, I 2172 + 0 = 2172, P 107 + 0 = 107
22:49:21 ADE: U 5625770, I 2172 + 0 = 2172, P 84 + 0 = 84
22:49:22 ADE: U 5613705, I 2172 + 0 = 2172, P 132 + 0 = 132
22:49:23 ADE: U 5615118, I 2172 + 0 = 2172, P -12 + 0 = -12
22:49:24 ADE: U 5617683, I 2172 + 0 = 2172, P 80 + 0 = 80
22:49:25 ADE: U 5616058, I 2172 + 0 = 2172, P 58 + 0 = 58
22:49:26 ADE: U 5618618, I 2160 + 0 = 2160, P 58 + 0 = 58
22:49:27 ADE: U 5614245, I 2160 + 0 = 2160, P 10 + 0 = 10
22:49:28 ADE: U 5619090, I 2160 + 0 = 2160, P 31 + 0 = 31
22:49:29 ADE: U 5614864, I 2160 + 0 = 2160, P 35 + 0 = 35
22:49:30 ADE: U 5631883, I 2160 + 0 = 2160, P 92 + 0 = 92
22:49:31 ADE: U 5634012, I 2160 + 0 = 2160, P 68 + 0 = 68
22:49:32 ADE: U 5640634, I 2242 + 0 = 2242, P 61 + 0 = 61
22:49:33 ADE: U 5602725, I 2253 + 0 = 2253, P 44 + 0 = 44
22:49:34 ADE: U 5609853, I 2253 + 0 = 2253, P 55 + 0 = 55
22:49:34 MQT: tele/bagnosala/STATE = {"Time":"2019-07-03T22:49:34","Uptime":"0T00:05:22","Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:04"}}
22:49:34 MQT: tele/bagnosala/SENSOR = {"Time":"2019-07-03T22:49:34","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":61.9},"ENERGY":{"TotalStartTime":"2019-07-03T22:48:06","Total":0.012,"Yesterday":0.000,"Today":0.012,"Period":0.00,"Power":0.35,"ApparentPower":4.53,"ReactivePower":4.50,"Factor":0.08,"Voltage":219.2,"Current":0.021},"TempUnit":"C"}
22:49:35 ADE: U 5611469, I 2253 + 0 = 2253, P 21 + 0 = 21
22:49:36 ADE: U 5607148, I 2253 + 0 = 2253, P 78 + 0 = 78
22:49:37 ADE: U 5607825, I 2253 + 0 = 2253, P 55 + 0 = 55
22:49:38 ADE: U 5581130, I 2253 + 0 = 2253, P -5 + 0 = -5
22:49:39 ADE: U 5596684, I 2253 + 0 = 2253, P 79 + 0 = 79
22:49:40 ADE: U 5594365, I 2228 + 0 = 2228, P 52 + 0 = 52
22:49:41 ADE: U 5575490, I 2228 + 0 = 2228, P 48 + 0 = 48
[... 31 lines cut ...]
22:50:13 ADE: U 5571058, I 2209 + 0 = 2209, P 103 + 0 = 103
22:50:14 ADE: U 5570764, I 2209 + 0 = 2209, P 60 + 0 = 60
22:50:15 ADE: U 5573245, I 2209 + 0 = 2209, P 84 + 0 = 84
22:50:16 ADE: U 5617266, I 2209 + 0 = 2209, P 64 + 0 = 64
22:50:17 ADE: U 5619467, I 2153 + 0 = 2153, P 22 + 0 = 22
22:50:18 ADE: U 5592697, I 2153 + 0 = 2153, P 96 + 0 = 96
22:50:19 ADE: U 5597584, I 2153 + 0 = 2153, P 144 + 0 = 144
22:50:20 ADE: U 5589615, I 2153 + 0 = 2153, P 55 + 0 = 55
22:50:21 ADE: U 5592406, I 2153 + 0 = 2153, P 12 + 0 = 12
22:50:22 ADE: U 5598818, I 2153 + 0 = 2153, P 29 + 0 = 29
22:50:23 ADE: U 5572894, I 2153 + 0 = 2153, P 60 + 0 = 60
22:50:24 ADE: U 5592721, I 2153 + 0 = 2153, P 52 + 0 = 52
22:50:25 ADE: U 5592933, I 2153 + 0 = 2153, P 62 + 0 = 62
22:50:26 ADE: U 5576651, I 2153 + 0 = 2153, P 69 + 0 = 69
22:50:27 ADE: U 5580521, I 2172 + 0 = 2172, P 70 + 0 = 70
22:50:28 ADE: U 5592298, I 2172 + 0 = 2172, P 39 + 0 = 39
[... cut ...]
22:50:36 MQT: tele/bagnosala/STATE = {"Time":"2019-07-03T22:50:36","Uptime":"0T00:06:24","Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":34,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"guilan","BSSId":"E0:46:9A:77:A6:DE","Channel":4,"RSSI":72,"LinkCount":1,"Downtime":"0T00:00:04"}}
22:50:36 MQT: tele/bagnosala/SENSOR = {"Time":"2019-07-03T22:50:36","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":62.4},"ENERGY":{"TotalStartTime":"2019-07-03T22:48:06","Total":0.036,"Yesterday":0.000,"Today":0.036,"Period":23.87,"Power":0.18,"ApparentPower":4.41,"ReactivePower":4.40,"Factor":0.04,"Voltage":218.9,"Current":0.020},"TempUnit":"C"}

guiumb, 2019-07-03:
I have to flash dev with precompiled ver 6.5.0 ? Actualli run on 6.5.0.16.

arendst, 2019-07-03:
There are pre compiled devs too but no need.

I see you have negative values where you should not have them. I will add a test for this to solve the issue in the next days.

guiumb, 2019-07-03:
Ok! Great!

arendst, 2019-07-04:
Give it a go.

guiumb, 2019-07-04:
I am in office now. As soon as I'll arrive to home I will compile the new code and i will flash dev.
I will forward you a feedback.

guiumb, 2019-07-04:
I have updated Shelly with last code and seems work: in 15 minutes with low load (<4 W) no historical data increased.
I will keep it on monitor and I will forward news.
Do you need some logs or dumps?

arendst, 2019-07-04:
No need as problem is solved.

ascillato2, 2019-07-04:
Closing this issue as it is solved. Thanks everyone for working on this.

closed 2019-07-04 by ascillato2