arendst/Tasmota #5969: WS2812 ghost switching

demian85, 2019-06-20:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : wemos d1 mini
- [ ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x] Provide the output of command ``status 0`` :
```
19:11:49 CMD: status 0
19:11:49 SRC: WebConsole from 192.168.0.11
19:11:49 RSL: Group 0, Index 1, Command STATUS, Data 0
19:11:49 MQT: stat/wemos/STATUS = {"Status":{"Module":18,"FriendlyName":["wemos","wemos"],"Topic":"wemos","ButtonTopic":"0","Power":2,"PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,1,4,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":0}}
19:11:49 MQT: stat/wemos/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/sonoff-display.bin","RestartReason":"Exception","Uptime":"0T00:00:38","StartupUTC":"2019-06-20T22:11:11","Sleep":50,"CfgHolder":4617,"BootCount":500,"SaveCount":5974,"SaveAddress":"F6000"}}
19:11:50 MQT: stat/wemos/STATUS2 = {"StatusFWR":{"Version":"6.5.0.12(a801585-display)","BuildDateTime":"2019-05-26T17:03:25","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
19:11:50 MQT: stat/wemos/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Garchatel WiFi",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["10008209","280500000100000000000000000000000000","00000000"]}}
19:11:50 MQT: stat/wemos/STATUS4 = {"StatusMEM":{"ProgramSize":531,"Free":472,"Heap":9,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","0FAAA39C","00178FA0","23B617C0","010028C0"]}}
19:11:50 MQT: stat/wemos/STATUS5 = {"StatusNET":{"Hostname":"wemos-2497","IPAddress":"192.168.0.12","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"BC:DD:C2:81:09:C1","Webserver":2,"WifiConfig":5}}
19:11:50 MQT: stat/wemos/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.10","MqttPort":1883,"MqttClientMask":"wemos","MqttClient":"wemos","MqttUser":"wemos-7wmaw9ta8h932o8p","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:11:50 MQT: stat/wemos/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 20 22:11:50 2019","Local":"Thu Jun 20 19:11:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"-03:00","Sunrise":"07:59","Sunset":"17:50"}}
19:11:50 MQT: stat/wemos/STATUS10 = {"StatusSNS":{"Time":"2019-06-20T19:11:50","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","AM2301":{"Temperature":22.0,"Humidity":73.6},"BMP280":{"Temperature":23.9,"Pressure":1019.6},"BH1750":{"Illuminance":12},"PressureUnit":"hPa","TempUnit":"C"}}
19:11:50 MQT: stat/wemos/STATUS11 = {"StatusSTS":{"Time":"2019-06-20T19:11:50","Uptime":"0T00:00:39","SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Dimmer":20,"Color":"330000","HSBColor":"0,100,20","Channel":[20,0,0],"Scheme":7,"Width":0,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```

```
Hey guys. I've been trying to solve this issue for some time but still cannot find a solution. 
I have a 8-led strip configured as ws2812 in GPIO D6. I tried other pins as well.
The strip powers on automagically for no reason. PowerOnState and PowerRetain are both disabled.
The strip is connected to 5v and I'm using a logic level converter for the data pin.
What can be causing this issue?

Thanks!

ascillato, 2019-06-21:
Hi,

Please, complete the last part of the template. Use weblog 4 and post here the output of the console when your issue happens. Thanks

demian85, 2019-06-21:
Right. But the thing is I can't keep monitoring the console 24x7 because
this can happen at any moment. I may need to configure some kind of alert
when power goes on so I can enter the console quick.

Maybe it is good to mention that I added a 200ohm resistor to the data pin
just in case.

meingraham, 2019-06-21:
If your problem is intermittent, they you need to set up a remote syslog and turn on syslog logging on the Tasmota device. SysLog is documented in the wiki.

demian85, 2019-06-21:
That's just great. I enabled syslog and weblog 4 and now the device keeps
rebooting itself. I cannot use the web admin anymore. Do I need to flash
firmware again?

andrethomas, 2019-06-21:
> The strip is connected to 5v and I'm using a logic level converter for the data pin.

This is not necessary as the WS2812 will read the logic level of 3.3V perfectly fine.

Have you considered that the level shifter may be introducing the problem? :)

meingraham, 2019-06-21:
Flash again - Probably.

You probably don't need full debug logging but definitely you don't need debug on the weblog if you're trying to capture intermittent issues. Set syslog 2 and weblog 2 to start. If that doesn't trap the issue, then up syslog to 4.

Cheers!

ascillato2, 2019-06-23:
@demian85 

Hi, any news on this?

demian85, 2019-06-23:
Bad news. I removed the level shifter and the problem persists. I didn't
have time yet to setup a syslog.
Can someone please explain how a Led strip that requires a driver to work
can be powered on by itself without any intervention of the
microcontroller? At least I would like to understand if some signal is
being picked up by mistake or the esp is magically turning the led on.
I double checked and no one is sending the power cmd to it...

ascillato, 2019-06-23:
I never had that issue. I have several of these leds installed for more than a year and they never power up by themselves.

May be your leds are defective. Some other people have reported in old issues weird problems with low quality clones of these leds.

demian85, 2019-06-23:
Okay, but it doesn't explain how the leds are controlled by themselves.
Does that make sense? They even power up using the Christmas scheme...

ascillato, 2019-06-23:
For a scheme, it requires Tasmota to send all those commands. So, in that case Tasmota is power them up.

So, for troubleshooting that, you need to set weblog or syslog to 4 to let Tasmota tells you, where that 'turn on' command cames from

+ Have you set any button or switch?
+ Are you powering the led strip using a more than 2 amps power source?

andrethomas, 2019-06-23:
Also make sure you have the correct number of pixels configured by checking the output of the  `Pixels` command

demian85, 2019-06-23:
Well, I found the logs when this happened, and it tells me nothing.
```
16:30:33 MQT: tele/wemos/STATE = {"Time":"2019-06-23T16:30:33","Uptime":"0T00:20:15","Vcc":2.732,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Dimmer":20,"Color":"333333","HSBColor":"0,0,20","Channel":[20,20,20],"Scheme":7,"Width":0,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:30:33 MQT: tele/wemos/SENSOR = {"Time":"2019-06-23T16:30:33","Switch1":"ON","Switch2":"ON","Switch3":"OFF","AM2301":{"Temperature":22.1,"Humidity":83.4},"BMP280":{"Temperature":24.4,"Pressure":1008.3},"BH1750":{"Illuminance":31},"PressureUnit":"hPa","TempUnit":"C"} (retained)
16:30:33 SRC: MQTT
16:30:33 SRC: MQTT
16:30:41 SRC: MQTT
16:30:41 SRC: MQTT
16:30:52 SRC: MQTT
16:30:58 SRC: MQTT
16:30:58 SRC: MQTT
16:44:59 SRC: MQTT
16:45:08 SRC: MQTT
16:45:08 SRC: MQTT
16:45:11 SRC: MQTT
16:45:37 MQT: tele/wemos/STATE = {"Time":"2019-06-23T16:45:37","Uptime":"0T00:09:14","Vcc":2.714,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":999,"POWER1":"ON","Dimmer":20,"Color":"333333","HSBColor":"0,0,20","Channel":[20,20,20],"Scheme":7,"Width":0,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":9,"RSSI":100,"LinkCount":2,"Downtime":"0T00:00:06"}}
16:45:37 MQT: tele/wemos/SENSOR = {"Time":"2019-06-23T16:45:37","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","AM2301":{"Temperature":21.6,"Humidity":82.4},"BMP280":{"Temperature":24.2,"Pressure":1008.5},"BH1750":{"Illuminance":30},"PressureUnit":"hPa","TempUnit":"C"} (retained)
16:45:37 SRC: MQTT
16:45:37 SRC: MQTT
```

As you can see, state says POWER1 = OFF and some time later it says ON, what happened in between? what was the trigger?

I have 3 switches but SwitchTopic<n> is empty for all 3 of them. I'm using rules to capture the state change:

`on Switch1#State do publish2 stat/wemos/SWITCH1 %value% endon on Switch2#State do publish2 stat/wemos/SWITCH2 %value% endon on Switch3#State do publish stat/wemos/SWITCH3 1 endon`

Switch 1 and 2 are motion sensors, switch 3 is just a button.
Pixels are correctly set to 8.

demian85, 2019-06-23:
Sorry, I just realized that 15 minutes passed between those 2 logs and I don't understand why. I've set up syslog with debug level. I'll wait and check again with more  detail.

demian85, 2019-06-23:
Okay, sorry for the spam, here it is. Device is restarting and then an empty POWER cmd that seems to be retained since old times is triggering the led on (???)

Check the exception message:
```
Jun 23 17:25:40 wemos-2497 ESP-MQT: tele/wemos/STATE = {"Time":"2019-06-23T17:25:40","Uptime":"0T00:49:17","Vcc":2.732,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","Dimmer":20,"Color":"333333","HSBColor":"0,0,20","Channel":[20,20,20],"Scheme":7,"Width":0,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":9,"RSSI":100,"LinkCount":2,"Downtime":"0T00:00:06"}}
Jun 23 17:25:40 wemos-2497 ESP-MQT: tele/wemos/SENSOR = {"Time":"2019-06-23T17:25:40","Switch1":"ON","Switch2":"ON","Switch3":"OFF","AM2301":{"Temperature":22.7,"Humidity":85.5},"BMP280":{"Temperature":24.4,"Pressure":1008.2},"BH1750":{"Illuminance":31},"PressureUnit":"hPa","TempUnit":"C"} (retained)
Jun 23 17:30:14 wemos-2497 ESP-WIF: Connected
Jun 23 17:30:14 wemos-2497 ESP-HTP: Web server active on wemos-2497 with IP address 192.168.0.12
Jun 23 17:30:16 wemos-2497 ESP-MQT: Attempting connection...
Jun 23 17:30:16 wemos-2497 ESP-MQT: Connected
Jun 23 17:30:16 wemos-2497 ESP-MQT: tele/wemos/LWT = Online (retained)
Jun 23 17:30:16 wemos-2497 ESP-MQT: cmnd/wemos/POWER =
Jun 23 17:30:16 wemos-2497 ESP-MQT: stat/wemos/RESULT = {"POWER1":"ON"}
Jun 23 17:30:16 wemos-2497 ESP-MQT: stat/wemos/POWER1 = ON
Jun 23 17:30:16 wemos-2497 ESP-MQT: tele/wemos/INFO1 = {"Module":"Generic","Version":"6.5.0(sonoff)","FallbackTopic":"cmnd/wemos_fb/","GroupTopic":"sonoffs"}
Jun 23 17:30:16 wemos-2497 ESP-MQT: tele/wemos/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos-2497","IPAddress":"192.168.0.12"}
Jun 23 17:30:16 wemos-2497 ESP-MQT: tele/wemos/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4000e1f0 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000"}
Jun 23 17:30:16 wemos-2497 ESP-MQT: stat/wemos/RESULT = {"POWER1":"ON"}
Jun 23 17:30:24 wemos-2497 ESP-MQT: Attempting connection...
Jun 23 17:30:24 wemos-2497 ESP-MQT: Connected
Jun 23 17:30:24 wemos-2497 ESP-MQT: tele/wemos/LWT = Online (retained)
Jun 23 17:30:24 wemos-2497 ESP-MQT: cmnd/wemos/POWER =
Jun 23 17:30:24 wemos-2497 ESP-MQT: stat/wemos/RESULT = {"POWER1":"ON"}
Jun 23 17:30:24 wemos-2497 ESP-MQT: stat/wemos/POWER1 = ON
Jun 23 17:30:29 wemos-2497 ESP-MQT: tele/wemos/STATE = {"Time":"2019-06-23T17:30:29","Uptime":"0T00:00:14","Vcc":2.714,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":782,"POWER1":"ON","Dimmer":20,"Color":"333333","HSBColor":"0,0,20","Channel":[20,20,20],"Scheme":7,"Width":0,"Fade":"OFF","Speed":2,"LedTable":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"Garchatel WiFi","BSSId":"CC:2D:E0:A6:11:71","Channel":9,"RSSI":100,"LinkCount":2,"Downtime":"0T00:00:06"}}
Jun 23 17:30:29 wemos-2497 ESP-MQT: tele/wemos/SENSOR = {"Time":"2019-06-23T17:30:29","Switch1":"ON","Switch2":"ON","Switch3":"OFF","AM2301":{"Temperature":22.9,"Humidity":84.7},"BMP280":{"Temperature":24.5,"Pressure":1008.0},"BH1750":{"Illuminance":31},"PressureUnit":"hPa","TempUnit":"C"} (retained)
Jun 23 17:30:44 wemos-2497 ESP-RUL: SWITCH1#STATE```

ascillato, 2019-06-23:
Yes, that is. You have retained messages in your mqtt broker and also you have some problem in your firmware that makes a reboot with exception.

You need to clear your mqtt broker database and also erase your firmware with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py also.

ascillato, 2019-06-23:
The exception list descriptions are in https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst

Mainly your restart issue is due to bad flashing (see wiki for the correct way) or not enough power due to small or noisy power supply, or bad regulator on your board.

Please, in order to solve your issue:

+ Check your power supply to be more than 2 amps for driving safely your leds.
+ erase your flash with esptool.py as explained in the wiki 
+ follow the steps for flashing explained in the wiki (use esptool.py) 
+ erase your mqtt broker database 

Thanks

ascillato2, 2019-06-24:
Hi,

Could you solve your issue?

demian85, 2019-06-24:
Will try later. You didn't even give me time to sleep!
Thanks!

ascillato2, 2019-06-26:
Hi, any news on this?

demian85, 2019-06-26:
Did not give it a try yet. I'll reopen if necessary.
Thanks.

ascillato2, 2019-06-26:
Ok. When you have time, tell us if you could solve your issue. Thanks

closed 2019-06-26 by ascillato2