arendst/Tasmota #5970: Tuya Dimmer doesn´t switch off completly

seliku, 2019-06-21:
Hi Everybody

Flashing this Tuya Dimmer with 6.5 went smoothly. I installed it and it responds to on/off and Dimming the Lights. But, If I toggle it to switch off, the light dims slowly down to about 10% and stays there.

The console says following:
`09:50:08 TYA: RX Packet: "55aa0007000501010001000e"
09:50:08 TYA: RX - Off State
09:50:08 TYA: RX Packet: "55aa00070008020200040000003d53"
09:50:08 TYA: RX Dim State=61
09:50:08 TYA: RX Packet: "55aa0007000501010001000e"
09:50:08 TYA: RX - Off State
09:50:08 TYA: RX Packet: "55aa00070008020200040000003a50"
09:50:08 TYA: RX Dim State=58
09:50:08 TYA: RX Packet: "55aa0007000501010001000e"
09:50:08 TYA: RX - Off State
09:50:08 TYA: RX Packet: "55aa0007000802020004000000384e"
09:50:08 TYA: RX Dim State=56
09:50:08 TYA: RX Packet: "55aa0007000501010001000e"
09:50:08 TYA: RX - Off State
09:50:08 TYA: RX Packet: "55aa0007000802020004000000354b"
09:50:08 TYA: RX Dim State=53
09:50:08 TYA: RX Packet: "55aa0007000501010001000e"
09:50:08 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000003248"
09:50:09 TYA: RX Dim State=50
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000002f45"
09:50:09 TYA: RX Dim State=47
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000002c42"
09:50:09 TYA: RX Dim State=44
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000002a40"
09:50:09 TYA: RX Dim State=42
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa0007000802020004000000273d"
09:50:09 TYA: RX Dim State=39
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa0007000802020004000000243a"
09:50:09 TYA: RX Dim State=36
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000002137"
09:50:09 TYA: RX Dim State=33
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000001e34"
09:50:09 TYA: RX Dim State=30
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000001c32"
09:50:09 TYA: RX Dim State=28
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa0007000802020004000000192f"
09:50:09 TYA: RX Dim State=25
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa0007000802020004000000162c"
09:50:09 TYA: RX Dim State=22
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000001329"
09:50:09 TYA: RX Dim State=19
09:50:09 TYA: RX Packet: "55aa0007000501010001000e"
09:50:09 TYA: RX - Off State
09:50:09 TYA: RX Packet: "55aa00070008020200040000001026"
09:50:09 TYA: RX Dim State=16
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa00070008020200040000000e24"
09:50:10 TYA: RX Dim State=14
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa00070008020200040000000b21"
09:50:10 TYA: RX Dim State=11
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa0007000802020004000000081e"
09:50:10 TYA: RX Dim State=8
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa0007000802020004000000051b"
09:50:10 TYA: RX Dim State=5
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa00070008020200040000000218"
09:50:10 TYA: RX Dim State=2
09:50:10 TYA: RX Packet: "55aa0007000501010001000e"
09:50:10 TYA: RX - Off State
09:50:10 TYA: RX Packet: "55aa00070008020200040000000016"
09:50:10 TYA: RX Dim State=0`

Setoption32 is 40. I dont know what else to check.

Thank you

jsponz, 2019-07-10:
You tested with the original firmware? 

I bought two units. The first one had this problem with Tasmota and I tested the second one with the original firmare with the same result. It was a hardware problem!

They sent me two new pieces...

ascillato2, 2019-07-11:
Closing issue as seems to be a hardware issue. Sorry.

If you need further assistant, please, do not hesitate on asking here or in the Tasmota Support Chat. Thanks.

closed 2019-07-11 by ascillato2