arendst/Tasmota #5973: sonoff http or mqtt

Buzda, 2019-06-22:
Hi there!

I am wondering what is suitable for my case, using espeasy or Tasmota. I know espeasy is using http requests and Tasmota is using mqtt.

I want to control my sonoff devices by a raspberry pi that is acting as a home automation hub, and it in turn send updates and receives commands from AWSIoT platform. For interacting with AWSIoT platform, it uses mqtt.

What are the pros and cons of using either? and will it cause problems if I control several sonoff devices with http, while using mqtt for AWSIoT? or better use mqtt for all AWSIoT and sonoff?

meingraham, 2019-06-22:
Tasmota can also use http.

s-hadinger, 2019-06-22:
What are you using on the Raspberry PI? AWS Greengrass?

I would rather keep it simple and do all in MQTT for the sake of consistency.

By the way, do you know you can directly talk to AWS IoT from Tasmota? Details here: https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT

closed 2019-06-22 by ascillato2