arendst/Tasmota #5975: DNS not resolving *.xyz.local hostname, mDNS disabled, Windows DNS server

FelR, 2019-06-22:
Hi folks,

I'm trying to configure my Shelly 2.5 modules. I'm using version 6.5.0.16.

My domain at home started at a time when mDNS (RFC 6762) wasn't known and Microsoft was recommending .local as domain for Active Directory installs. Therefore all my servers have a fqdn like server1.xyz.local.

In Tasmota mDNS is turned off by setting option 55 to 0. Nevertheless the software isn't able to resolve the ip of my mqtt broker (mqtt-broker.xyz.local). Wireshark shows no DNS packets are sent from the tasmota device to the dns server.

In version 6.3.0.8 it's working fine.

FelR, 2019-06-22:
Update: I recompiled versions 6.5.0.16 using Arduino Core 2.3.0. Now it's working.

But I don't understand why. When mDNS is disabled ALL dns queries have to be sent to the configured DNS server. It must not happen that those queries are swallowed by the system with no chance for processing, neither by the mDNS daemon nor by the configured DNS server.

ascillato2, 2019-06-22:
Great to know that it is working now for you.

In the first attempt may be you didn't compile with mDNS enabled. 

If you need assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-22 by ascillato2

FelR, 2019-06-23:
Hi,

sorry to bring this up again. But I don't think this issue is solved. When using the released bin file, these dns queries don't work. When I compile it myself with Arduino Core 2.3.0 it works. So for me there seems to be a bug in Arduino Core.

My issue looks similiar to #4670 and #4781.

Even in these two issues no solution was found.