arendst/Tasmota #5979: Wi-Fi

probonopd, 2019-06-24:
**Have you look for this feature in other issues and in the wiki?**

Yes

**Is your feature request related to a problem? Please describe.**

I would like the device to connect to the Wi-Fi, fetch the time and maybe configuration, then switch off the Wi-Fi (e.g., during night time) and just execute rules locally. Upon device power cycling, it would connect to Wi-Fi again and the thing would repeat.

**Describe the solution you'd like**

A command to switch Wi-Fi on and off in rules.

**Describe alternatives you've considered**

Not using Tasmota for the task.

ascillato2, 2019-06-24:
Closing this issue as it is **duplicated** (#4330). Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-24 by ascillato2

probonopd, 2019-06-24:
Please reopen this ticket.

I had seen #4330 and had come to the conclusion that
* This is a different feature request. I am not asking to automatically turn off the Wi-Fi, but to have a function to turn __on and off__ Wi-Fi in __rules__
* The answer given in #4330, "turning off the wifi of your smart device make it not a home automation device" does not apply to this feature request

Also I have read the policy. This is a feature request and now a "how do I..." type question.

ascillato, 2019-06-24:
Hi,

Sorry about the phrase "read the contributing....". Was not intended to you. As we have a lot of issues and requests, we use predefined answers. I forgot to delete that sentence. In fact, you did the feature request correctly and you did explain it very well. 

The request to have a command to turn off/on wifi was asked before and as pointed in the referenced issue, is not in the actual main goal of Tasmota. Anyway, that issue has the label: **requested feature - hold over**, for future reference and may be someone would work on that. So, it is not discarded but not to be worked on right now.

ascillato, 2020-06-16:
Now it is available the command `WIFI` to turn ON and OFF the wifi from the Tasmota Console. Please, use latest development version.

ilvalle, 2020-09-20:
Hi there, I've tried the `wifi off` command, is it documented any where?My experience is that It disconnects the devices from my wifi, however it starts to advertise a new wifi `tasmota-xxxx` with the network incomplete 

```
20:35:14 RSL: tasmota/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3250","IPAddress":"(IP unset)","Gateway":"192.168.100.250","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.250","Mac":"C8:2B:96:02:EC:B2","Webserver":2,"WifiConfig":5,"WifiPower":17.0}}
```