arendst/Tasmota #5980: Wemos D1 Mini pro 16 MB can't connect to Wifi

z1969, 2019-06-24:
After flashing the original 6.5 Binary sonoff-DE.bin the Wemos won't startup in AP-Mode.
I tried also Version 6.1 to 6.4. On this releases the Wemos start in AP-Mode and also scanning SSID's are possible but after restart it start's again in AP Mode.
It Looks like the Wifi-Credentials are not saved.

After that I try ESPeasy to cross check and this works.

The Wemos D1 Mini 4MB work perfectly
For flashing I use NodeMCU-Pyflasher

Is the 16MB Version currently not supported?

Kind Regards
Chris

PS: Many thanks for this great Software

ascillato2, 2019-06-24:
Hi, Tasmota works on any flash size bigger than 500k, but requires to be flashed as explained in the wiki.
Please, use the precompiled bins for flashing. There is no need to make different binaries for different memory maps. The same binary will work. 

If you need further assistant, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-06-24 by ascillato2

PipeDeveloper, 2019-10-15:
Same problem here, but im using the pre-compiled sonoff.bin (english recomended), Wemos D1 mini PRO start up in AP-Mode but after configure my SSID and password it doesnt connect.

mike2nl, 2019-10-15:
@ascillato2 
Hi all,
yesterday i helped out another dutch Team member and we have tested a lot with the Wemos D1 mini original. We are sure that they are original because the new Logo LoLin.

He used 1MB and 4MB bin files voa OTA. After the update and setup the wifi things he can't connect anymore. Then i compiled three version for him (3x wemos). 2x 4MB and 1x 1MB so far i remember right. He used everytime one new Wemos D1 Mini. 

With my compiled versions, uploaded via discord, he was able to connect in no time. He has done settings and the devices still running. When some screenshots are needed i can download them from the personal message chat from discord

- His used core's is:
before update:  V6.6.0.3 Core 2.5.2 
cores working until: V6.6.0.12 Core2_6
After that no more connection possible after setup the wifi.
- My compiled version core:
V6.6.0.18 Core 2_6pre (0.0.1 based on a link from **Jason2866, Thanks BTW**)
With the compiled binary which i uploded to discord all units were working so as has to be.

The question for me is: 
Which are the board settings while compile for the Wemos D1 mini? I don't say there is something wrong!

But when self compiled (my work for him) 1MB bin and after that a 4MB bin and after that a 1MB bin again and all is working as has to be, then there can/must be some things. Mostly user issues that's right in my own experience and i had discussed that with Jason2866 in private. But we must get it clear what the users have done.

But one point which can give some clearity is to setup/reset tghe Wemos D1 mini back to factory settings and start again.

Fatctory setting description:
https://wiki.wemos.cc/tutorials:get_started:revert_to_at_firmware

Another idea is that i will upload my bin files to google drive let say and he edit the OTQ url to that and download them from there and we can test it. Is this an idea at all to see how things working for him?

z1969, 2019-10-15:
Hi everybody

I solved my problem with the compiled 4MB version from:
http://thehackbox.org/tasmota/

After flashing that binary, the Wemos D1 pro with 16MB works and a I was able to connect to the Wlan.
Also an OTA update to latest official Version from GitHub works without any issues.

Maybe this work around help you.

Kind Regards
Chris

ascillato, 2019-10-15:
The different memory map version are not OTA compatible between them. You need to erase all flash with esptool.py as explained in the wiki and flash again with the precompiled bins with esptool.py in order to that change to work. 

Doesn't matter if you choose 1mb or 4mb or any other memory map version for a 16mb version. What does matter is to use the same version all the time because if you do an OTA to a different memory map version, your device will stuck.

It is recommended to use the 1mb precompiled bin for all devices so you can update without issues. Having devices with different memory map firmwares will unnecessarily overcomplicating your installation.

carloszacu, 2020-05-02:
what I did is using the generic ESP6266 board instead of the wemos one, that turned out OK for me.

PipeDeveloper, 2020-05-02:
Problem was because tasmota doesnt recognize SSID with special characters like $%(!.- that was the problem and thats why it didn't connect properly after putting wifi info at the first connection.