arendst/Tasmota #5981: Shelly 1 - switchtopic disables switch

vb8080, 2019-06-24:
Hi,

my Shelly 1 is working as designed and as it should .
I can switch from WEB and i can use the wall switch either.
everything ok.

Now i want to use the GPIO01 to add an PIR Motion Sensormodule HC-SR501 to have a notification as a home alert system when i'm not home...

so i hooked up 2 wire, one to GND and the other to GPIO01, confugured everything and testd it by connecting those wires together... and the light turned out .. 

WEB
21:59:25 MQT: stat/light/RESULT = {"POWER":"OFF"}
21:59:25 MQT: stat/light/POWER = OFF
21:59:25 MQT: stat/light/RESULT = {"POWER":"ON"}
21:59:25 MQT: stat/light/POWER = ON

Button
21:59:33 MQT: stat/light/RESULT = {"POWER":"OFF"}
21:59:33 MQT: stat/light/POWER = OFF
21:59:33 MQT: stat/light/RESULT = {"POWER":"ON"}
21:59:33 MQT: stat/light/POWER = ON

GPIO01
21:59:46 MQT: stat/light/RESULT = {"POWER":"OFF"}
21:59:46 MQT: stat/light/POWER = OFF
21:59:47 MQT: stat/light/RESULT = {"POWER":"ON"}
21:59:47 MQT: stat/light/POWER = ON

great everything works as designed...
now i thought i just need to do a switchtopic and i'm fine...
but the behavior i got was not what i was expecting:

22:00:24 CMD: switchtopic radartest
22:00:24 MQT: stat/light/RESULT = {"SwitchTopic":"radartest"}

WEB
22:00:27 MQT: stat/light/RESULT = {"POWER":"OFF"}
22:00:27 MQT: stat/light/POWER = OFF
22:00:28 MQT: stat/light/RESULT = {"POWER":"ON"}
22:00:28 MQT: stat/light/POWER = ON

wall button
22:00:42 MQT: cmnd/radartest/POWER1 = TOGGLE
22:00:42 MQT: cmnd/radartest/POWER1 = TOGGLE

GPIO01
22:00:46 MQT: cmnd/radartest/POWER2 = TOGGLE
22:00:47 MQT: cmnd/radartest/POWER2 = TOGGLE

from the web interface i still can turn the light on and off...
but not anymore from my wall mounted switch ....
seems it chnaged everywhere the switchtopic... 

How do i cange the behavior that i only change the "switchtopic" only for GPIO01 ?
i looked evrywhere but could not find something ....

thanks for the reply...

digiblur, 2019-06-24:
set the switchtopic back to 0 and use a rule is the best way to accomplish this.

https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches#switchtopic

vb8080, 2019-06-24:
Great thanks! 

i configured the GPIO01 to use Switch02
and added the following rule and now it's working as i wanted ...

thanks a lot!

Rule ADDED
23:50:42 CMD: Rule1 on switch2#state do publish cmnd/testradarmqtt/POWER %value% endon
23:50:42 MQT: stat/light/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":445,"Rules":"on switch2#state do publish cmnd/testradarmqtt/POWER %value% endon"}
23:50:51 CMD: Rule1 1
23:50:51 MQT: stat/light/RESULT = {"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":445,"Rules":"on switch2#state do publish cmnd/testradarmqtt/POWER %value% endon"}

WEB
23:50:59 MQT: stat/light/RESULT = {"POWER":"OFF"}
23:50:59 MQT: stat/light/POWER = OFF
23:51:00 MQT: stat/light/RESULT = {"POWER":"ON"}
23:51:00 MQT: stat/light/POWER = ON

lightswitch
23:51:25 MQT: stat/light/RESULT = {"POWER":"OFF"}
23:51:25 MQT: stat/light/POWER = OFF
23:51:26 MQT: stat/light/RESULT = {"POWER":"ON"}
23:51:26 MQT: stat/light/POWER = ON

GPIO01
23:51:30 RUL: SWITCH2#STATE performs "publish cmnd/testradarmqtt/POWER 2"
23:51:30 MQT: cmnd/testradarmqtt/POWER = 2
23:51:30 RUL: SWITCH2#STATE performs "publish cmnd/testradarmqtt/POWER 2"
23:51:30 MQT: cmnd/testradarmqtt/POWER = 2

closed 2019-06-24 by ascillato2